raspberrypi/firmware #1437: 3.5mm audio interface cannot output sound, but HDMI can output sound

Dolphinzzx, 2020-07-29:
I am using Raspberry Pi4(8G). 3.5mm audio interface cannot output sound, but HDMI can output sound.
I have set [sudo raspi-config]---[7 Advanced Options]--[A4 Audio]--[1 Headphones]. But 3.5mm audio interface cannot output sound all the time.

popcornmix, 2020-07-29:
I've just tried with latest RaspiOS 32 image and used the speaker icon on taskbar.
I right-click and choose "AV Jack" and the next program I run has audio.
I also tried with command line raspi-config.

Note: It won't change audio destination for a running application - it just affects the default audio destination used when apps start.

Dolphinzzx, 2020-07-29:
> I've just tried with latest RaspiOS 32 image and used the speaker icon on taskbar.
> I right-click and choose "AV Jack" and the next program I run has audio from 3.5mm socket.
> I also tried with command line raspi-config.
> 
> Note: It won't change audio destination for a running application - it just affects the default audio destination used when apps start.

I also latest RaspiOS 32 image. and i also right-click and choose "AV Jack" . But speaker is not make sound.

popcornmix, 2020-07-29:
What app are you using to make sound? I'm just running speaker-test from console.

Trotter73, 2020-07-29:
Hi,

 This will be another me too post until I get to the Pi and get some hard data for you.

Pi4 4Gb, on the 4 series kernels there was no issue, update to the latest 5 series kernels and the audio stops, I have a cinnamon desktop so the immediate thing to notice is that the various "system" sounds are missing. Cinnamon was installed over the top of PiOS32...

On from that, other specific cases, Doom (the game) did not output any sound, neither did moc player (command line audio player). 

I cant be more specific because I rolled back to the 4 series mainly because of the screen corruption, and have not done any real investigation. The screen issues would seem to be fixed so I am more than happy to update and investigate further..

What would help with the diagnosis popcornmix ? Let me know and I'll get the info 

Regards...

popcornmix, 2020-07-29:
Ideally the steps to reproduce, starting from a clean installed PiOS32 image. What do I install, click and run to see your issue?

Output from:
vcgencmd version
uname -a
aplay -l
may provide some clues.

Dolphinzzx, 2020-07-29:
> Ideally the steps to reproduce, starting from a clean installed PiOS32 image. What do I install, click and run to see your issue?
> 
> Output from:
> vcgencmd version
> uname -a
> aplay -l
> may provide some clues.

[image]

popcornmix, 2020-07-29:
I think "Subdevices: 3/4" means alsa is in use (I don't see that except when speaker-test is running). 
What does:
fuser -v /dev/snd/*
report?

arrowgent, 2020-08-19:
i also experienced this with raspbian (rpi4b)
some update caused hdmi to become the primary audio output, the 3.5mm no longer output audio

i do not have pulseaudio

editing config.txt (raspi-config did not properly set this option -> advanced -> audio)
```
###hdmi audio off, default to 3.5mm analogue
hdmi_ignore_edid_audio=1
```
resolved this odd change for me.