raspberrypi/firmware #1438: SC16is752 ttySCx device not longer availible

Thomas-GeDaD, 2020-07-29:
**Describe the bug**
I used several SC16IS752 on a Pi 4. 
After the Last Update, the ttySCx devices are not longer available. After downgrading, the devices are available again. (4.19)
Seems the devices are recognized correct, but the device in /dev/ are not created.
Any Idears?

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
PI4
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage5
* Which firmware version (`vcgencmd version`)?
Jul 17 2020 10:59:17 
Copyright (c) 2012 Broadcom
version 21a15cb094f41c7506ad65d2cb9b29c550693057 (clean) (release) (start)
* Which kernel version (`uname -a`)?
Linux marine-control-server 5.4.51-v8+ #1329 SMP PREEMPT Wed Jul 29 10:33:36 BST 2020 aarch64 GNU/Linux

**Logs**
-dmesg:
[    5.551930] serial serial0: tty port ttySC0 registered
[    5.583411] serial serial1: tty port ttySC1 registered
[    5.620841] serial serial2: tty port ttySC2 registered
[    5.626098] serial serial3: tty port ttySC3 registered
[    5.641702] serial serial4: tty port ttySC4 registered
[    5.646916] serial serial5: tty port ttySC5 registered

-sudo vcdbg log msg
005693.525: brfs: File read: /mfs/sd/overlays/sc16is752-i2c.dtbo
005744.434: Loaded overlay 'sc16is752-i2c'
005744.463: dtparam: int_pin=13
005744.950: dtparam: addr=0x4c
005745.863: dtparam: xtal=14745600
005893.915: brfs: File read: 1356 bytes
005895.258: brfs: File read: /mfs/sd/overlays/sc16is752-i2c.dtbo
005946.342: Loaded overlay 'sc16is752-i2c'
005946.366: dtparam: int_pin=12
005946.834: dtparam: addr=0x49
005947.744: dtparam: xtal=14745600
006097.264: brfs: File read: 1356 bytes
006098.597: brfs: File read: /mfs/sd/overlays/sc16is752-i2c.dtbo
006149.818: Loaded overlay 'sc16is752-i2c'
006149.841: dtparam: int_pin=6
006150.311: dtparam: addr=0x48
006151.224: dtparam: xtal=14745600

-cat /proc/tty/drivers
unknown              /dev/ttySC    235 0-7 serial

**Additional context**
Add any other relevant context for the problem.

Thomas-GeDaD, 2020-07-29:
[image]

pelwell, 2020-07-29:
To be clear, you are saying that the kernel driver is outputing:
```
[ 5.551930] serial serial0: tty port ttySC0 registered
[ 5.583411] serial serial1: tty port ttySC1 registered
[ 5.620841] serial serial2: tty port ttySC2 registered
[ 5.626098] serial serial3: tty port ttySC3 registered
[ 5.641702] serial serial4: tty port ttySC4 registered
[ 5.646916] serial serial5: tty port ttySC5 registered
```
and the output from `cat /proc/tty/drivers` include:
```
unknown /dev/ttySC 235 0-7 serial
```
but that there is no `/dev/ttySC*`?

Thomas-GeDaD, 2020-07-29:
yes, exactly

pelwell, 2020-07-29:
Try this:
1. Comment out the dtoverlay=sc16is752-i2c entries in config.txt and reboot.
2. Run `sudo udevadm monitor -p` in one terminal.
3. Run `sudo dtoverlay sc16is752-i2c int_pin=13 addr=0x4c xtal=14745600`.
4. See if /dev/ttySC* exists.
5. Post the output from udevadm.

Thomas-GeDaD, 2020-07-29:
KERNEL[96.856680] add      /devices/platform/soc/fe804000.i2c/i2c-1/1-004c (i2c)
ACTION=add
DEVPATH=/devices/platform/soc/fe804000.i2c/i2c-1/1-004c
SUBSYSTEM=i2c
OF_NAME=sc16is752
OF_FULLNAME=/soc/i2c@7e804000/sc16is752@4c
OF_COMPATIBLE_0=nxp,sc16is752
OF_COMPATIBLE_N=1
MODALIAS=of:Nsc16is752T(null)Cnxp,sc16is752
SEQNUM=1789

KERNEL[96.884522] add      /module/regmap_i2c (module)
ACTION=add
DEVPATH=/module/regmap_i2c
SUBSYSTEM=module
SEQNUM=1790

KERNEL[96.887343] add      /module/regmap_spi (module)
ACTION=add
DEVPATH=/module/regmap_spi
SUBSYSTEM=module
SEQNUM=1791

UDEV  [96.892598] add      /module/regmap_i2c (module)
ACTION=add
DEVPATH=/module/regmap_i2c
SUBSYSTEM=module
SEQNUM=1790
USEC_INITIALIZED=96889007

UDEV  [96.892770] add      /module/regmap_spi (module)
ACTION=add
DEVPATH=/module/regmap_spi
SUBSYSTEM=module
SEQNUM=1791
USEC_INITIALIZED=96891677

KERNEL[96.893584] add      /module/sc16is7xx (module)
ACTION=add
DEVPATH=/module/sc16is7xx
SUBSYSTEM=module
SEQNUM=1792

KERNEL[96.894100] add      /bus/i2c/drivers/sc16is7xx (drivers)
ACTION=add
DEVPATH=/bus/i2c/drivers/sc16is7xx
SUBSYSTEM=drivers
SEQNUM=1793

KERNEL[96.894241] add      /bus/spi/drivers/sc16is7xx (drivers)
ACTION=add
DEVPATH=/bus/spi/drivers/sc16is7xx
SUBSYSTEM=drivers
SEQNUM=1794

UDEV  [96.897778] add      /devices/platform/soc/fe804000.i2c/i2c-1/1-004c (i2c)
ACTION=add
DEVPATH=/devices/platform/soc/fe804000.i2c/i2c-1/1-004c
SUBSYSTEM=i2c
OF_NAME=sc16is752
OF_FULLNAME=/soc/i2c@7e804000/sc16is752@4c
OF_COMPATIBLE_0=nxp,sc16is752
OF_COMPATIBLE_N=1
MODALIAS=of:Nsc16is752T(null)Cnxp,sc16is752
SEQNUM=1789
USEC_INITIALIZED=96859644

UDEV  [96.898261] add      /bus/spi/drivers/sc16is7xx (drivers)
ACTION=add
DEVPATH=/bus/spi/drivers/sc16is7xx
SUBSYSTEM=drivers
SEQNUM=1794
USEC_INITIALIZED=96898036

UDEV  [96.898925] add      /bus/i2c/drivers/sc16is7xx (drivers)
ACTION=add
DEVPATH=/bus/i2c/drivers/sc16is7xx
SUBSYSTEM=drivers
SEQNUM=1793
USEC_INITIALIZED=96898701

UDEV  [96.899796] add      /module/sc16is7xx (module)
ACTION=add
DEVPATH=/module/sc16is7xx
SUBSYSTEM=module
SEQNUM=1792
USEC_INITIALIZED=96899559

=> No ttySCx available

pelwell, 2020-07-29:
A kernel configuration change in our 5.4 tree (`CONFIG_SERIAL_DEV_BUS=y`) has had the unintended consequence of preventing the char device from being created. More time is needed to understand why the other serial devices are unaffected and whether this is fixable.

pelwell, 2020-07-29:
See https://github.com/raspberrypi/linux/pull/3682 for some context.