raspberrypi/firmware #1440: RPi3B DPI, no output on any pins.

acidtech, 2020-07-30:
**Describe the bug**
Problem enabling DPI mode on RPi3B

**To reproduce**
List the steps required to reproduce the issue.

Enable DPI with this in config.txt

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
dtoverlay=vc4-fkms-v3d

##Enable DPI gpio
gpio=0-9,12-17,20-25=a2

enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x070016
#dpi_timings=640 1 44 2 42 480 1 16 2 14 0 0 0 75 0 28000000 1
dpi_timings=640 1 44 2 42 480 1 16 2 14 0 0 0 60 0 22364160 1

also added ssh.txt

**Expected behaviour**
I would expect the display to function. I would expect output on any/most of the DPI pins.

**Actual behaviour**
Check DPI pins with a scope. No DPI pins are outputing anything.  I checked HSync, VSync, DE DCLK and all color pins.

**System**
-bash: raspinfo: command not found

* Which model of Raspberry Pi? Pi3B
* Which OS and version (`cat /etc/rpi-issue`)? 2020-05-27-raspios-buster-lite-armhf installed
* Which firmware version (`vcgencmd version`)?
Apr 15 2020 11:43:08
Copyright (c) 2012 Broadcom
version 82f9bb929ce2186eb1824178c1ae82902ad6275c (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux

**Logs**
Tell me what I can provide.

**Additional context**
I havent setup a DPI screen on a Pi3 in a long time so I assume I am doing something stupid.  Please help. Thanks.

acidtech, 2020-07-30:
I think I may have figured it out. Pi4 has wider range of dotclocks supported I assume?  On the right track I think?  I put in a dpi_timing for another display and I see something on mine(wrong but have output now) so I assume that is the case.

acidtech, 2020-07-30:
And yes, that worked. I bumped my dotclock up to 32mhz and then increased my front porch and back porch to compensate.  Still fiddling but I would like a list of dot clock ranges supported on all the RPi models so I dont do this again in 2 years.

HuBowen1031, 2020-08-18:
It is an experiment project, It may very unstable,so,it will have many problem. But you chose that,you have to accept and adapt.

acidtech, 2020-08-18:
HuBowen1031, it appears english is not your first language so I will assume
a misunderstanding.

1. I have fixed my problem by setting a higher dotclk.
2. I asked for the list of discrete values and ranges of dotclk supported
by the RPi3B because it is NOT documented.  Since that information is in
the firmware here is the ONLY place I could ask and know someone should
have it.

Nathan Scherdin

HuBowen1031, 2020-08-19:
well,I am glad that you can use it normally,and I think you are clever,I am a Chinese,so,if I have some mistakes,please forgive me.

HuBowen1031, 2020-08-19:
Oh,I have a big problem,That is my USB drivers did not work. Can you tell me why?(Device:Rpi 4B,Sandisk 32G storage card)

acidtech, 2020-08-19:
Start a new thread. Don't hijack this one.

Nathan Scherdin

6by9, 2020-08-19:
@timg236 You were more involved in the original PV config than me. Is there a minimum pixel clock rate on Pi4?

timg236, 2020-08-19:
@6by9 - Not for PV but PV runs at core freq. Perhaps it's limited by DPI DIV which has a 4-bit integer divisor

acidtech, 2020-08-19:
The problem is Pi3, Not Pi4. Pi4 works at Dotclk/PV frequencies that the
Pi3B(+) does not. I physically tested that.  For example I used a 24mhz
clock on Pi4. No problems. Same dpi_settings on Pi3 black screen.  Up the
clock to 32mhz and Pi3 turns on with a reasonable display. But I dont want
to run a higher refresh rate than 60 since there is little benefit and
based on posts I've read there is some hit to processing power(I assume on
the VC processor).

There was a post a year or more ago about the Pi3 frequency limitations but
I can't find it anymore and the info isn't documented on the DPI page.

Nathan Scherdin

6by9, 2020-08-19:
Apologies - misread the issue and confused it with some complaints over DPI on Pi4.

Pi0-3 the pixel clock comes from a PLL. I don't recall the potential range for the divider on that (@timg236 ?).

If you add ```start_debug=1``` to config.txt, do you get any asserts logged via ```sudo vcdbg log assert```? Sorry, I don't have a Pi3 hooked up at present to try it myself.

acidtech, 2020-08-19:
No problem. I dont have mine hooked up right now either.  In the middle of
a move to a bigger office.  I'll set up a test asap and get back to you.

Nathan Scherdin

rizvanovic, 2023-04-06:
@acidtech Hi Nathan,

I'm trying to get in touch with you and found this thread. I have some questions regarding basicmicro. 

Kind regards,

Teo