raspberrypi/firmware #1442: RPI4 boot problem, waiting for root device

Nopel2020, 2020-08-03:
**Describe the bug**
My RPI4 won't boot anymore with the latest firmware (commit bd816dbac723e04f8f8b06bb2e16d767cda7692c)

I am running a Debian arm64 userland, with a custom compiled 64bit raspberrypi 5.4 kernel (custom compiled because I manually compile in Wireguard and ZFS modules).

My /boot is on an SD card, my / is on a zpool on an SSD on USB. I use an initrd with the zfs modules to mount the /. My cmdline.txt includes 'root=ZFS=rpool/ROOT/ezck rootfstype=zfs rootdelay=5'
I need the rootdelay because otherwise it tries to import the zpool before the SSD is recognised.
This has always worked fine, but starting with commit bd816dbac723e04f8f8b06bb2e16d767cda7692c of the firmware, the boot gets stuck on 'waiting for root device rpool/ROOT/ezck'

**To reproduce**
Update fixup4cd.dat and start4cd.elf to commit bd816dbac723e04f8f8b06bb2e16d767cda7692c.

**Expected behaviour**
Succesful boot.

**Actual behaviour**
Boot gets stuck on 'waiting for root device rpool/ROOT/ezck'.

**System** 
Using an RPI4 4GB.
OS: Debian 10.5 arm64
Kernel version: 5.4.51-v8+

timg236, 2020-08-03:
Have you tried increasing the rootdelay time?

Nopel2020, 2020-08-03:
Yes, even raised it to 600 seconds, but nothing. I can also see in the dmesg output that the SSD is recognised in time anyway, before it tries to import the zpool and gets stuck, so I don't think the rootdelay is the problem.

Nopel2020, 2020-08-05:
I noticed there have been some updates to the firmware, so I tried the latest commit 46e2c3ee6ceb0f64b9c4ba4dafeb852ffb4396d3 and apparently this fixed my issue! Booting normally from a ZFS root again. Thanks all!

closed 2020-08-05 by Nopel2020