raspberrypi/firmware #1443: Analog audio output "pop" upon reboot (and after startup playing audio)

snblackout, 2020-08-04:
**Is this the right place for my bug report?**
Yes.  Audio output related (analog audio)

**Describe the bug**
When analog audio output is set as default ALSA output, when you issue a reboot, upon the last phases of the shutdown process, there is an audible "pop" that is sent through the analog output and quite noticeable on the speakers if kept turned up.

Also upon resuming the OS and opening say, Chrome, and starting to play a video with audio there is another "pop" but is much more quiet and less noticeable, but is there if you listen via headphones.

**To reproduce**
Set ALSA to use the analog audio output as default.  Open a web browser or any other application that plays sound/music.  Now issue a reboot and keep your speakers on and listen for the audible "pop".  Also upon re-entering the OS, open Chrome and goto a YouTube video and listen closely (with probably headphones) and you hear another, much quieter, but there "pop".

**Expected behaviour**
Computer analog audio output should remain quiet during reboots, etc.

**Actual behaviour**
Explained above.  Audible "pop" when rebooting.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? 

Pi 4 Model B 2GB

* Which OS and version (`cat /etc/rpi-issue`)? 

Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage4

* Which firmware version (`vcgencmd version`)?

May 11 2020 18:59:41
Copyright (c) 2012 Broadcom
version 21bfdeee3a6ea823e2113b983390acd1eec8edfb (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux PIHOSTNAME 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux

Thank you for taking the time to look into this.

timg236, 2020-08-04:
The pop is the audio LDO being turned off, if it wasn't turned off there would be noise when the ROM reads from the SPI EEPROM. Unfortunately, it's just a limitation of the hardware

6by9, 2020-08-04:
The audio output is also driven from a PWM device via a high pass filter (a capacitor to block the DC), so at start/stop you're going from/to fixed high or low, to/from 50% duty cycle. That too is unavoidable and will create an audible click or pop as the capacitor charges up.

snblackout, 2020-08-04:
Thanks for the information.  Too bad that it can't be fixed in firmware.  Are the engineering team aware of this for next designs to hopefully alleviate this artifact in the design?

6by9, 2020-08-05:
The analogue audio output is done as a minimal cost thing - it uses a total of 14 external components as can be seen in the [schematic](https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/rpi_SCH_4b_4p0_reduced.pdf), and on the original Pis it was even fewer as it skipped the buffer.

If you want really high quality analogue audio then you need to look at external I2S DACs from one of the many 3rd party suppliers.

There is a config.txt option to keep the audio device open after first use, which avoids clicks between playing audio. IIRC that's ```force_pwm_open=1```.

closed 2025-02-14 by popcornmix