raspberrypi/firmware #1444: After upgrading to firmware 1.20200723-1 eth0 is crashing after some days uptime on Pi3+

cfilipem, 2020-08-05:
**Describe the bug**
After upgrading a headless Pi3+ to latest apt kernel and firmware, 1.20200723-1, eth0 started to crash after some days of uptime. It happened me twice. I never had this on 1.20200601-1.
Don't know what triggers the crash of eth0. Meanwhile, as this is a production system, I've reverted to 1.20200601-1.

**To reproduce**

Don't know how.

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW

```
Raspberry Pi 3 Model B Plus Rev 1.3
```

* Which OS and version (`cat /etc/rpi-issue`)?

```
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2
```

* Which firmware version (`vcgencmd version`)?

```
Jul 17 2020 11:04:27
Copyright (c) 2012 Broadcom
version 21a15cb094f41c7506ad65d2cb9b29c550693057 (clean) (release) (start)
```

* Which kernel version (`uname -a`)?

```
Linux rpi 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020 armv7l GNU/Linux
```

**Logs**

```
[298495.237629] ------------[ cut here ]------------
[298495.237666] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:448 dev_watchdog+0x310/0x314
[298495.237676] NETDEV WATCHDOG: eth0 (lan78xx): transmit queue 0 timed out
[298495.237684] Modules linked in: binfmt_misc wireguard ip6_udp_tunnel udp_tunnel sha256_generic libsha256 cfg80211 rfkill 8021q garp stp llc bcm2835_codec(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops raspberrypi_hwmon videobuf2_v4l2 snd_bcm2835(C) videobuf2_common snd_pcm videodev snd_timer mc snd vc_sm_cma(C) uio_pdrv_genirq uio fixed nft_masq nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv4 nft_counter nf_tables nfnetlink ip_tables x_tables ipv6 nf_defrag_ipv6
[298495.237864] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.4.51-v7+ #1327
[298495.237868] Hardware name: BCM2835
[298495.237872] Backtrace:
[298495.237891] [<8010d458>] (dump_backtrace) from [<8010d750>] (show_stack+0x20/0x24)
[298495.237901]  r6:80d00000 r5:00000000 r4:80d93ff4 r3:dd7dfe8a
[298495.237914] [<8010d730>] (show_stack) from [<808b2324>] (dump_stack+0xe0/0x124)
[298495.237927] [<808b2244>] (dump_stack) from [<8011fd24>] (__warn+0xec/0x104)
[298495.237939]  r8:000001c0 r7:00000009 r6:80b0c14c r5:00000000 r4:80d01cec r3:dd7dfe8a
[298495.237950] [<8011fc38>] (__warn) from [<8011fdc0>] (warn_slowpath_fmt+0x84/0xc0)
[298495.237962]  r9:80b0c14c r8:000001c0 r7:807bff40 r6:00000009 r5:80b0c164 r4:80d04f48
[298495.237974] [<8011fd40>] (warn_slowpath_fmt) from [<807bff40>] (dev_watchdog+0x310/0x314)
[298495.237986]  r9:01c70270 r8:b7386000 r7:00000000 r6:80d03d00 r5:b73862a8 r4:00000000
[298495.238001] [<807bfc30>] (dev_watchdog) from [<8019b6a0>] (call_timer_fn+0x40/0x180)
[298495.238011]  r8:807bfc30 r7:00000100 r6:80d00000 r5:b73862a8 r4:b77fa440
[298495.238025] [<8019b660>] (call_timer_fn) from [<8019c7f0>] (run_timer_softirq+0x288/0x654)
[298495.238037]  r9:00000000 r8:b73862a8 r7:80d00000 r6:01c70270 r5:80d01dc4 r4:b77fa440
[298495.238050] [<8019c568>] (run_timer_softirq) from [<8010249c>] (__do_softirq+0x1a4/0x418)
[298495.238062]  r10:00000004 r9:00000282 r8:b709d000 r7:00000100 r6:80d00000 r5:00000001
[298495.238066]  r4:80d03084
[298495.238080] [<801022f8>] (__do_softirq) from [<80125be4>] (irq_exit+0x100/0x110)
[298495.238091]  r10:b7fff840 r9:80d00000 r8:b709d000 r7:00000001 r6:00000000 r5:00000000
[298495.238096]  r4:80ca22a4
[298495.238111] [<80125ae4>] (irq_exit) from [<80180ad4>] (__handle_domain_irq+0x70/0xc4)
[298495.238124] [<80180a64>] (__handle_domain_irq) from [<80102228>] (bcm2836_arm_irqchip_handle_irq+0x60/0xa4)
[298495.238135]  r8:80d9943c r7:80d01f0c r6:ffffffff r5:60000013 r4:00000000 r3:80d01ed8
[298495.238147] [<801021c8>] (bcm2836_arm_irqchip_handle_irq) from [<80101a3c>] (__irq_svc+0x5c/0x7c)
[298495.238153] Exception stack(0x80d01ed8 to 0x80d01f20)
[298495.238160] 1ec0:                                                       80109b94 00000000
[298495.238172] 1ee0: 60000093 8011a2e0 80d04f68 80d00000 80d04fb0 00000001 80d9943c 80c68a40
[298495.238183] 1f00: b7fff840 80d01f34 80d0535c 80d01f28 00000000 80109b98 60000013 ffffffff
[298495.238189]  r4:80109b98 r3:dd7dfe8a
[298495.238205] [<80109b64>] (arch_cpu_idle) from [<808d2c84>] (default_idle_call+0x34/0x48)
[298495.238217] [<808d2c50>] (default_idle_call) from [<80153494>] (do_idle+0xec/0x170)
[298495.238227] [<801533a8>] (do_idle) from [<801537f4>] (cpu_startup_entry+0x28/0x2c)
[298495.238239]  r8:00000000 r7:80c68a40 r6:80daa580 r5:80daa5d8 r4:000000ce r3:80d01f50
[298495.238252] [<801537cc>] (cpu_startup_entry) from [<808cc5a8>] (rest_init+0xbc/0xc0)
[298495.238268] [<808cc4ec>] (rest_init) from [<80c00bd8>] (arch_call_rest_init+0x18/0x1c)
[298495.238274]  r4:80d04f48 r3:80c97260
[298495.238286] [<80c00bc0>] (arch_call_rest_init) from [<80c01090>] (start_kernel+0x4b4/0x4e8)
[298495.238297] [<80c00bdc>] (start_kernel) from [<00000000>] (0x0)
[298495.238304] ---[ end trace d68254835429d4e0 ]---

```

pelwell, 2020-08-05:
Unfortunately "transmit queue <n> timed out" is a catch-all error indicating that transmission is no longer proceeding as expected.

* What was the Pi doing at the point of failure?
* How active would Ethernet have been?
* Do you have any custom settings in your config.txt?

cfilipem, 2020-08-05:
> Unfortunately "transmit queue timed out" is a catch-all error indicating that transmission is no longer proceeding as expected.
> 
> * What was the Pi doing at the point of failure?

This is a remote Pi running Homebridge, Telegraf for some SNMP and ICMP network monitoring and has a WireGuard tunnel.

> * How active would Ethernet have been?

At the time of the crash ethernet had about 70kbit/s peak.

> * Do you have any custom settings in your config.txt?

Only these settings:

```
dtoverlay=disable-wifi
dtoverlay=disable-bt
gpu_mem=64
```

pelwell, 2020-08-05:
You could try updating to the very latest kernel and firmware (`sudo rpi-update`), but I'm not aware of any specific changes that would affect your configuration.

Unfortunately any bug with such a generic failure mode is hard to pinpoint even if very reproducible, and bugs with specific failure modes that may take days to reproduce are probably worse, but the combination of the two makes this issue almost impossible to debug.

cfilipem, 2020-08-05:
For now I will stick with version 1.20200601-1 on this system, which has been working fine on this system since June 11.
This Pi is in a remote location with access only via ethernet, so It's risky to play with kernel and firmware. On both crashes I had to remotely power off/on the plug to recover access.

cfilipem, 2021-02-25:
This issue seems to be solved in firmware 1.20210201-1 with this kernel change: https://github.com/raspberrypi/linux/commit/981e1807748af57283a11d566b787aed6107dae9

closed 2021-03-01 by cfilipem