raspberrypi/firmware #1446: Pi GPU always powers up in 60hz mode, ignoring flags to run at 59.94hz until applied a 2nd time

alanbork, 2020-08-05:
**Describe the bug**

The Pi GPU always powers up in 60hz mode, ignoring flags to run at 59.94hz (the NTSC refresh rate). This is true whether the pi is configured to use NTSC via config.txt, or if the display is powered down via tvservice and then powered back on requesting the NTSC refresh rate.  Neither KMS or FKMS is loaded, so I guess that means this is a bug in the firmware?

note: if you change the video mode again once the GPU is already powered up you get the proper refresh rate (60hz or 59.94hz). It's only when you go from GPU power off to power on that there is a failure. While this sounds like a minor issue it has two unfortunate consequences; some TVs totally fail when given 60hz instead of 59.94; for those users it's impossible to see any boot messages. Second, I've observed that switching between 60hz and 59.94hz can cause errors in the GPU's measure of when vsync happens, as reported via vc_dispmanx_vsync_callback. I've observed this with two different TVs that have problems responding to 60hz signals and expect 59.94hz. While that again sounds like an edge case it in fact represents a full half of the tvs I've tested (all older models, however). See the forum link under Additional context.

**To reproduce**
method 1: boot time

Recently it was discovered that you could request the NTSC refresh rate using boot/config.txt with the (still undocumented?) flag:

hdmi_pixel_clock_type=1

Unfortunately the pi still boots the GPU up at 60hz. If you run tvservice -s it prints out that the refresh rate is 59.94hz, but the GPU outputs 60hz until you set the mode again using tvservice -t -e [...];

method 2: tvservice

after booting you can power down the display output and then bring it back up again requesting NTSC refresh rate:

tvservice -0
tvservice -t -e="CEA 4"
[...additional call here to fbset required to get a working console]

tvserice -s now reports NTSC refresh rate, but the true output is 60hz. 

**Expected behaviour**
In both the bootime and post-boot tvservice based methods the display should power up at 59.94hz. Ideally it should do so without first initializing in 60hz mode, since there's some evidence that switching to 59.94hz from 60hz can cause problems keeping track of the true timestamp for vsync (see additional context below).

**Actual behaviour**
root@raspberrypi:~# tvservice -s
state 0xa [HDMI CEA (4) RGB lim 16:9], 1280x720 @ 59.94Hz, progressive

actual gpu refresh rate: 60hz. 

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW

PiZero

* Which OS and version (`cat /etc/rpi-issue`)?

I've upgrade to buster, but this is the contents of /etc/rpi-issue

Raspberry Pi reference 2018-04-18
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, d6c238c1b2b1b070a574d3e25048ca442e3e221f, stage2

* Which firmware version (`vcgencmd version`)?

root@raspberrypi:~# vcgencmd version
Jul 17 2020 11:04:27
Copyright (c) 2012 Broadcom
version 21a15cb094f41c7506ad65d2cb9b29c550693057 (clean) (release) (start)

* Which kernel version (`uname -a`)?

root@raspberrypi:~# uname -a
Linux raspberrypi 5.4.51+ #1327 Thu Jul 23 10:53:06 BST 2020 armv6l GNU/Linux

**Logs**
not sure this is relevant..
Jul 31 22:07:55 raspberrypi kernel: [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Jul 31 22:07:55 raspberrypi kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Jul 31 22:07:55 raspberrypi kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero Rev 1.3
Jul 31 22:07:55 raspberrypi kernel: [    0.000000] Kernel command line: bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=720 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 Jul 31 22:07:55 raspberrypi kernel: [    0.140822] raspberrypi-firmware soc:firmware: Attached to firmware from 2018-04-16 18:16

**Additional context**
forum post discussing this issue and a related problem with drift between true vsync and reported vsync:

https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=281548&p=1708231#p1708231

6by9, 2020-08-06:
@popcornmix Can you confirm this with the HDMI analyser please?
I had a look through the code and it seems to set the PIXEL_CLOCK_TYPE property in platform_config_init before setting the the mode, so I would have expected it to work.

popcornmix, 2020-08-06:
hdmi_pixel_clock_type=0
```
Pixel clock: 74.251MHz
H freq: 45.00kHZ 
V freq: 60.00Hz
```
hdmi_pixel_clock_type=1
```
Pixel clock: 74.251MHz
H freq: 45.00kHZ 
V freq: 60.00Hz
```
```
$ vcgencmd get_config int |grep hdmi_pixel_clock_type
hdmi_pixel_clock_type:0=1
```

$ tvservice -t -e "CEA 4"
Powering on HDMI with explicit settings (CEA mode 4)
```
Pixel clock: 74.176MHz
H freq: 44.96kHZ 
V freq: 59.94Hz
```
i@b3plus:~ $ tvservice -e "CEA 4"
Powering on HDMI with explicit settings (CEA mode 4)
```
Pixel clock: 74.251MHz
H freq: 45.00kHZ 
V freq: 60.00Hz
```

So, tvservice seems to do the right thing but hdmi_pixel_clock_type is ignored.

popcornmix, 2020-08-06:
You don't need analyser to test this:
```
pi@b3plus:~ $ tvservice -e "CEA 4"
Powering on HDMI with explicit settings (CEA mode 4)
pi@b3plus:~ $ vcgencmd measure_clock pixel
frequency(29)=74250000
pi@b3plus:~ $ tvservice -t -e "CEA 4"
Powering on HDMI with explicit settings (CEA mode 4)
pi@b3plus:~ $ vcgencmd measure_clock pixel
frequency(29)=74176000
```

6by9, 2020-08-06:
> You don't need analyser to test this:

Thanks - the a useful trick. I've got the debugger up to look at other things already, so I'll have a poke around to see why ```hdmi_pixel_clock_type=1``` gets reset.

6by9, 2020-08-06:
Hmm, something funny on a Pi4 using that technique
```
pi@raspberrypi:~ $ tvservice -s
state 0xa [HDMI DMT (68) RGB full 16:10], 1920x1200 @ 60.00Hz, progressive
pi@raspberrypi:~ $ vcgencmd measure_clock pixel
frequency(29)=149989744
```
```
pi@raspberrypi:~ $ tvservice -s
state 0xa [HDMI DMT (68) RGB full 16:10], 1920x1200 @ 59.94Hz, progressive
pi@raspberrypi:~ $ vcgencmd measure_clock pixel
frequency(29)=150002928
```
1920x1200@60 (DMT mode 68) should be 154MHz normally.

Time to check a Pi3 (should be identical to a Pi0).

6by9, 2020-08-06:
OK, Pi3 behaves sensibly.

popcornmix, 2020-08-06:
```
pi@b3plus:~ $ tvservice -e "DMT 68"
Powering on HDMI with explicit settings (DMT mode 68)
pi@b3plus:~ $ tvservice -s
state 0xa [HDMI DMT (68) RGB full 16:10], 1920x1200 @ 60.00Hz, progressive
pi@b3plus:~ $ vcgencmd measure_clock pixel
frequency(29)=154000000
pi@b3plus:~ $ tvservice -t -e "DMT 68"
Powering on HDMI with explicit settings (DMT mode 68)
pi@b3plus:~ $ tvservice -s
state 0xa [HDMI DMT (68) RGB full 16:10], 1920x1200 @ 59.94Hz, progressive
pi@b3plus:~ $ vcgencmd measure_clock pixel
frequency(29)=153846000
```
and hdmi analyser agrees.
Pi4 doesn't use pixel clock - it has it's own pll and clock manager, so I'm not quite sure what that is measuring.

popcornmix, 2020-08-06:
@timg236

timg236, 2020-08-06:
It's the HDMI/PV bus clock, you can't measure it via the normal PLL measure clock because it's not a CM clock

6by9, 2020-08-06:
Don't panic over Pi4 - you're right that it doesn't use the pixel clock.

I'm following through on Pi3 with the debugger when these properties get set.
I've a half guess that we've started in a default mode, applied the properties, then set activated the mode. Because it thinks it is the same mode, that doesn't actually get actioned.

6by9, 2020-08-06:
Potential patch uploaded.
It looks like we take in the desired mode, alter it for NTSC timings, but then pass the original mode to HDMI driver to program into the hardware.

I now get the expected clock reported via ```vcgencmd measure_clock pixel``` based on ```hdmi_pixel_clock_type=1``` and ```tvservice -t -e...```.

6by9, 2020-08-06:
Change merged internally, so it should be in the next rpi-update.

popcornmix, 2020-08-06:
The fix should be in latest rpi-update firmware.

alanbork, 2020-08-06:
> The fix should be in latest rpi-update firmware.

So if I run rpi-update on my pi0 now it will get the new firmware? I'm looking forward to testing this and super happy that this was taken care of so quickly!

popcornmix, 2020-08-06:
Yep.

alanbork, 2020-08-06:
downloaded, tested, working! thanks everybody. 

PS this didn't solve my drifting vsync issue, but because I had this firmware to test with I was able to eliminate the possibility that the pi was at fault. As unlikely as it seems, I have two TVs from different manufacturers which refresh their display at a random/drifting offset relative to what the PI thinks they do. I can only conclude that these TVs run off an internal frame buffer/sync signal and make absolutely no attempt to synchronize their "vertical refresh period" with the external input.

closed 2020-08-06 by alanbork