raspberrypi/firmware #1451: Raspberry Pi 4 missing firmware mitigation for Spectre v4

hotaru2k3, 2020-08-11:
**Describe the bug**
the Raspberry Pi 4's Cortex-A72 cores are vulnerable to Spectre v4 (Speculative Store Bypass, CVE-2018-3639). [according to ARM](https://developer.arm.com/support/arm-security-updates/speculative-processor-vulnerability), there's a firmware mitigation available for this vulnerability, but the mitigation seems to not be present on the Raspberry Pi 4.

**To reproduce**
1. download and build https://github.com/google/safeside
2. run the `spectre_v4` demo

**Expected behaviour**
```
Leaking the string: Does not converge
```

**Actual behaviour**
```
Leaking the string: It's a s3kr3t!!!
Done!
```

**System**
 * Which model of Raspberry Pi?
Pi 4
* Which OS and version (`cat /etc/rpi-issue`)?
Arch Linux ARM aarch64
* Which firmware version (`vcgencmd version`)?
```
Jul 13 2020 13:56:29
Copyright (c) 2012 Broadcom
version adcebbdb7b415c623931e80795ba3bae68dcc4fa (clean) (release) (start_x)
```
* Which kernel version (`uname -a`)?
```
Linux marten 5.8.0-1-ARCH #1 SMP Sun Aug 9 00:03:44 UTC 2020 aarch64 GNU/Linux
```

6by9, 2020-08-11:
The linked page appears to say
```
For Cortex-A57 and Cortex-A72:

    Set bit 55 (disable load pass store) of CPUACTLR_EL1 (S3_1_C15_C2_0).
```
so I guess that needs to go into the ARM stub.

The issue is likely to be that this has a significant performance hit for all those who don't care about Spectre, so you now have to gain another version of the stub.

hotaru2k3, 2020-08-11:
> The issue is likely to be that this has a significant performance hit for all those who don't care about Spectre, so you now have to gain another version of the stub.

maybe there could be a config.txt option to enable or disable it?

6by9, 2020-08-11:
> maybe there could be a config.txt option to enable or disable it?

Yes, things are possible, it just becomes further maintenance overheads.

You can already override the built-in stubs by adding an appropriate file (probably ```armstub8-gic.bin``` in your case) to /boot. That's why I was looking for the stub source files - now found https://github.com/raspberrypi/tools/tree/master/armstubs

hotaru2k3, 2020-08-11:
I built armstub8-gic.bin with the following changes:

```
diff --git a/armstubs/armstub8.S b/armstubs/armstub8.S
index f85eb521..188c0d55 100644
--- a/armstubs/armstub8.S
+++ b/armstubs/armstub8.S
@@ -65,6 +65,9 @@
 #define SCR_VAL \
     (SCR_RW | SCR_HCE | SCR_SMD | SCR_RES1_5 | SCR_RES1_4 | SCR_NS)
 
+#define CPUACTLR_EL1		S3_1_C15_C2_0
+#define CPUACTLR_EL1_DLPS	BIT(55)
+
 #define CPUECTLR_EL1		S3_1_C15_C2_1
 #define CPUECTLR_EL1_SMPEN	BIT(6)
 
@@ -124,6 +127,10 @@ _start:
 	mov x0, #CPUECTLR_EL1_SMPEN
 	msr CPUECTLR_EL1, x0
 
+	/* mitigate Spectre v4 */
+	mov x0, #CPUACTLR_EL1_DLPS
+	msr CPUACTLR_EL1, x0
+
 #ifdef GIC
         bl      setup_gic
 #endif
```

and after booting with that armstub, the `spectre_v4` demo does gives the desired result ("Does not converge").

6by9, 2020-08-11:
Thanks. I didn't know your knowledge level, but provided the links in case you could make use of them.

Feel free to create a PR for the stub. Ideally wrapped in a ```#ifdef``` so that it can be built only when desired. We always prefer to provide the correct attribution for contributions to the system.

Do you have any numbers on the impact that this mitigation has? I know some of them on x86 were HUGE, but I haven't been following this one.

hotaru2k3, 2020-08-11:
PR created: https://github.com/raspberrypi/tools/pull/115

> Do you have any numbers on the impact that this mitigation has?

so far all the testing I've done shows no impact other than Spectre v4 no longer working. do you have any recommendations of benchmarks that would be likely to show a difference?

alanbork, 2020-09-16:
personally I'd want ZERO impact before enabling this by default. How many of us are using our Pi's in a multi-user environment?

hotaru2k3, 2020-09-17:
the PR wouldn't enable the mitigation by default (it's wrapped in a `#ifdef`). enabling it by default would definitely require more testing to determine what, if any, performance impact it has.

andrum99, 2020-09-17:
> the PR wouldn't enable the mitigation by default (it's wrapped in a `#ifdef`)

What PR? If you mean the code above, it's not wrapped in an ifdef - the one ifdef shown sets up the GIC, if GIC mode is enabled.

hotaru2k3, 2020-09-17:
https://github.com/raspberrypi/tools/pull/115, which is linked three comments above yours.

AaronDewes, 2020-09-30:
According to this: https://github.com/speed47/spectre-meltdown-checker, the Raspberry Pi 4 is also vulnerable to CVE-2018-3640.

hotaru2k3, 2020-09-30:
it is, but ARM says:

> This side-channel can be used to determine the values held in system registers that should not be accessible. While it is undesirable for lower exception levels to be able to access these data values, for the majority of system registers, the leakage of this information is not material.

and

> Although Arm is addressing this issue in new CPU releases, in general, it is not believed that software mitigations for this issue are necessary.
> For system registers that are not in use when working at a particular exception level and which are felt to be sensitive, it would in principle be possible for the software of a higher exception level to substitute in dummy values into the system registers while running at that exception level. In particular, this mechanism could be used in conjunction with the mitigation for Variant 3 to ensure that the location of the VBAR_EL1 while running at EL0 is not indicative of the virtual address layout of the EL1 memory, so preventing the leakage of information useful for compromising KASLR.

mitigating CVE-2018-3640 would probably require changes to the Linux kernel, rather than firmware, and probably isn't worth the trouble.

ell1e, 2021-09-16:
Any updates on this? People are using the Raspberry increasingly for normal desktop tasks, so this seems like it should be fixed (with the secure variant being the default). Does the Linux kernel contain the necessary mitigations as well?

alanbork, 2021-09-16:
idk, pi4 is pretty slow as a desktop os, I'd certainly prefer it not the default, or at the least very easy to switch.

pelwell, 2021-09-16:
Spectre mitigation is enabled in the kernel, and https://github.com/raspberrypi/tools/pull/117 was merged last October. Do you have some reason to believe this is not sufficient?

JamesH65, 2021-09-16:
Because the issue is still open?

ell1e, 2021-09-16:
> the issue is still open

That, and the referenced https://github.com/raspberrypi/tools/pull/115 hasn't been merged. Was it superseded by https://github.com/raspberrypi/tools/pull/117 then?

pelwell, 2021-09-16:
Hmm... the PR for https://github.com/raspberrypi/tools/pull/117 suggests that it _could_ supersede https://github.com/raspberrypi/tools/pull/115, but although we've enabled the standard kernel mitigation I don't see anything that actually does set the "disable load pass store" bit of CPUACTLR_EL1.

Since the rewrite, https://github.com/raspberrypi/tools/pull/115 no longer applies and will need to be updated, but that will still leave the mitigation as optional (enabled either by an external stub file or by an as-yet-non-existent firmware config.txt setting).

alanbork, 2021-09-16:
and (perhaps) more importantly, if it's in there by default, how do we disable the mitigations?

pelwell, 2021-09-16:
If _what_ is in there by default?:
> but that will still leave the mitigation as optional (enabled either by an external stub file or by an as-yet-non-existent firmware config.txt setting).

alanbork, 2021-09-16:
the cpu-slowing mitigation. we posted at the same time, so now that I've read your post it sounds like no spectre mitigations are enabled by default, which is good.

ell1e, 2021-09-16:
> enabled either by an external stub file or by an as-yet-non-existent firmware config.txt setting

Wouldn't it make the most sense if it was somehow set through the kernel mitigations option as well? Or is that not feasible? Any other way is likely to be missed by someone: if it defaults to on people who use the raspberry without untrusted code will get unexpected slowness and if it defaults to off people will set the kernel mitigations option and think they're safe. Neither of these outcomes seems ideal.

alanbork, 2021-09-16:
Not sure what you are suggesting, but config.txt has the best visibility I
know of.

ell1e, 2021-09-16:
The mitigations kernel option is for sure more known under the general Linux population. Since it needs to be set to the desired value anyway, having a separate config.txt option that unexpectedly also controls a part of this has IMHO for sure less visibility.

pelwell, 2021-09-16:
If the kernel mitigation is enabled on the command line then detecting it and setting the magic bit accordingly is feasible,

ell1e, 2021-09-16:
If that would reliably work with major distros and partitioning setups (like with LUKS/unlocking initramf), then that sounds like the best variant to me personally, linking it to the kernel mitigations. But maybe someone else has input on this as well.

alanbork, 2021-09-16:
The whole breakdown between cmdline.txt and config.txt is confusing as
heck, you never know where to look for your options. But at least
config.txt allows comments so you can list what options are available and
briefly describe their function. cmdline.txt is hugely less user friendly.

ell1e, 2021-09-16:
Via [rpi4-uefi](https://rpi4-uefi.dev) + the generic Fedora ARM64 image I think the kernel options usually reside in `/etc/default/grub` which I think might support comments (don't quote me on that), so maybe that would be interesting to you. I am wondering now though if whatever detection @pelwell had in mind would still work when using rp4-uefi.

pelwell, 2021-09-17:
No - it wouldn't work with UEFI, grub or any other ways of passing in a kernel command line, which makes me think the config.txt setting to set the "disable load pass store" bit (and only do that) is the most general solution.

ell1e, 2021-09-17:
Yes, that might be correct then. My suggestion would still be to enable it by default, but you'll probably get tons of voices (as above) disagreeing. But IMHO having an unexpected discovery you ran slower than desired is better than having an unexpected discovery you thought all mitigations were on when they were not.

alanbork, 2021-09-17:
I'm not too worked up about the default setting as long as it's well
documented as to how to change it (and when it comes to pi documentation,
"well documented" rarely applies), and there's a status message printed
telling you that it's enabled/disabled during boot. Of course bootup is
full of so many status messages most of which are so completely irrelevant
to anything a user would care about that an actually important status
message would be easy to miss.

ell1e, 2025-12-03:
Any mitigation being added should likely be enabled by default. Most users not familiar with CPU vulnerabilities won't understand that they can often be exploited by web pages. (Sorry if I'm missing something why that wouldn't be a concern here.)

atomic-kernel, 2025-12-06:
> Note, on a Pi5, lscpu reports:
> 
> ```
> Vulnerability Spectre v1:                Mitigation; __user pointer sanitization
> Vulnerability Spectre v2:                Mitigation; CSV2, BHB
> ```

This seems to be because raspberry pi 5 is not affected by spectre v2 vulnerabilities, not because the firmware supports spectre v2 mitigation. According to kernel code:
https://elixir.bootlin.com/linux/v6.18/source/arch/arm64/kernel/proton-pack.c#L135

If spectre v2 vulnerability and firmware mitigation both exist, "Branch predictor hardening" should be printed.

atomic-kernel, 2025-12-06:
> Hi everyone, I found that the Raspberry Pi 4b's firmware is not only missing spectre v4 mitigation but also missing spectre v2 mitigation. Based on the kernel source code, the Cortex-A72 architecture is likely affected by the Spectre v2 vulnerability. https://elixir.bootlin.com/linux/v6.18/source/arch/arm64/kernel/proton-pack.c#L152
> 
> I expected the Linux kernel will enable Spectre v2 mitigation, but it doesn't. Based on my debugging, it returns at this point: https://elixir.bootlin.com/linux/v6.18/source/arch/arm64/kernel/proton-pack.c#L273 because `arm_smccc_1_1_invoke(ARM_SMCCC_ARCH_FEATURES_FUNC_ID, ARM_SMCCC_ARCH_WORKAROUND_1, &res);` returns SMCCC_RET_NOT_SUPPORTED from here: https://elixir.bootlin.com/linux/v6.18/source/arch/arm64/kernel/proton-pack.c#L181
> 
> I urgently need firmware that supports Spectre v2 mitigation measures to do some testing. Is there any way to enable it? Thanks very much

I discovered that the Raspberry Pi 4B UEFI firmware supports Spectre v2 mitigation. After using [this firmware](https://github.com/pftf/RPi4), [spectre_v2_enable_fw_mitigation](https://elixir.bootlin.com/linux/v6.18/source/arch/arm64/kernel/proton-pack.c#L266)() seems to successfully install [call_smc_arch_workaround_1](https://elixir.bootlin.com/linux/v6.18/source/arch/arm64/kernel/proton-pack.c#L232)(), the branch predictor hardening callback function.

However, after enabling the firmware mitigation, I did not observe the expected performance degradation. It seems there's something wrong with how this mitigation implemented? I'm not sure, and I'm completely unfamiliar with the firmware.

JamesH65, 2026-02-24:
Here's some data from ARM. 

Variant 4

**The risk of Spectre v4 is now considered to be lower than was initially thought. Arm recommends that developers consider whether mitigations are necessary before implementing them**.

Mitigating this issue can be achieved by disabling memory disambiguation at boot by setting the relevant control register bit. The specific bit varies by processor micro-architecture.

For Cortex-A57 and Cortex-A72:

Set bit 55 (disable load pass store) of CPUACTLR_EL1 (S3_1_C15_C2_0).

For Cortex-A73:

Set bit 3 of S3_0_C15_C0_0.

For Cortex-A75:

Set bit 35 of CPUACTLR_EL1 (S3_0_C15_C1_0)

For Cortex-A76:

Set bit 16 of CPUACTLR2_EL1 (S3_0_C15_C1_1)

Alternatively, a dynamic approach may be used to enable the mitigation during EL3 initialization and disable it at lower exception levels, when the risk is deemed low enough. SMCCC_ARCH_WORKAROUND_2 has been created to allow callers at lower exception levels to temporarily disable the mitigation in their execution context. For more information on this approach, please refer to the [SMC Calling Convention specification](https://developer.arm.com/documentation/den0028/latest).

If the system uses Trusted Firmware-A, developers should also refer to the Trusted Firmware Advisory [TFV-7](https://trustedfirmware-a.readthedocs.io/en/latest/security_advisories/security-advisory-tfv-7.html).

ell1e, 2026-02-24:
Do you have any source why it needs to be more on device than JavaScript? Even if this is true, it may render e.g. docker, flatpak, and many other sandboxes useless. Don't people commonly run cloud services on Raspberry Pi 5s?

> missing if it was detected as AI spam

Perhaps your spam filter needs some revision then.

ell1e, 2026-02-24:
I know a ton of hobbyists who run containers on their Raspberry PIs, or use Raspberry Pi 5 as a desktop for browsing.

ell1e, 2026-02-25:
It was already mentioned above that the Raspberry Pi 5 is apparently also affected. Do you want me to file a different issue?

ell1e, 2026-03-29:
@JamesH65 any update on this, given that apparently there isn't much of a slowdown? The mitigation does seem to be ready to ship here: https://github.com/raspberrypi/tools/pull/115 My apologies for taking up your time.