raspberrypi/firmware #1452: Rpi 3B + Kingston DTR3.0 G2 = No USB Boot

himuura, 2020-08-11:
**Describe the bug**
Well, i simply cannot boot any OS from a Kingston DTR3.0 G2 16/32 GB usb drive. I activated the OTP bit, it went fine, flashed the Raspbian OS onto the Kingston drive, removed de uSD and attached the USB. It blinks a couple of times but no image, no rainbow screen, nothing.

**To reproduce**
Well, you just need to have a Kingston DTR3.0 G2 drive...

**Expected behaviour**
It should boot without the uSD inserted...Kingston is a major OEM of usb drivers, uSD and SSD AFAIK...

**Actual behaviour**
Nothing...just a black screen, no image, simply nothing. Just a couple of led flashes from the USB drive

**System**

* Which model of Raspberry Pi? 
Pi3B
* Which OS and version (`cat /etc/rpi-issue`)?
Latest Raspbian OS (but a bunch of other OS as well)
* Which firmware version (`vcgencmd version`)?
Latest i guess...
* Which kernel version (`uname -a`)?
Latest.

**Logs**
Nothing to show here, move along...

**Additional context**
Pi3B should be able to boot without uSD. Haven't tried an HDD though....

timg236, 2020-08-11:
You could try putting the latest bootcode.bin on a sdcard. On Pi3 USB boot is in the ROM and obviously can’t be changed but this might be a good compromise. If you happen to have a Pi4 please let us know if it works on that.

himuura, 2020-08-12:
Well, tried that with pi2 v1.1 worked pretty well (just as a side note)!! Im actually surprised by the snapiness increase from the USB boot. But as for the Pi 3B, the same method should work without issues but my problems is/was that i actually needed the uSD card for other things and didn't want to buy a new one xD Any solid performance USB drives you can recommend that work out of the box (after OTP the bit) with the Pi 3 B?