raspberrypi/firmware #1453: Raspberry Pi 2 Model B Rev 1.1 fails to boot with firmware 1.20200723

srcshelton, 2020-08-12:
**Describe the bug**

With the files from tag `1.20200723` on the `/boot` partition, my RPi2B fails to boot: it enables HDMI video-out and shows the rainbow display (to show the GPU firmware has loaded correctly?), but then progresses no further.  The power LED is light and the SD activity LED flashes 7 times, pauses, then repeats.  I understand this indicates that the GPU has been unable to find a (valid?) kernel image to load.

Making *no change* at all other than reverting to the files from tag `1.20200601`, and the problem goes away - the system boots as anticipated.

(I'm currently deploying `1.20200717` to see whether this works or not)

**To reproduce**
Attempt to boot a RPi2B with firmware `1.20200723`.

**Expected behaviour**
Kernel image is loaded (from `kernel7.img`) and the boot process moves to Linux kernel-space processing.

**Actual behaviour**
The SD activity light repeats sequences of 7 flashes, and the system stays on the rainbow boot-display.

**System**

* Which model of Raspberry Pi? Pi2Bv1.1
* Which kernel version (`uname -a`)? Stock RPi kernel 5.4.51 (although no kernel image is loaded regardless)

srcshelton, 2020-08-12:
Update: Firmware `1.20200717` also fails to load the kernel.

srcshelton, 2020-08-12:
cmdline.txt:
```
ipv6.disable=0 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=noop rootwait
```

config.txt:
```
gpu_mem_512=316
gpu_mem_256=128
cma_lwm=16
cma_hwm=32
cma_offline_start=16
```

pelwell, 2020-08-13:
Firmware 1.20200723 boots fine for me on a 2B rev 1.1, even with your config.txt(*) and cmdline.txt. It's not surprising that 1.20200717 would behave the same because the firmware is unchanged between the two.

If you have a serial cable you can connect to GND + GPIO14 + GPIO15 (pins 6, 8 & 10), try adding `enable_uart=1` and `uart_2ndstage=1` to config.txt. The terminal app needs to be set to 115200 baud, no flow control.

Otherwise I suggest downloading an image to a spare SD card and upgrading that (or grab a nightly build from here: https://downloads.raspberrypi.org/nightlies/).

(*) Those cma settings are pointless now - we aren't sharing memory with the ARM in that way any more.

srcshelton, 2020-08-17:
> If you have a serial cable you can connect to GND + GPIO14 + GPIO15 (pins 6, 8 & 10), try adding `enable_uart=1` and `uart_2ndstage=1` to config.txt. The terminal app needs to be set to 115200 baud, no flow control.

Would it be technically feasible to have VideoCore *write* to the boot partition?

If so, I could raise a feature-request to suggest allowing diagnostic data to be written-out to a file on the SD card... although I wouldn't be surprised if the firmware's FAT32 filesystem code is read-only.

> (*) Those cma settings are pointless now - we aren't sharing memory with the ARM in that way any more.

I'll check to see where this particular config.txt came from... it may have been shipped by Arch at some point?

srcshelton, 2020-10-17:
I believe this is may be due to #1457 / #1465 - testing ASAP...

srcshelton, 2023-07-18:
Almost three years on, and I solved this issue: the problem is that this system's `config.txt` contained:

```
start_file=start_cd.elf
fixup_file=fixup_cd.dat
```

… which up to firmware `1.20200601` had the desired effect of loading the cut-down firmware.  From `1.20200723` onwards this instead leads to a boot failure.  It's a shame that this change wasn't publicised more widely at the time - or, indeed, that setting these values doesn't either implicitly activate `gpu_mem=16` or at least output some form of error rather than hanging.

closed 2023-07-18 by srcshelton