raspberrypi/firmware #1454: Rasbian RPi 4 kodi unable to create gui dtoverlay=vc4-fkms-v3d

Balast5, 2020-08-13:
Hello! Comrades, help. There is an RPi 4 connected to a TV (Samsung UE46ES6550) with an HDMI cable, the RPI has PIOS (Rasbian) installed.
I get a black TV screen (no signal) when I try to run RPi 4 after a clean install. after correcting the "config.txt" file, the system starts. 

config.txt:
disable_overscan=1
dtparam=audio=on
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2
hdmi_force_hotplug=1
hdmi_ignore_edid=0xa5000080
gpu_mem=256

Installed Kodi, according to the forums and instructions, you need to add the dtoverlay = vc4-fkms-v3d parameter to config.txt.
When adding a parameter and loading Rasbian, I get a black TV screen (No signal). Without this parameter, the system boots, but kodi refuses to start. (unable to create gui )
Tell me, can anyone come across this issue.

config.txt:
disable_overscan=1
dtparam=audio=on
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
hdmi_force_hotplug=1
hdmi_ignore_edid=0xa5000080
gpu_mem=256

Please, help give advice on how to get out of this situation.

popcornmix, 2020-08-13:
You added a lot of non-default options to config.txt - can you determine which are essential to get an image on screen.
Ensure display is connected and on right AV channel before powering the Pi. Do you see the rainbow splash screen?

Add the config.txt settings one at a time to see which is required to get an image.

Balast5, 2020-08-13:
> You added a lot of non-default options to config.txt - can you determine which are essential to get an image on screen.
> Ensure display is connected and on right AV channel before powering the Pi. Do [...]

I get black screen when I am loading the system with default config.txt (attached)
[config.txt](https://github.com/raspberrypi/firmware/files/5070273/01_origin_config.txt)
If I remove the following parameter from config.txt - 
dtoverlay=vc4-fkms-v3d
and then add this one -
hdmi_ignore_edid=0xa5000080
the system starts working and I can see the rainbow and desktop.
[config.txt](https://github.com/raspberrypi/firmware/files/5070291/config.txt)

popcornmix, 2020-08-18:
Can you post your edid? Without hdmi_ignore_edid in config.txt run (e.g through ssh) 
```
tvservice -d edid.dat
base64 edid.dat
```
and paste it here

mi-hol, 2020-10-28:
@Balast5 , as the issue is resolved, why don't you 'close' it?

closed 2020-10-28 by pelwell

Balast5, 2020-10-29:
Sorry