raspberrypi/firmware #1455: pull up in config.txt, but the level is 0.

xantocoder, 2020-08-14:
**Is this the right place for my bug report?**
This repository contains the GPU firmware used on the Raspberry Pi. This software is the closed source part of the Raspberry Pi system, it includes booting (including network booting and USB booting), low-level power and clock control, FKMS and legacy HDMI control (not full KMS that is in the linux kernel), hardware legacy codecs (MPEG2, H264 and VC1), encode hardware including the ISP (image sensor pipeline) and camera control, audio output (analogue and HDMI audio).

If you believe that the issue you are seeing is within this area, this is the right place. If not, we have other repositories for the linux kernel at [github.com/raspberrypi/linux](https://github.com/raspberrypi/linux) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
pull up in config.txt, but the level is 0.

**To reproduce**
I set gpio 15 as pull up in config.txt.

################## config.txt ##################
# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Enable DRM VC4 V3D driver
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
arm_64bit=1
initramfs initramfs.gz
gpio=15=ip,pu
[gpio15=0]
cmdline=recovery.txt
################## config.txt ##################

when I boot up into RPIOS x64 beta the first time, the level is 1. That's correct. 
After I connect the GPIO15 with GND and reboot, the level is 0. That's also correct.
Then I remove the wire between GPIO15 and GND and reboot, the level is 0.
This is the status using 'raspi-gpio get 15'.
         GPIO 15: level=0 fsel=0 func=INPUT pull=UP

**Expected behaviour**
After I remove the wire between GPIO15 and GND, the level is 1 because of the setting in config.txt [gpio=15=ip,pu].

**Actual behaviour**
After I remove the wire between GPIO15 and GND, the level is 0 because of the setting in config.txt [gpio=15=ip,pu].

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi?                            
Raspberry Pi 4B 1Gb

* Which OS and version (`cat /etc/rpi-issue`)?    
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 30e2dd32ba47cc3bec15ab1413c16a17e5797775, stage4

* Which firmware version (`vcgencmd version`)?  
version 369ed4e44cb5a080a2dfa7f854ae4ff46b7c9ef9 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64 GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

pelwell, 2020-08-14:
Please break the process down into more, clear steps. Use `raspi-gpio get 15` at each step to show the state. Try to contrast the behaviour before the reboot to the behaviour after. Is that different from a power off?

For example:
1. Boot from cold: `GPIO 15: level=1 fsel=0 func=INPUT pull=UP`
2. With GPIO 15 tied to GND: `GPIO 15: level=0 fsel=0 func=INPUT pull=UP`
3. With GPIO 15 _not_ tied to GND: `GPIO 15: level=1 fsel=0 func=INPUT pull=UP`
4. After `sudo reboot`: `GPIO 15: level=1 fsel=0 func=INPUT pull=UP`
5. With GPIO 15 tied to GND: `GPIO 15: level=0 fsel=0 func=INPUT pull=UP`
6. With GPIO 15 _not_ tied to GND: `GPIO 15: level=1 fsel=0 func=INPUT pull=UP`

xantocoder, 2020-08-14:
I tried again. It's really a strange behavior.
############## the first boot ##############
1. Boot from cold: GPIO 15: level=1 fsel=0 func=INPUT pull=UP
2. With GPIO 15 tied to GND: `GPIO 15: level=0 fsel=0 func=INPUT pull=UP`
3. With GPIO 15 _not_ tied to GND: `GPIO 15: level=0 fsel=0 func=INPUT pull=UP`
############## the second boot ##############
4. After `sudo reboot`(not tied): `GPIO 15: level=1 fsel=0 func=INPUT pull=UP`
5. With GPIO 15 tied to GND:  `GPIO 15: level=0 fsel=0 func=INPUT pull=UP`
############## the third boot ##############
6. After `sudo reboot` (With GPIO 15 tied to GND): `GPIO 15: level=1 fsel=0 func=INPUT pull=UP`
7. With GPIO 15 _not_ tied to GND:  `GPIO 15: level=0 fsel=0 func=INPUT pull=UP`
############## the fourth boot ##############
8. After `sudo reboot`(not tied): `GPIO 15: level=1 fsel=0 func=INPUT pull=UP`
############## power off and boot ##############
9. boot with not tied: `GPIO 15: level=1 fsel=0 func=INPUT pull=UP`
############## power off and boot ##############
10. boot with tied: `GPIO 15: level=1 fsel=0 func=INPUT pull=UP`
11. With GPIO 15 _not_ tied to GND: `GPIO 15: level=0 fsel=0 func=INPUT pull=UP`

very confusing behavior.

pelwell, 2020-08-14:
It is really strange.

Could you run a similar test but with a different GPIO? It's starting to look like a hardware fault.

xantocoder, 2020-08-17:
Thanks for your quick reply. I fixed this using the latest version of raspberry pi os which is built from pi-gen. 
System
Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

Which model of Raspberry Pi?
Raspberry Pi 4B 1Gb

Which OS and version (cat /etc/rpi-issue)?
Raspberry Pi reference 2020-08-16
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, c45c8e2f3217ea3e76c1f173f9cc0195cf98e11f, stage2

Which firmware version (vcgencmd version)?
Aug  6 2020 16:22:25 
Copyright (c) 2012 Broadcom
version af3edc2de473197cdfe1ff5a8ff2d34095d5b336 (clean) (release) (start)

Which kernel version (uname -a)?
Linux raspberrypi 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux

> It is really strange.
> 
> Could you run a similar test but with a different GPIO? It's starting to look like a hardware fault.

pelwell, 2020-08-17:
Does that mean you can close the issue?

xantocoder, 2020-08-18:
> Does that mean you can close the issue?

Yes, close the issue.  Thanks.

closed 2020-08-18 by xantocoder