raspberrypi/firmware #1456: Change resolutions on the fly via GPIO in Pi4

rtomasa, 2020-08-16:
Hi, I was wondering if anybody knows if using `vcgencmd hdmi_timings` is still feasible in the Pi4 when the video is output via GPIO with adapters like RGB-Pi or VGA666.
As per my tests due to the move to the new fkms driver, trying to change the timming on the fly will hang the OS (tested in Raspberry OS Lite with no X).

I've tested with RGB-Pi adapter using vanilla Raspbian OS Lite and making the proper configurations in the config.txt to make it work (set overlay, dpi mode, etc.). Then tested to manually execute `vcgencmd hdmi_timings ...` and not working no matter if you eneble or disable the new vc4 driver from the config.txt

Any light on this?

6by9, 2020-08-16:
Not currently as the DPI driver has very limited ways to pass timings in. If there were easier ways to pass the timings in then it might be worth plumbing in the DRM/KMS calls to FKMS to allow adding/changing modes.

With the full KMS driver (vc4-kms-v3d) you can use the standard Linux DRM/KMS APIs to change resolution already.

rtomasa, 2020-08-16:
AFAIK the full KMS vc4-kms-v3d driver is still in developmen and even if using xrandr for changing timings work, it would require the ful XORG stack overcomplicating the things if I do not need any UI.
Also that means that vcgencmd is now deprecated or at least many of their commands in the Pi4. The Raspberry Pi 2/3 was an amazing board for emulation mainly because of this timing feature, and now that Pi4 is much more powerfull, do not have the timing freq limitation and the interlaced mode seems to be working too, we step back to only the fixed resolutions preconfigured in the config.txt works.
I do not clearly understand why is this not feasible even when disabling the vc4-fkms-v3d which supposedly uses the custom driver like in previous Pi3.

6by9, 2020-08-17:
vc4-kms-v3d is still under development, but DPI was one of the first things that was running.

When fkms is disabled then I would also expect changes to vcgencmd hdmi_timings to work, however it's not a high priority to me to investigate. A lot of the display pipeline has changed on Pi4, and we're investing time in making it work through the standard Linux APIs rather than through the firmware.

DRM/KMS is available to any application, although one is allowed to be master at a time. You do not need X11 and xrandr - that's just one route to the relevant ioctls.
The libdrm library is another option (https://cgit.freedesktop.org/drm/libdrm/), and the included modetest app allows you to set the mode and add planes. Those settings are not persistent as the emulated framebuffer will be reinstated on exit.

rtomasa, 2020-08-17:
Very helpful your explanation. Also I can confirm that using experimental kms works when forcing DPI via custom dtbo, and disabling the driver kms/fkms in an updated Raspbian is now also workig (I’d tested a few weeks ago and fbset was still crashing but now works 😊). Thanks a lot

rtomasa, 2020-08-18:
After testing again I did realize that actualy only the framefuffer resolution was changed but the real timing was unaltered :-(

I've tested in a fully updated Raspian OS Lite in a RPi4. The test is simple, execute the two following instructions for trying to set a super res from CLI:
`vcgencmd hdmi_timings 1920 1 48 192 240 240 1 9 5 9 0 0 0 60.0 0 37872000 1`
`fbset -depth 8 && fbset -depth 24 && fbset -depth 32`

I've executed this test 3 times trying one by one all the different available drivers:
`dtoverlay=vc4-fkms-v3d` --> this completly hang the system when fbset is executed
`dtoverlay=vc4-kms-kippah-7inch` --> uses DPI via KMS which is required for GPIO video. Using this driver there is no timing change but fbset works
Using the fallback driver commenting all the above --> no timing change either but fbset works too

@6by9 Even when you stated that this is not a high priority issue, is there any posibility to see this working back again in the future?

closed 2020-08-18 by rtomasa