raspberrypi/firmware #1457: Can't boot release 1.20200717 to 1.20200811 with a hybrid MBR + GPT from SD on rpi4

iosabi, 2020-08-16:
**Describe the bug**
I was using a Hybrid MBR + GPT setup where the rpi4 boot would only care about the MBR piboot partition in the sdcard (first entry of the MBR, formatted as a fat32) but then the linux kernel would use the GPT partition table thanks to a *second* entry in the MBR signaling that there is a GPT table. This setup was working fine on rpi3 and rpi4 with 32-bit and 64-bit.

The following tests are performed with the EEPROM at 2020-04-16 with the UART debug output enabled using a recent third party linux kernel (5.4.40). (`RPi: BOOTLOADER release VERSION:a5e1b95f DATE: Apr 16 2020`).

Firmware releases 1.20200717 through 1.20200811 don't work anymore with this set up. The symptom is that the boot process gets stuck at:
```
Starting start4.elf @ 0xfec00200
```
and then the green LED blinks 7 times, indicating a missing kernel.

The first commit where this happens is 2b76cfc6f57d4943144b9ceb5b57d3d455d6a8fd, at the parent commit (9d4f640625529b104303a31482f5ccbd2d8da51b) everything still works. Presumably this is due to the new GPT support in start4.elf. A new release (like 1.20200811) with the old start4.elf and fixup4.dat files form the old commit (9d4f640625529b104303a31482f5ccbd2d8da51b) works fine, but using any newer one fails.

I saw issue #1420 where regular GPT was fixed in 66fc5eaac3d0af1d5a7ffa616086cbfaefd72e98, however this is not the case for the hybrid MBR+GPT case, nor I was able to boot a regular GPT with just one 0xEE entry in the MBR.

Here's the description of the MBR + GPT I'm using:

The first sector contains a MBR with two partitions: partition1 is a FAT32 boot partition, starting a 1MiB (sector 2048) and 64 MiB in size, and the second partition entry is an entry with type 0xee just to signal that there is a GPT table. In the regular GPT configuration this second partition would be the only partition entry in the MBR and would cover the whole sdcard. In my case I made it start at sector 1 but be only 2047 sectors in size so it doesn't overlap with the first entry, linux doesn't care about this size and would happily parse a valid GPT pointing outside of this MBR partition, but some tools would complain if you have two overlapping MBR partitions.

MBR portion:
```
Device          Boot Start    End Sectors    Size Id Type
sdcard.bin.mbr1 *     2048 133119  131072     64M  c W95 FAT32 (LBA)
sdcard.bin.mbr2          1   2047    2047 1023.5K ee GPT
``` 

GPT portion:
```
Device       Start     End Sectors Size Type
sdcard.bin1   2048  133119  131072  64M Microsoft LDM metadata
sdcard.bin2 133120 4194303 4061184   2G Linux filesystem
```
This example has only two partitions in the GPT which could be well used with just the MBR but in my use case I have many more. I made this more common 2 partition example trying to figure out what the problem is.

**To reproduce**

1. Create a GPT with two partitions (piboot and root) of 64 MiB and 1983 MiB in a 2049 MiB empty file image. This image will have a MBR with just one type 0xEE entry.
2. Copy valid boot and root partition contents to the image.
3. Override the first sector of the image with an MBR that contains two entries, the first entry a type 0x0c partition with the exact same location of the "boot" partition of the GPT, a second entry 0xee as described.

4. Flash this image to an sdcard with the start4.elf and fixup4.dat from 9d4f640625529b104303a31482f5ccbd2d8da51b and verify it boots all the way to linux (kernel reads root=/dev/mmcblk0p2 without problems).
5. Flash this image to an sdcard with a newer start4.elf and fixup4.dat and verify it fails.

Here is a a script to generate this scheme: https://pastebin.com/RPhE7YFj

**Expected behaviour**
start4.elf would either:
* parse the MBR and use the first partition there ignoring whether there's a 0xEE entry or a GPT at all (old behavior)
* parse the MBR, find a 0xEE entry, then parse the GPT and use the first partition there which in my case is the same 

**Actual behaviour**

I wasn't able to make start4.elf find the kernel with the newer versions, either with a regular GPT and with a Hybrid+GPT. It hangs with blinking green LED 7 times.

**System**

* Which model of Raspberry Pi?
rpi4 4GB, EEPROM at 2020-04-16 critical

* Which firmware version (`vcgencmd version`)?
1.20200601 works ok
1.20200717 to 1.20200811 don't work.

**Additional context**
I would like to update an existing deployment of rpi with this hybrid MBR+GPT that used to work. Ideally, I would like to not have to modify the MBR for this to work since it was working before. I wasn't able to boot any sdcard with GPT in the latest firmware. I would appreciate if you can point me to one GPT-based sdcard image running the latest firmware.

rtreffer, 2020-08-28:
I can confirm that the same issue exists for the Raspberry pi 3. Reverting `start.elf` (and `fixup.dat`) to an older version makes hybrid MBR/GPT layouts boot.

erik73, 2020-08-29:
I have the same problem. Confirmed om RPI3 and RPI4

erik73, 2020-09-04:
This is now solved for me. The fix for #1465, commit https://github.com/raspberrypi/firmware/commit/baec4d28b54c7f9c30aabdbc61fa0c4dedcb3e55 was the solution.

iosabi, 2020-09-04:
Indeed this was fixed by that commit. Thanks for letting me know!

closed 2020-09-04 by iosabi