raspberrypi/firmware #1458: Booting from an eMMC module with an eMMC-SD adapter fails on Raspberry Pi 2 B and Raspberry Pi B

t-linden, 2020-08-17:
A Raspberry Pi 3 B and 4 B can succesfully boot from an eMMC module with an eMMC-SD adapter. A succeful boot on a Raspberry Pi 3 B gives this output from bootcode.bin over a serial console:
"Raspberry Pi Bootcode

Found SD card, config.txt = 1, start.elf = 1, recovery.elf = 0, timeout = 0
Read File: config.txt, 1845 (bytes)

Raspberry Pi Bootcode
Read File: config.txt, 1845
Read File: start.elf, 2883204 (bytes)
Read File: fixup.dat, 6745 (bytes)
MESS:00:00:07.312064:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:07.316989:0: brfs: File read: 1845 bytes
..."

On a Raspberry Pi 2 or a Raspberry Pi B the boot process fails and the messages shown by bootcode.bin are as follows:

"Raspberry Pi Bootcode

No SD card found, try USB
Trying USB
..."
If one boots without any card inserted then no messages are obviously given. Clearly the bootcode.bin file is found on the eMMC module, because otherwise the error messages could not be given.

**To reproduce**
Put bootcode.bin on an eMMC module on an eMMC-SD adapter and try to boot.

**Expected behaviour**
Also the older Raspberry Pi models would boot like the Raspberry Pi 3 B,

**Actual behaviour**
The boot process fails as the eMMC module is not recognized by bootcode.bin.

**System**
Raspberry Pi 2 B and Raspberry Pi B both with 256 MB and 512 MB of RAM

t-linden, 2020-09-03:
If I use a 16 GB eMMC module in a Raspberry Pi 1 B then it boots fine.  But if I use a 32 GB eMMC Nano Pi eMMC module then the boot fails as described in the previous comment.

I connected a Raspberry Pi SD-uSD adapter to a 512 MB Raspberry Pi 1 B with a Kingston Mobilite G2 USB 2 SD/uSD reader with a SanDisk 8 GB SD-card inserted in the Kingston card reader. If I insert a 16 GB eMMC/uSD RasPiKey module into the Raspberry Pi SD-uSD adapter then booting from the eMMC module works fine as can be seen here:

Raspberry Pi Bootcode

Found SD card, config.txt = 1, start.elf = 1, recovery.elf = 0, timeout = 0
Read File: config.txt, 1798 (bytes)

Raspberry Pi Bootcode
Read File: config.txt, 1798
Read File: start.elf, 2877988 (bytes)
Read File: fixup.dat, 6736 (bytes)
MESS:00:00:02.078719:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.083298:0: brfs: File read: 1798 bytes
MESS:00:00:02.163416:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.167941:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not define
d
MESS:00:00:02.184531:0: gpioman: gpioman_get_pin_num: pin WL_LPO_CLK not defined
MESS:00:00:02.190236:0: gpioman: gpioman_get_pin_num: pin BT_ON not defined
MESS:00:00:02.196914:0: gpioman: gpioman_get_pin_num: pin WL_ON not defined
MESS:00:00:02.375946:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not define
d
MESS:00:00:02.381758:0: *** Restart logging
MESS:00:00:02.385644:0: brfs: File read: 1798 bytes
MESS:00:00:02.420651:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_di
splay_state instead
MESS:00:00:02.429126:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:02.433408:0: Read command line from file 'cmdline.txt':
MESS:00:00:02.439293:0: 'console=serial0,115200 console=tty1 root=PARTUUID=d9b3f
436-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash plymout
h.ignore-serial-consoles'
MESS:00:00:02.455805:0: brfs: File read: 160 bytes
MESS:00:00:02.463373:0: brfs: File read: /mfs/sd/bcm2708-rpi-b.dtb
MESS:00:00:02.467860:0: Loading 'bcm2708-rpi-b.dtb' to 0x100 size 0x5d9e
MESS:00:00:02.564334:0: brfs: File read: 23966 bytes
MESS:00:00:02.568407:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.573248:0: dtparam: audio=on
MESS:00:00:04.431517:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not define
d
MESS:00:00:04.460069:0: brfs: File read: 1798 bytes
MESS:00:00:04.909446:0: brfs: File read: /mfs/sd/kernel.img
MESS:00:00:04.913326:0: Loading 'kernel.img' to 0x8000 size 0x4cbd38
MESS:00:00:04.919410:0: Device tree loaded to 0x1bfe9e00 (size 0x6185)
MESS:00:00:04.926532:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:04.933363:0: uart: Baud rate change done...
MESS:00:00:04.936791:0: uart: Baud rate change done...
MESS:00:00:04.942168:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER n
ot defined
MESS:00:00:06.894856:0: vchiq_core: vchiq_init_state: slot_zero = 0x5b580000, is
_master = 1
MESS:00:00:06.907984:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_di
splay_state instead
MESS:00:00:06.921502:0: TV service:host side not connected, dropping notificatio
n 0x00000002, 0x00000002, 0x00000044
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.75+ (dom@buildbot) (gcc version 4.9.3 (crossto
ol-NG crosstool-ng-1.22.0-88-g8460611)) #1270 Tue Sep 24 18:38:54 BST 2019
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr
=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruc
tion cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Model B Rev 2
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x49c with
 crng_init=0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M bcm2708_fb.fbwidth=1920 bcm
2708_fb.fbheight=1200 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_
[... 706 lines cut ...]
[  OK  ] Reached target Network is Online.
         Starting /etc/rc.local Compatibility...
         Starting LSB: Start/stop apache2 web server...
         Starting LSB: Brings up/down network automatically...
[FAILED] Failed to start LSB: Start/stop apache2 web server.
See 'systemctl status apache2.service' for details.
[  OK  ] Created slice User Slice of UID 1000.
         Starting User Runtime Directory /run/user/1000...
[  OK  ] Started LSB: Brings up/down network automatically.
[  OK  ] Started User Runtime Directory /run/user/1000.
         Starting User Manager for UID 1000...
[  OK  ] Started User Manager for UID 1000.
[  OK  ] Started Session c1 of user pi.
[   53.183357] rc.local[390]: /home/pi/motion-mmal/motion: error while loading s
hared libraries: libavformat.so.53: cannot open shared object file: No such file
 or directory
[FAILED] Failed to start /etc/rc.local Compatibility.
See 'systemctl status rc-local.service' for details.
         Starting Hold until boot process finishes up...
         Starting Terminate Plymouth Boot Screen...

Raspbian GNU/Linux 10 rpicam ttyAMA0

rpicam login: pi
Password:

t-linden, 2021-01-12:
I have a few Friendly Elec 32 GB eMMC modules that will not boot on the Raspberry Pi B versions with 256 MB and 512 MB RAM. Being able to use these eMMC modules on the older Raspberry Pi B versions would be useful to have faster random I/O compared to a standard SD-card.The problem seems to be a timing issue in bootcode.bin as shown above where the eMMC module can be accessed without problem from the Linux kernel when only the bootcode.bin has been provided from a USB-device. If needed I could send one of the problematic eMMC modules for testing. 

I borrowed from an Odroid-N2 an 128 GB eMMC module and tested it as well in the Raspberry Pi:s that I have. That eMMC module did boot also on the Raspberry Pi B versions with 256 MB and 512 MB RAM.

Could you comment on this Friendly Elec eMMC boot problem @timg236 ?

timg236, 2021-01-12:
For Pi4 boot issues please can you raise an rpi-eeprom but with the log from BOOT_UART=1 to see whether the EMMC is detected at all by the Raspberry Pi. Can Linux mount it if you boot from USB?

t-linden, 2021-01-13:
The issue here is specific to the Raspberry Pi 1 A, 256 MB, Raspberry Pi 1 B rev. 2, 512 MB, Raspberry Pi 1 B rev. 2, 256 MB, Raspberry Pi 1 B rev. 1, 256 MB and is only seen with the 32 GB Friendly Elec eMMC module with a corresponging uSD-adaptor in that these modules will not boot on these older Raspberry Pi models. I have tried most of the available bootcode.bin versions, but I have not found a version that works. Shown above is the logs where I put bootcode.bin on a USB-reader and was able to access the 32 GB Frienddly Elec eMMC module from Linux on a Raspberry Pi 1 B. The Friendly Elec 32 GB eMMC modules boots as expected on the more recent models that I have: Raspberry Pi 4 B, Raspberry Pi 3 B and Raspberry Pi 2 B.

I also have a 16 GB RasPiKey UUGear eMMC module and it boots fine on all Raspberry Pi models listed previously. A 128 GB Odroid eMMC module with corresponding uSD-adaptor also boots on all the Raspberry Pi models listed above.

I don't have access to Raspberry Pi Zero models, so I don't know if they would boot with the 32 GB Friendly Elec eMMC module.

t-linden, 2021-01-14:
I forgot to mention that I have tried also creating the file "timeout" in the /boot folder, but this has not solved the problem.
  [https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/README.md](url)

t-linden, 2021-01-14:
The eMMC chip on the RasPiKey 16 GB eMMC module is a Samsung KLMAG1JETD-B041 and on the Friendly Elec eMMC the 32 GB chip is a Samsung KLMBG2JETD-B041, so they are very similar.

t-linden, 2021-01-20:
When a Raspberry Pi boots succesfully from the (u)SD-reader the activity LED will flash in an irregular fasion with a normal intensity. The activity LED only gives a very faint light when the boot hangs with a Friendly Elect 32 GB eMMC module. It could be that the LED is flashing with such a high frequence that it gives the appearance of a constant very faint light.

t-linden, 2021-02-01:
Is there any bootcode.bin version that could output more debugging information about the detection of the SD/eMMC-storage?

timg236, 2021-02-01:
Modifying BOOT_UART=1 in bootcode.bin will show all available debug

t-linden, 2021-02-01:
OK, I have tried to understand the problem with that option already but so far unsuccesfully as you can see above. Do you have suggestions for how to debug this further?