raspberrypi/firmware #1460: i2c-sensor sps30 driver does not work

lpwk, 2020-08-24:
Applying "dtoverlay=i2c-sensor,sps30' returns no data from sensor, the same in udevadm monitor returns error: * Failed to apply overlay '1_i2c-sensor' (kernel)

pelwell, 2020-08-25:
1. What do you mean by "the same in udevadm monitor"?
2. With the overlay in config.txt, what does `i2cdetect -y 1` report?
3. _Without_ the overlay in config.txt, try `sudo dtoverlay i2c-sensor sps30`.

lpwk, 2020-08-25:
1. 'The same' means put the same command in terminal, i exactly put:
`udevadm monitor -p -k -u &`
then I put:
`dtoverlay i2c-sensor sps30`
and I got:
`* Failed to apply overlay '0_i2c-sensor' (kernel)`
2. `i2cdetect -y 1` reports the same device address (0x69) with and without overlay added to config.txt.
When I add overlay for BME280 its address changes to 'UU', I expected the same behavior for SPS30.
3. `sudo dtoverlay i2c-sensor sps30` generates the same result as in point 1: `* Failed to apply overlay '0_i2c-sensor' (kernel)`
The sensor itself works, I can read data by the Python script without problem. I can run dtoverlay and read data from other sensors (BME280, ADS1115), too.
Sometimes, when run `dtoverlay i2c-sensor sps30` the terminal session hangs, the same command run in second terminal window generates error `* Failed to apply overlay '1_i2c-sensor' (kernel)` instead of `* Failed to apply overlay '0_i2c-sensor' (kernel)`

pelwell, 2020-08-25:
The first error is likely to be because you still had "dtoverlay=i2c-sensor,sps30" in config.txt.

The reason the driver doesn't load is probably that you are running a standard RPi kernel and CONFIG_SPS30 is not enabled in our defconfigs.

lpwk, 2020-08-25:
You're right, after commenting out dtoverlay error disappeared, but driver still doesn't load.
Yes, I run standard kernel - I expected if option is described in readme file it is enabled in it...

pelwell, 2020-08-25:
I've just added CONFIG_SPS30=m to our standard defconfig files, so the driver will be present in future kernel releases.

lpwk, 2020-08-25:
Thank you. I'll wait then.

closed 2020-08-25 by lpwk

pelwell, 2020-08-25:
Are you not tempted to build your own kernel to test it? It doesn't take _that_ long on a Pi 4, and the instructions are fairly good: https://www.raspberrypi.org/documentation/linux/kernel/building.md