raspberrypi/firmware #1462: [RPi3A+] System slowdown with FKMS

emrsmsrli, 2020-08-30:
I enabled the FKMS through config.txt
```
dtoverlay=vc4-fkms-v3d
gpu_mem=128
```
And after pi shows the initial boot texts, there is a long black screen. It lasts for 30 or more seconds. Than the desktop is shown but it is very slow until it is fully initialized (like icons, desktop files and such). Also, opening up applications like Firefox or Chromium is extremely slow. Compiling with GCC usually freezes the OS (i.e. compiling libfmt). Is this a known thing or can I fix it somehow?

**System**
raspinfo command:
https://paste.ofcode.org/Y7s53KmPC2xKBLTNCkqcSG

pelwell, 2020-08-30:
Why give a quarter of the RAM to the VPU? With FKMS the graphics driver allocates memory from Linux, not from the VPU memory pools. Try reducing gpu_mem to 64 (or even lower).

emrsmsrli, 2020-08-30:
Oh, didn't know that. Thanks! Black screen lasts shorter now, but the OS is still slow in general compared to legacy driver.