raspberrypi/firmware #1463: WiFi firmware brcmfmac crashes when using virtual AP interface

ingo-h, 2020-08-31:
**Describe the bug**
For a WiFi router/repeater I use a Raspberry Pi 4B and created an access point with an uplink client connection to the local internet router using a virtual interface of type `__ap` as created with:

    rpi ~$ sudo /sbin/iw dev wlan0 interface add ap@wlan0 type __ap

The router/repeater works stable with [2020-02-13-raspbian-buster-lite](https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/2020-02-13-raspbian-buster-lite.zip). After an upgrade to [2020-05-27-raspios-buster-lite-armhf](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-05-28/2020-05-27-raspios-buster-lite-armhf.zip) I noticed from time to time a crash of the **brcmfmac** firmware, either silently doesn't reply, or with error message

    Aug 28 09:04:03 raspberrypi kernel: brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted

Disabling power save does not make a difference with:

    rpi ~$ sudo iw dev wlan0 set power_save off

**To reproduce**
I have made a stress test that associated and disassociated a laptop with a static ip address to the access point in a loop for example with this script using systemd-networkd:

    laptop ~$ cat wifistress.sh
    #! /bin/bash -e
    while true; do
        systemctl start wpa_supplicant@wlan0.service
        sleep 10
        ping -W 10 -c600 -A 192.168.50.1
        sleep 1
        systemctl stop wpa_supplicant@wlan0.service
        sleep 10
    done

192.168.50.1 is the internet router of the uplink on the RasPi so routing is also tested. Calling the script with:

    laptop ~$ time sudo ./wifistress.sh

I flashed the image (see above) and run it on the RasPi and setup the testing with:

    rpi ~$ sudo -Es
    rpi ~# apt update
    rpi ~# apt install hostapd

    rpi ~# cat > /etc/hostapd/hostapd.conf <<EOF
    interface=ap@wlan0
    driver=nl80211
    ssid=RPiNet
    country_code=DE
    hw_mode=g
    channel=1
    auth_algs=1
    wpa=2
    wpa_passphrase=verySecretPassword
    wpa_key_mgmt=WPA-PSK
    wpa_pairwise=TKIP
    rsn_pairwise=CCMP
    EOF

    rpi ~# systemctl disable --now dhcpcd.service
    rpi ~# rfkill unblock wlan
    rpi ~# echo 'allowinterfaces wlan0' >> /etc/dhcpcd.conf

    rpi ~# cat > initifs.sh <<EOF
    /sbin/iw dev wlan0 interface add ap@wlan0 type __ap
    echo 1 > /proc/sys/net/ipv4/ip_forward
    iptables -t nat -A POSTROUTING -o wlan0 -j MASQUERADE
    /usr/sbin/hostapd -B /etc/hostapd/hostapd.conf
    ip addr add 192.168.4.1/24 dev ap@wlan0
    EOF

    rpi ~# chmod +x initifs.sh

    rpi ~# cat > /etc/wpa_supplicant/wpa_supplicant.conf <<EOF
    country=DE
    ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
    update_config=1

    network={
        ssid="TestNet"
        psk="verySecretPassword"
        key_mgmt=WPA-PSK
    }
    EOF

Reboot, then execute:

    rpi ~$ sudo systemctl daemon-reload
    rpi ~$ sudo ./initifs.sh
    rpi ~$ sudo systemctl start dhcpcd.service
    rpi ~$ journalctl -f &

On the laptop start the stress test.

**Expected behaviour**
The stress test from the laptop should run without a crash of the brcmfmac firmware as seen with `2020-02-13-raspbian-buster-lite`. There I stopped it after 15 hours without a crash.

**Actual behaviour**
With `2020-05-27-raspios-buster-lite-armhf` the stress test failed after typically 3 to 12 minutes. I haven't seen it running longer than 1 hour.

**System**
`2020-02-13-raspbian-buster-lite` comes with [firmware-brcm80211_20190114-1+rpt4_all.deb](http://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/).

`2020-05-27-raspios-buster-lite-armhf` comes with [firmware-brcm80211_20190114-1+rpt6_all.deb](http://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/).

A workaround to fix the crash is to downgrade to the downloaded `firmware-brcm80211_20190114-1+rpt4_all.deb`:

    rpi ~$ sudo dpkg --purge firmware-brcm80211
    rpi ~$ sudo dpkg --install firmware-brcm80211_20190114-1+rpt4_all.deb
    rpi ~$ sudo  systemctl reboot

This runs stable all the time as before. The crash is 100 % reproducible when installing the downloaded `firmware-brcm80211_20190114-1+rpt6_all` again.

**Logs**
Typical crash:  
on the laptop called with `time sudo ./wifistress.sh`

    --- snip ---
    64 bytes from 192.168.50.1: icmp_seq=598 ttl=63 time=2.05 ms
    64 bytes from 192.168.50.1: icmp_seq=599 ttl=63 time=4.35 ms
    64 bytes from 192.168.50.1: icmp_seq=600 ttl=63 time=2.05 ms

    --- 192.168.50.1 ping statistics ---
    600 packets transmitted, 600 received, 0% packet loss, time 174ms
    rtt min/avg/max/mdev = 2.023/3.099/46.393/3.916 ms, pipe 5, ipg/ewma 3.626/2.405 ms
    connect: Network is unreachable

    real    3m38.373s
    user    0m1.055s
    sys     0m2.573s
    laptop ~$

on the RasPi from the journal:

    --- snip ---
    Aug 28 09:03:16 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 IEEE 802.11: associated
    Aug 28 09:03:16 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 IEEE 802.11: associated
    Aug 28 09:03:16 raspberrypi hostapd[377]: ap@wlan0: AP-STA-CONNECTED 00:17:9a:b9:6e:a9
    Aug 28 09:03:16 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 RADIUS: starting accounting session F4BE3FBE49C7A46D
    Aug 28 09:03:16 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 WPA: pairwise key handshake completed (RSN)
    Aug 28 09:03:16 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 RADIUS: starting accounting session F4BE3FBE49C7A46D
    Aug 28 09:03:16 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 WPA: pairwise key handshake completed (RSN)
    Aug 28 09:03:26 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 IEEE 802.11: disassociated
    Aug 28 09:03:26 raspberrypi hostapd[377]: ap@wlan0: AP-STA-DISCONNECTED 00:17:9a:b9:6e:a9
    Aug 28 09:03:26 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 IEEE 802.11: disassociated
    Aug 28 09:03:26 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 IEEE 802.11: disassociated
    Aug 28 09:03:26 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 IEEE 802.11: disassociated
    Aug 28 09:03:39 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 IEEE 802.11: associated
    Aug 28 09:03:39 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 IEEE 802.11: associated
    Aug 28 09:03:39 raspberrypi hostapd[377]: ap@wlan0: AP-STA-CONNECTED 00:17:9a:b9:6e:a9
    Aug 28 09:03:39 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 RADIUS: starting accounting session 2E9827C3218501DA
    Aug 28 09:03:39 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 WPA: pairwise key handshake completed (RSN)
    Aug 28 09:03:39 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 RADIUS: starting accounting session 2E9827C3218501DA
    Aug 28 09:03:39 raspberrypi hostapd[377]: ap@wlan0: STA 00:17:9a:b9:6e:a9 WPA: pairwise key handshake completed (RSN)
    Aug 28 09:04:03 raspberrypi kernel: brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted

pelwell, 2020-08-31:
Add `brcmfmac.debug=0x100000` to cmdline.txt and repeat the test. When the firmware crashes it will report its crash state in the kernel log.

ingo-h, 2020-08-31:
@pelwell Here is it:

    Aug 31 17:52:35 raspberrypi kernel: [  126.667920] brcmfmac: CONSOLE: 000119.812 wl0: unable to find iovar "toe_ol"
    Aug 31 17:52:35 raspberrypi kernel: [  126.667926] brcmfmac: CONSOLE: 000119.812 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
    Aug 31 17:52:35 raspberrypi kernel: [  126.667930] brcmfmac: CONSOLE: 000119.812 wl0: wl_open
    Aug 31 17:52:35 raspberrypi kernel: [  126.667934] brcmfmac: CONSOLE: 000119.821 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
    Aug 31 17:52:35 raspberrypi kernel: [  126.675802] IPv6: ADDRCONF(NETDEV_UP): ap@wlan0: link is not ready
    Aug 31 17:52:35 raspberrypi kernel: [  126.777923] brcmfmac: CONSOLE: 000119.895 wl0: wl_open
    Aug 31 17:52:35 raspberrypi kernel: [  126.777929] brcmfmac: CONSOLE: 000119.904 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
    Aug 31 17:52:35 raspberrypi kernel: [  126.790485] IPv6: ADDRCONF(NETDEV_CHANGE): ap@wlan0: link becomes ready
    Aug 31 17:52:35 raspberrypi kernel: [  126.817933] brcmfmac: CONSOLE: 000119.949 wl0: link up (wl0.2)
    Aug 31 17:52:48 raspberrypi kernel: [  140.562418] 8021q: 802.1Q VLAN Support v1.8
    Aug 31 17:52:49 raspberrypi kernel: [  140.652205] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
    Aug 31 17:52:49 raspberrypi kernel: [  140.652350] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
    Aug 31 17:52:49 raspberrypi kernel: [  140.668003] brcmfmac: CONSOLE: 000133.754 wl0: unable to find iovar "toe_ol"
    Aug 31 17:52:49 raspberrypi kernel: [  140.668010] brcmfmac: CONSOLE: 000133.754 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
    Aug 31 17:52:49 raspberrypi kernel: [  140.838056] brcmfmac: CONSOLE: 000133.921 wl0: wl_open
    Aug 31 17:52:49 raspberrypi kernel: [  140.838068] brcmfmac: CONSOLE: 000133.930 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
    Aug 31 17:52:49 raspberrypi kernel: [  140.898009] brcmfmac: CONSOLE: 000133.981 wl0: link up (wl0.2)
    Aug 31 17:52:49 raspberrypi kernel: [  141.037983] brcmfmac: CONSOLE: 000134.097 wl0: unable to find iovar "nd_hostip_clear"
    Aug 31 17:52:49 raspberrypi kernel: [  141.037991] brcmfmac: CONSOLE: 000134.097 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
    Aug 31 17:52:49 raspberrypi kernel: [  141.037998] brcmfmac: CONSOLE: 000134.123 wl0: unable to find iovar "nd_hostip_clear"
    Aug 31 17:52:49 raspberrypi kernel: [  141.038004] brcmfmac: CONSOLE: 000134.123 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
    Aug 31 17:52:56 raspberrypi kernel: [  148.078040] brcmfmac: CONSOLE: 000141.123 wl0: link up (wl0)
    Aug 31 17:52:56 raspberrypi kernel: [  148.178228] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
    Aug 31 17:52:56 raspberrypi kernel: [  148.218056] brcmfmac: CONSOLE: 000141.213 wl0: link up (wl0.2)
    Aug 31 17:52:56 raspberrypi kernel: [  148.218069] brcmfmac: CONSOLE: 000141.268 wl0: unable to find iovar "nd_hostip_clear"
    Aug 31 17:52:56 raspberrypi kernel: [  148.218081] brcmfmac: CONSOLE: 000141.268 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
    Aug 31 17:52:56 raspberrypi kernel: [  148.238051] brcmfmac: CONSOLE: 000141.293 wl0.0: wlc_send_bar: seq 0x1 tid 7
    Aug 31 17:52:56 raspberrypi kernel: [  148.277932] brcmfmac: CONSOLE: 000141.333 wl0.0: wlc_send_bar: seq 0x4 tid 0
    Aug 31 20:08:09 raspberrypi kernel: [  159.630529] brcmfmac: CONSOLE: 000152.629 wl0.0: wlc_send_bar: seq 0x1 tid 6
    Aug 31 20:09:32 raspberrypi kernel: [  243.000099] brcmfmac: CONSOLE: 000235.673 
    Aug 31 20:09:32 raspberrypi kernel: [  243.000110] brcmfmac: CONSOLE: No PMKID found in RSNIE
    Aug 31 20:09:56 raspberrypi kernel: [  267.210329] brcmfmac: CONSOLE: 000259.789 
    Aug 31 20:09:56 raspberrypi kernel: [  267.210341] brcmfmac: CONSOLE: No PMKID found in RSNIE
    Aug 31 20:10:19 raspberrypi kernel: [  290.330610] brcmfmac: CONSOLE: 000282.816 
    Aug 31 20:10:19 raspberrypi kernel: [  290.330621] brcmfmac: CONSOLE: No PMKID found in RSNIE
    Aug 31 20:10:43 raspberrypi kernel: [  313.430774] brcmfmac: CONSOLE: 000305.830 
    Aug 31 20:10:43 raspberrypi kernel: [  313.430786] brcmfmac: CONSOLE: No PMKID found in RSNIE
    Aug 31 20:11:06 raspberrypi kernel: [  336.580881] brcmfmac: CONSOLE: 000328.900 
    Aug 31 20:11:06 raspberrypi kernel: [  336.580893] brcmfmac: CONSOLE: No PMKID found in RSNIE
    Aug 31 20:11:35 raspberrypi kernel: [  365.610999] brcmfmac: CONSOLE: 000357.823 
    Aug 31 20:11:35 raspberrypi kernel: [  365.611010] brcmfmac: CONSOLE: No PMKID found in RSNIE
    Aug 31 20:11:58 raspberrypi kernel: [  388.661129] brcmfmac: CONSOLE: 000380.787 
    Aug 31 20:11:58 raspberrypi kernel: [  388.661140] brcmfmac: CONSOLE: No PMKID found in RSNIE
    Aug 31 20:12:21 raspberrypi kernel: [  411.761433] brcmfmac: CONSOLE: 000403.806 
    Aug 31 20:12:21 raspberrypi kernel: [  411.761444] brcmfmac: CONSOLE: No PMKID found in RSNIE
    Aug 31 20:12:44 raspberrypi kernel: [  434.901662] brcmfmac: CONSOLE: 000426.863 
    Aug 31 20:12:44 raspberrypi kernel: [  434.901673] brcmfmac: CONSOLE: No PMKID found in RSNIE
    Aug 31 20:13:07 raspberrypi kernel: [  458.063167] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted
    Aug 31 20:13:07 raspberrypi kernel: [  458.142216] brcmfmac: CONSOLE: 
    Aug 31 20:13:07 raspberrypi kernel: [  458.142229] brcmfmac: CONSOLE: 000449.891 sp+294 001d8deb
    Aug 31 20:13:07 raspberrypi kernel: [  458.142239] brcmfmac: CONSOLE: 000449.891 sp+2bc 001bde8b

pelwell, 2020-09-01:
Is that where it ends? I was expecting something more like this:
```
Jun  3 11:34:30 pxe-server kernel: [10998.001525] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted
Jun  3 11:34:30 pxe-server kernel: [10998.073414] brcmfmac: CONSOLE: 010951.974 
Jun  3 11:34:30 pxe-server kernel: [10998.073420] brcmfmac: CONSOLE: FWID 01-72f6ece2
Jun  3 11:34:30 pxe-server kernel: [10998.073424] brcmfmac: CONSOLE: flags 1
Jun  3 11:34:30 pxe-server kernel: [10998.073428] brcmfmac: CONSOLE: 010951.974 
Jun  3 11:34:30 pxe-server kernel: [10998.073432] brcmfmac: CONSOLE: TRAP 4(25fbe0): pc 1b7622, lr 1e9461, sp 25fc38, cpsr a000019f, spsr a00001bf
Jun  3 11:34:30 pxe-server kernel: [10998.073436] brcmfmac: CONSOLE: 010951.975   dfsr 80d, dfar 28
Jun  3 11:34:30 pxe-server kernel: [10998.073441] brcmfmac: CONSOLE: 010951.975   r0 0, r1 80000000, r2 0, r3 2ab7, r4 259e00, r5 246c94, r6 241884
Jun  3 11:34:30 pxe-server kernel: [10998.073445] brcmfmac: CONSOLE: 010951.975   r7 0, r8 24188e, r9 24189c, r10 241854, r11 25fda8, r12 25fecc
Jun  3 11:34:30 pxe-server kernel: [10998.073449] brcmfmac: CONSOLE: 010951.975 
Jun  3 11:34:30 pxe-server kernel: [10998.073453] brcmfmac: CONSOLE:    sp+0 00000046 001d02b7 00000011 00000010
Jun  3 11:34:30 pxe-server kernel: [10998.073457] brcmfmac: CONSOLE: 010951.975   sp+10 0000000f 0025fda8 00255838 00000000
Jun  3 11:34:30 pxe-server kernel: [10998.073461] brcmfmac: CONSOLE: 
Jun  3 11:34:30 pxe-server kernel: [10998.073465] brcmfmac: CONSOLE: 010951.975 sp+4 001d02b7
Jun  3 11:34:30 pxe-server kernel: [10998.073469] brcmfmac: CONSOLE: 010951.975 sp+24 001d076f
Jun  3 11:34:30 pxe-server kernel: [10998.073472] brcmfmac: CONSOLE: 010951.975 sp+5c 001db89d
Jun  3 11:34:30 pxe-server kernel: [10998.073476] brcmfmac: CONSOLE: 010951.975 sp+1c4 001b0045
Jun  3 11:34:30 pxe-server kernel: [10998.073480] brcmfmac: CONSOLE: 010951.975 sp+230 00001001
Jun  3 11:34:30 pxe-server kernel: [10998.073484] brcmfmac: CONSOLE: 010951.975 sp+23c 000541ab
Jun  3 11:34:30 pxe-server kernel: [10998.073488] brcmfmac: CONSOLE: 010951.975 sp+27c 001b2933
Jun  3 11:34:30 pxe-server kernel: [10998.073492] brcmfmac: CONSOLE: 010951.975 sp+294 001d8deb
Jun  3 11:34:30 pxe-server kernel: [10998.073495] brcmfmac: CONSOLE: 010951.975 sp+2bc 001bde8b
Jun  3 11:34:30 pxe-server kernel: [10998.073499] brcmfmac: CONSOLE: 010951.975 sp+2f4 001c5a89
Jun  3 11:34:30 pxe-server kernel: [10998.073503] brcmfmac: CONSOLE: 010951.975 sp+31c 001a1d71
Jun  3 11:34:30 pxe-server kernel: [10998.073507] brcmfmac: CONSOLE: 010951.975 sp+32c 001a1dad
Jun  3 11:34:30 pxe-server kernel: [10998.073511] brcmfmac: CONSOLE: 010951.975 sp+33c 0019a589
Jun  3 11:34:30 pxe-server kernel: [10998.073515] brcmfmac: CONSOLE: 010951.975 sp+344 0019ac59
Jun  3 11:34:30 pxe-server kernel: [10998.073518] brcmfmac: CONSOLE: 010951.975 sp+354 00199cf7
Jun  3 11:34:30 pxe-server kernel: [10998.073522] brcmfmac: CONSOLE: 010951.975 sp+360 000001df
```

ingo-h, 2020-09-01:
@pelwell That is the complete log from booting the RasPi to some minutes after the crash. There was nothing more. But to be sure I will try again.

ingo-h, 2020-09-01:
@pelwell Seems I had more luck now. This is what I get with a new try:

    --- snip ---
    Sep  1 14:25:54 raspberrypi kernel: [  618.060228] brcmfmac: CONSOLE: 000609.768 
    Sep  1 14:25:54 raspberrypi kernel: [  618.060239] brcmfmac: CONSOLE: No PMKID found in RSNIE
    Sep  1 14:26:19 raspberrypi kernel: [  642.760582] brcmfmac: CONSOLE: 000634.412 
    Sep  1 14:26:19 raspberrypi kernel: [  642.760594] brcmfmac: CONSOLE: No PMKID found in RSNIE
    Sep  1 14:26:30 raspberrypi kernel: [  654.200643] brcmfmac: CONSOLE: 000643.384 coex channel 1
    Sep  1 14:26:30 raspberrypi kernel: [  654.200653] brcmfmac: CONSOLE: 000643.384 coex channel 6
    Sep  1 14:26:30 raspberrypi kernel: [  654.200661] brcmfmac: CONSOLE: 000643.384 coex channel 11
    Sep  1 14:26:44 raspberrypi kernel: [  668.307073] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted
    Sep  1 14:26:44 raspberrypi kernel: [  668.387467] brcmfmac: CONSOLE: 000659.824 wl0: wlc_userscb_alloc: no SCBs available to reclaim
    Sep  1 14:26:44 raspberrypi kernel: [  668.387481] brcmfmac: CONSOLE: 000659.824 wl0: wlc_ap_authresp: out of scbs for 00:17:9a:b9:6e:a9
    Sep  1 14:26:44 raspberrypi kernel: [  668.387490] brcmfmac: CONSOLE: 000659.824 
    Sep  1 14:26:44 raspberrypi kernel: [  668.387500] brcmfmac: CONSOLE: FWID 01-72f6ece2
    Sep  1 14:26:44 raspberrypi kernel: [  668.387509] brcmfmac: CONSOLE: flags 1
    Sep  1 14:26:44 raspberrypi kernel: [  668.387518] brcmfmac: CONSOLE: 000659.824 
    Sep  1 14:26:44 raspberrypi kernel: [  668.387529] brcmfmac: CONSOLE: TRAP 4(25fbe0): pc 1b7622, lr 1e9461, sp 25fc38, cpsr a000019f, spsr a00001bf
    Sep  1 14:26:44 raspberrypi kernel: [  668.387538] brcmfmac: CONSOLE: 000659.824   dfsr 80d, dfar 28
    Sep  1 14:26:44 raspberrypi kernel: [  668.387549] brcmfmac: CONSOLE: 000659.824   r0 0, r1 80000000, r2 0, r3 1c5, r4 259e00, r5 23405c, r6 1
    Sep  1 14:26:44 raspberrypi kernel: [  668.387559] brcmfmac: CONSOLE: 000659.824   r7 0, r8 24090e, r9 24091c, r10 0, r11 25fda8, r12 924ce
    Sep  1 14:26:44 raspberrypi kernel: [  668.387568] brcmfmac: CONSOLE: 000659.824 
    Sep  1 14:26:44 raspberrypi kernel: [  668.387578] brcmfmac: CONSOLE:    sp+0 00000000 00000002 00000001 00000000
    Sep  1 14:26:44 raspberrypi kernel: [  668.387589] brcmfmac: CONSOLE: 000659.824   sp+10 00000000 00000000 00000001 0005e3d9
    Sep  1 14:26:44 raspberrypi kernel: [  668.387597] brcmfmac: CONSOLE: 
    Sep  1 14:26:44 raspberrypi kernel: [  668.387606] brcmfmac: CONSOLE: 000659.824 sp+1c 0005e3d9
    Sep  1 14:26:44 raspberrypi kernel: [  668.387616] brcmfmac: CONSOLE: 000659.824 sp+4c 001ab1b5
    Sep  1 14:26:44 raspberrypi kernel: [  668.387625] brcmfmac: CONSOLE: 000659.824 sp+a0 00001431
    Sep  1 14:26:44 raspberrypi kernel: [  668.387634] brcmfmac: CONSOLE: 000659.824 sp+c0 00001431
    Sep  1 14:26:44 raspberrypi kernel: [  668.387644] brcmfmac: CONSOLE: 000659.824 sp+c4 00023467
    Sep  1 14:26:44 raspberrypi kernel: [  668.387653] brcmfmac: CONSOLE: 000659.825 sp+d4 00002c01
    Sep  1 14:26:44 raspberrypi kernel: [  668.387663] brcmfmac: CONSOLE: 000659.825 sp+118 00000103
    Sep  1 14:26:44 raspberrypi kernel: [  668.387672] brcmfmac: CONSOLE: 000659.825 sp+12c 00023369
    Sep  1 14:26:44 raspberrypi kernel: [  668.387681] brcmfmac: CONSOLE: 000659.825 sp+130 0002336d
    Sep  1 14:26:44 raspberrypi kernel: [  668.387691] brcmfmac: CONSOLE: 000659.825 sp+13c 000263cb
    Sep  1 14:26:44 raspberrypi kernel: [  668.387700] brcmfmac: CONSOLE: 000659.825 sp+154 0019d08f
    Sep  1 14:26:44 raspberrypi kernel: [  668.387709] brcmfmac: CONSOLE: 000659.825 sp+1c4 001b0045
    Sep  1 14:26:44 raspberrypi kernel: [  668.387719] brcmfmac: CONSOLE: 000659.825 sp+1dc 0019a9d3
    Sep  1 14:26:44 raspberrypi kernel: [  668.387728] brcmfmac: CONSOLE: 000659.825 sp+23c 0000860b
    Sep  1 14:26:44 raspberrypi kernel: [  668.387738] brcmfmac: CONSOLE: 000659.825 sp+27c 001b2933
    Sep  1 14:26:44 raspberrypi kernel: [  668.387747] brcmfmac: CONSOLE: 000659.825 sp+294 001d8deb

m-roberts, 2020-11-02:
Any update on this?

jlanza, 2020-12-14:
I don't know if this will help. But when I configured wlan0 on RPI3B as ap, I get:
```
  199.621638] WARNING: CPU: 0 PID: 313 at drivers/net/wireless/broadcom/brcm80211/brcmfmac/cfg80211.c:5225 brcmf_cfg80211_set_pmk+0x74/0x88 [brcmfmac]
[  199.621648] Modules linked in: cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc r8188eu(C) brcmfmac lib80211 brcmutil sha256_generic libsha256 raspberrypi_hwmon cfg80211 rfkill bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) snd_bcm2835(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 snd_pcm videobuf2_common snd_timer videodev snd mc vc_sm_cma(C) uio_pdrv_genirq uio fixed ip_tables x_tables ipv6
[  199.621841] CPU: 0 PID: 313 Comm: wpa_supplicant Tainted: G        WC        5.4.79-v7+ #1373
[  199.621846] Hardware name: BCM2835
[  199.621851] Backtrace: 
[  199.621873] [<8010dfa4>] (dump_backtrace) from [<8010e324>] (show_stack+0x20/0x24)
[  199.621886]  r7:ffffffff r6:00000000 r5:60000013 r4:80e97f70
[  199.621903] [<8010e304>] (show_stack) from [<80901174>] (dump_stack+0xd8/0x11c)
[  199.621920] [<8090109c>] (dump_stack) from [<80120548>] (__warn+0xe0/0x108)
[  199.621934]  r10:80e8bb40 r9:00000009 r8:7f3f27e8 r7:00001469 r6:00000009 r5:7f3f27e8
[  199.621941]  r4:7f41fc28 r3:00000000
[  199.621955] [<80120468>] (__warn) from [<80120910>] (warn_slowpath_fmt+0x70/0xc0)
[  199.621966]  r7:00001469 r6:7f41fc28 r5:80e04f88 r4:00000000
[  199.622122] [<801208a4>] (warn_slowpath_fmt) from [<7f3f27e8>] (brcmf_cfg80211_set_pmk+0x74/0x88 [brcmfmac])
[  199.622136]  r9:b2b2c1c0 r8:b2b2c000 r7:b67a4000 r6:b677f028 r5:b4ff9bdc r4:b67a4000
[  199.622540] [<7f3f2774>] (brcmf_cfg80211_set_pmk [brcmfmac]) from [<7f251e60>] (nl80211_set_pmk+0x15c/0x270 [cfg80211])
[  199.622548]  r5:b1ff6800 r4:80e04f88
[  199.622776] [<7f251d04>] (nl80211_set_pmk [cfg80211]) from [<808244d4>] (genl_rcv_msg+0x234/0x4b0)
[  199.622790]  r10:80e8bb40 r9:b1ffe3c0 r8:80e04f88 r7:00000000 r6:7f2946fc r5:b57f8414
[  199.622796]  r4:7f29d0c8
[  199.622811] [<808242a0>] (genl_rcv_msg) from [<80822d7c>] (netlink_rcv_skb+0xc8/0x120)
[  199.622825]  r10:00000000 r9:80e04f88 r8:0000004c r7:b57f8400 r6:808242a0 r5:80e04f88
[  199.622831]  r4:b1ffe3c0
[  199.622844] [<80822cb4>] (netlink_rcv_skb) from [<808233cc>] (genl_rcv+0x34/0x44)
[  199.622856]  r8:00000000 r7:b1ffe3c0 r6:0000004c r5:b1ffe3c0 r4:80e90a2c
[  199.622871] [<80823398>] (genl_rcv) from [<80822540>] (netlink_unicast+0x188/0x204)
[  199.622878]  r5:b3194400 r4:b664b400
[  199.622892] [<808223b8>] (netlink_unicast) from [<808227b4>] (netlink_sendmsg+0x1f8/0x3d4)
[  199.622906]  r9:0000004c r8:b3194400 r7:b1ffe3c0 r6:80e04f88 r5:b4ff9f44 r4:00000008
[  199.622922] [<808225bc>] (netlink_sendmsg) from [<807a24fc>] (sock_sendmsg+0x44/0x54)
[  199.622936]  r10:b4ff9de4 r9:00000000 r8:b637af80 r7:80e04f88 r6:00000000 r5:b637af80
[  199.622942]  r4:b4ff9f44
[  199.622957] [<807a24b8>] (sock_sendmsg) from [<807a2b30>] (____sys_sendmsg+0x200/0x21c)
[  199.622965]  r5:00000000 r4:b4ff9f44
[  199.622980] [<807a2930>] (____sys_sendmsg) from [<807a3320>] (___sys_sendmsg+0x7c/0xa8)
[  199.622993]  r10:00000128 r9:00000000 r8:00000000 r7:b637af80 r6:00000000 r5:b4ff9f44
[  199.622999]  r4:80e04f88
[  199.623015] [<807a32a4>] (___sys_sendmsg) from [<807a4974>] (__sys_sendmsg+0x60/0x9c)
[  199.623028]  r9:b4ff8000 r8:801011c4 r7:b637af80 r6:00000000 r5:7eea0050 r4:80e04f88
[  199.623044] [<807a4914>] (__sys_sendmsg) from [<807a49cc>] (sys_sendmsg+0x1c/0x20)
[  199.623055]  r7:00000128 r6:00b6c1b8 r5:00bb0198 r4:00b6c230
[  199.623071] [<807a49b0>] (sys_sendmsg) from [<80101000>] (ret_fast_syscall+0x0/0x28)
[  199.623078] Exception stack(0xb4ff9fa8 to 0xb4ff9ff0)
[  199.623090] 9fa0:                   00b6c230 00bb0198 00000004 7eea0050 00000000 00000000
[  199.623103] 9fc0: 00b6c230 00bb0198 00b6c1b8 00000128 76ee0000 ffffffff 00000001 00000004
[  199.623112] 9fe0: 0000006c 7eea0008 76ec55bc 76a20980
[  199.623121] ---[ end trace 4fea4fe235a57bf8 ]---
```

I don't know if this is related, but it seems so.

aaflatooni, 2020-12-27:
I am encountering the same problem even with firmware-brcm80211_20190114-1+rpt4_all.deb.
The one difference is that I have had to recompile my Kernel (needed some parameters enabled). I haven't changed any of the default settings for the Networking section of the Kernel.
Any chance there could be a parameter in the Kernel that is causing it?

RobertBerrie, 2020-12-30:
I have had the wireless adapter on my Pi4 crash for months.  I finally stumbled across this discussion:
https://github.com/raspberrypi/linux/issues/3849
where they point to a version of the broadcom driver that has been working for oaver two weeks now (which is a record for me and my Orbi network).
Driver is found on this link: 
https://drive.google.com/file/d/1YOvhSmyCmQkoF9vcsRDD5oxFkj0VGuM7/view?usp=sharing
It needs to be renamed (and backup your original as always).
Hope this is helps.

pelwell, 2020-12-31:
That version is likely to be released early in the new year.

aaflatooni, 2021-01-03:
Thanks,
Do we know exactly when it will be released? Also, are we confident that it will fix the issue?

Here is the log from the driver:
"The previous CYW43455 clm_blob provides limited access to the higher
5GHz channels (100+) - the BCM43456 (surprisingly) seems to work and
doesn't have this problem."

"surprisingly" doesn't convey confidence :-)

pelwell, 2021-01-05:
There is a release candidate firmware available for testing here: https://drive.google.com/file/d/11szy6tlDFL5d7A6JdZLEwr0ZE1yQfNkE/view?usp=sharing

All being well it will be released in the next few weeks.

aaflatooni, 2021-01-05:
Thanks,
The google drive has brcmfmac43455-sdio.bin. Can you confirm if I would be installing using the following commands:

rpi ~$ sudo dpkg --purge firmware-brcm80211
rpi ~$ sudo dpkg --install brcmfmac43455-sdio.bin

pelwell, 2021-01-06:
No, not like that - this is a simple file substitution:
```
pi@raspberrypi:~$ sudo mv /lib/firmware/brcm/brcmfmac43455-sdio.bin{,.bak}
pi@raspberrypi:~$ sudo cp brcmfmac43455-sdio.bin /lib/firmware/brcm/
```
Once the new firmware is officially released the upgrade process will become:
```
pi@raspberrypi:~$ sudo apt update
pi@raspberrypi:~$ sudo apt install firmware-brcm80211
```

aaflatooni, 2021-01-10:
Thanks
I am getting the following error:

```
Jan 10 17:05:59 chpi kernel: brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2
Jan 10 17:05:59 chpi kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Jan 10 17:05:59 chpi kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e
Jan 10 17:05:59 chpi kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled

```
wpa_supplicant hangs after that and anything that I try like "ifconfig" or "iw wlan0 info" just sits there. 
Any thoughts?

aaflatooni, 2021-01-11:
My problem was unrelated.
I did a fresh install and see that "failed with error -2" is there as well.

pelwell, 2021-01-11:
That "error" message is alarming, but it just shows that we aren't currently using per-device "nvram" files. What the messages don't show is that it then goes on to load `brcm/brcmfmac43455-sdio.txt` successfully.

delijati, 2021-03-03:
Hi @pelwell i assume the new firmware is not part in https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-01-12/ ? If not do i get it via  [1]?
[1]
```
pi@raspberrypi:~$ sudo apt update
pi@raspberrypi:~$ sudo apt install firmware-brcm80211
```

GuillaumeRossolini, 2021-03-03:
@delijati you can simply:
sudo apt update && sudo apt upgrade
the firmware will be part of it