raspberrypi/firmware #1464: V4L2 recurent Exception stack after few hours/days of use, v4l2-ctl freeze, any hardware any Linux version

Serveurperso, 2020-09-02:
**Describe the bug**
Add a clear and concise description of what you think the bug is.

It's a old hard to get but bug but happens quite often

We use V4L2 on about <= 50 online 24/7 system, any raspbery PI (2/3A+/3B+/4) any camera (v1 any/v2 official/HQ official), Last & clean stable Buster from continuous integration (always up to date but no rpi-update).

cmdline : v4l2-ctl process freeze, never terminate

/var/log/syslog say :

```
Sep  2 14:06:52 raspberrypi systemd[1683]: Reached target Sockets.
Sep  2 14:06:52 raspberrypi systemd[1683]: Reached target Basic System.
Sep  2 14:06:52 raspberrypi systemd[1683]: Reached target Default.
Sep  2 14:06:52 raspberrypi systemd[1683]: Startup finished in 156ms.
Sep  2 14:06:52 raspberrypi systemd[1]: Started User Manager for UID 0.
Sep  2 14:06:52 raspberrypi systemd[1]: Started Session c7 of user root.
Sep  2 14:07:00 raspberrypi kernel: [184847.809038] bcm2835_mmal_vchiq: timed out waiting for sync completion
Sep  2 14:07:03 raspberrypi kernel: [184850.859089] bcm2835_mmal_vchiq: timed out waiting for sync completion
Sep  2 14:07:19 raspberrypi kernel: [184866.609349] bcm2835_mmal_vchiq: timed out waiting for sync completion
Sep  2 14:07:22 raspberrypi kernel: [184869.649375] bcm2835_mmal_vchiq: timed out waiting for sync completion
Sep  2 14:07:26 raspberrypi kernel: [184873.489445] bcm2835_mmal_vchiq: timed out waiting for sync completion
Sep  2 14:07:29 raspberrypi kernel: [184876.539490] bcm2835_mmal_vchiq: timed out waiting for sync completion
Sep  2 14:07:30 raspberrypi kernel: [184877.567074] bcm2835-v4l2: queue_setup: capture port not configured
Sep  2 14:08:27 raspberrypi kernel: [184935.410453] INFO: task HCEC Notify:31211 blocked for more than 120 seconds.
Sep  2 14:08:27 raspberrypi kernel: [184935.410462]       Tainted: G        WC        4.19.118-v7l+ #1311
Sep  2 14:08:27 raspberrypi kernel: [184935.410468] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Sep  2 14:08:27 raspberrypi kernel: [184935.410475] HCEC Notify     D    0 31211  31206 0x00000005
Sep  2 14:08:27 raspberrypi kernel: [184935.410501] [<c09d9ee4>] (__schedule) from [<c09da554>] (schedule+0x50/0xa8)
Sep  2 14:08:27 raspberrypi kernel: [184935.410514] [<c09da554>] (schedule) from [<c09de53c>] (schedule_timeout+0x200/0x428)
Sep  2 14:08:27 raspberrypi kernel: [184935.410524] [<c09de53c>] (schedule_timeout) from [<c09dcad0>] (__down+0x98/0xec)
Sep  2 14:08:27 raspberrypi kernel: [184935.410536] [<c09dcad0>] (__down) from [<c0278e44>] (down+0x54/0x68)
Sep  2 14:08:27 raspberrypi kernel: [184935.410551] [<c0278e44>] (down) from [<c087dfac>] (vchiq_release+0x160/0x354)
Sep  2 14:08:27 raspberrypi kernel: [184935.410566] [<c087dfac>] (vchiq_release) from [<c03d6d04>] (__fput+0xa4/0x1e8)
Sep  2 14:08:27 raspberrypi kernel: [184935.410580] [<c03d6d04>] (__fput) from [<c03d6eb8>] (____fput+0x18/0x1c)
Sep  2 14:08:27 raspberrypi kernel: [184935.410593] [<c03d6eb8>] (____fput) from [<c0242734>] (task_work_run+0xbc/0xe0)
Sep  2 14:08:27 raspberrypi kernel: [184935.410607] [<c0242734>] (task_work_run) from [<c0226bd0>] (do_exit+0x448/0xc14)
Sep  2 14:08:27 raspberrypi kernel: [184935.410619] [<c0226bd0>] (do_exit) from [<c0227434>] (do_group_exit+0x4c/0xe4)
Sep  2 14:08:27 raspberrypi kernel: [184935.410631] [<c0227434>] (do_group_exit) from [<c02337b4>] (get_signal+0x1a4/0x8cc)
Sep  2 14:08:27 raspberrypi kernel: [184935.410643] [<c02337b4>] (get_signal) from [<c020c744>] (do_signal+0x84/0x474)
Sep  2 14:08:27 raspberrypi kernel: [184935.410653] [<c020c744>] (do_signal) from [<c020cd40>] (do_work_pending+0xdc/0xf4)
Sep  2 14:08:27 raspberrypi kernel: [184935.410663] [<c020cd40>] (do_work_pending) from [<c0201034>] (slow_work_pending+0xc/0x20)
Sep  2 14:08:27 raspberrypi kernel: [184935.410670] Exception stack(0xda813fb0 to 0xda813ff8)
Sep  2 14:08:27 raspberrypi kernel: [184935.410678] 3fa0:                                     b6d93384 00000189 00000001 00000000
Sep  2 14:08:27 raspberrypi kernel: [184935.410687] 3fc0: 00000000 ffffffff 00000000 000000f0 be9da3fa b6d926e0 b6d942a8 b51fcf9c
Sep  2 14:08:27 raspberrypi kernel: [184935.410695] 3fe0: 00000000 b51fcdb0 00000000 b6d20088 80000010 b6d93384
Sep  2 14:08:27 raspberrypi kernel: [184935.410704] INFO: task HCEC Notify:31235 blocked for more than 120 seconds.
Sep  2 14:08:27 raspberrypi kernel: [184935.410710]       Tainted: G        WC        4.19.118-v7l+ #1311
Sep  2 14:08:27 raspberrypi kernel: [184935.410715] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Sep  2 14:08:27 raspberrypi kernel: [184935.410721] HCEC Notify     D    0 31235  31230 0x00000005
Sep  2 14:08:27 raspberrypi kernel: [184935.410737] [<c09d9ee4>] (__schedule) from [<c09da554>] (schedule+0x50/0xa8)
Sep  2 14:08:27 raspberrypi kernel: [184935.410748] [<c09da554>] (schedule) from [<c09de53c>] (schedule_timeout+0x200/0x428)
Sep  2 14:08:27 raspberrypi kernel: [184935.410757] [<c09de53c>] (schedule_timeout) from [<c09dcad0>] (__down+0x98/0xec)
Sep  2 14:08:27 raspberrypi kernel: [184935.410766] [<c09dcad0>] (__down) from [<c0278e44>] (down+0x54/0x68)
Sep  2 14:08:27 raspberrypi kernel: [184935.410777] [<c0278e44>] (down) from [<c087dfac>] (vchiq_release+0x160/0x354)
Sep  2 14:08:27 raspberrypi kernel: [184935.410790] [<c087dfac>] (vchiq_release) from [<c03d6d04>] (__fput+0xa4/0x1e8)
Sep  2 14:08:27 raspberrypi kernel: [184935.410801] [<c03d6d04>] (__fput) from [<c03d6eb8>] (____fput+0x18/0x1c)
Sep  2 14:08:27 raspberrypi kernel: [184935.410812] [<c03d6eb8>] (____fput) from [<c0242734>] (task_work_run+0xbc/0xe0)
Sep  2 14:08:27 raspberrypi kernel: [184935.410824] [<c0242734>] (task_work_run) from [<c0226bd0>] (do_exit+0x448/0xc14)
Sep  2 14:08:27 raspberrypi kernel: [184935.410835] [<c0226bd0>] (do_exit) from [<c0227434>] (do_group_exit+0x4c/0xe4)
Sep  2 14:08:27 raspberrypi kernel: [184935.410846] [<c0227434>] (do_group_exit) from [<c02337b4>] (get_signal+0x1a4/0x8cc)
Sep  2 14:08:27 raspberrypi kernel: [184935.410855] [<c02337b4>] (get_signal) from [<c020c744>] (do_signal+0x84/0x474)
Sep  2 14:08:27 raspberrypi kernel: [184935.410864] [<c020c744>] (do_signal) from [<c020cd40>] (do_work_pending+0xdc/0xf4)
Sep  2 14:08:27 raspberrypi kernel: [184935.410874] [<c020cd40>] (do_work_pending) from [<c0201034>] (slow_work_pending+0xc/0x20)
Sep  2 14:08:27 raspberrypi kernel: [184935.410880] Exception stack(0xd40d1fb0 to 0xd40d1ff8)
Sep  2 14:08:27 raspberrypi kernel: [184935.410888] 1fa0:                                     b6dc5384 00000189 00000001 00000000
Sep  2 14:08:27 raspberrypi kernel: [184935.410896] 1fc0: 00000000 ffffffff 00000000 000000f0 bec5e3fa b6dc46e0 b6dc62a8 b51fcf9c
Sep  2 14:08:27 raspberrypi kernel: [184935.410903] 1fe0: 00000000 b51fcdb0 00000000 b6d52088 80000010 b6dc5384
Sep  2 14:08:27 raspberrypi kernel: [184935.410911] INFO: task VCHIQ completio:31526 blocked for more than 120 seconds.
[... 69 lines cut ...]

All Raspbery PI from 2 to 4 (RPI 2 / 3A+ / 3B+ / 4)

* Which OS and version (`cat /etc/rpi-issue`)?

Last Buster, same bug since Jessie.

* Which firmware version (`vcgencmd version`)?

All since few years.

* Which kernel version (`uname -a`)?

Last / All since Jessie

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

Complete /var/log/syslog here :
https://www.serveurperso.com/temp/syslog

**Additional context**
Add any other relevant context for the problem.

I'm Available for any help and debug. I keep a buggy system without reboot for help.

satmandu, 2020-09-02:
Also seeing a similar issue with rpi-update to current firmware on rpi4b using raspi-os 32 bit and kernel 5.4.61.

My logs: http://paste.debian.net/1162309/

Serveurperso, 2020-09-02:
It's not similar issue.
Mine only appear after many hours of usage of V4L2 H.264 and a lot of v4l2-ctl binary call.
raspistill freeze also.
After a random time from 1 day to few days of usage.
This is a Video Core / Kernel crash...

Serveurperso, 2020-09-02:
Intermittently streaming  : automatic stream sleep after few minute of inactivity on the robot);
But with dynamic control changes when stream is enabled : automatic low bitrate on network latency, resume to high bitrate on good network, and live brighness/contrast modification.

Only when streaming is disabled, raspistill can take some 1 (normal) or 3 (software HDR mode) snapshots every 1 minute.
If streaming is enabled, any remaining raspistill process are killed before.

No problem for many days, but some time I get this kernel crash. It's not a memory leak.
When there is the problem, I must reboot to get video working again.
And reboot is longer.

Serveurperso, 2020-09-02:
The only way is the long way:( I can't reproduce, it's every few day I get a random robot in this strange state.
I can try to make a stress tester :(

6by9, 2020-09-02:
Low bitrate is likely to be the key. I'm already aware of an issue in the firmware where the codec doesn't close cleanly if bitrate control caused the last frame submitted to be skipped, and results in the camera subsystem wedging. It's a question of finding the time to investigate it.

Serveurperso, 2020-09-02:
I don't think it's the bitrate (?) because :

When the problem appear v4l2-ctl is freeze and a callback in never called.

Because problem never happen here (contast/brigh adjustement):
```
    contrastBoost = tx.getCommande1(hard.CONTRASTBOOSTSWITCH);
   if(contrastBoost != oldContrastBoost) {
    if(contrastBoost) {
     exec("v4l2-ctl", SYS.V4L2 + " -c brightness=" + confVideo.BRIGHTNESSBOOST +
                                    ",contrast=" + confVideo.CONTRASTBOOST, function() {
     });
    } else {
     exec("v4l2-ctl", SYS.V4L2 + " -c brightness=" + confVideo.BRIGHTNESS +
                                    ",contrast=" + confVideo.CONTRAST, function() {
     });
    }
    oldContrastBoost = contrastBoost;
   }
```

Because problem never happen here (BITRATE adjustement):
```
 if(predictiveLatency < SYS.LATENCYALARMEND && latencyAlarm) {
  trace(predictiveLatency + " ms latency, return to configured video bitrate");
  exec("v4l2-ctl", SYS.V4L2 + " -c video_bitrate=" + confVideo.BITRATE, function() {
  });
  latencyAlarm = false;
 } else if(predictiveLatency > SYS.LATENCYALARMBEGIN && !latencyAlarm) {
  trace(predictiveLatency + " ms latency, stop the motors and switch to reduced video bitrate");
  exec("v4l2-ctl", SYS.V4L2 + " -c video_bitrate=" + SYS.EMERGENCYBITRATE, function() {
  });
  latencyAlarm = true;
 }

```

Problem ALWAYS appear here :
The v4l2-ctl here freeze and my Node.js callback is never called... never I must reboot:(
It's the more complete configuration, call is few time by days when the user make modification change.

```
 exec("v4l2-ctl", SYS.V4L2 + " -v width=" + confVideo.WIDTH +
                                ",height=" + confVideo.HEIGHT +
                                ",pixelformat=4" +
                             " -p " + confVideo.FPS +
                             " -c h264_profile=0" +
                                ",repeat_sequence_header=1" +
                                ",rotate=" + confVideo.ROTATE +
                                ",video_bitrate=" + confVideo.BITRATE +
                                ",brightness=" + brightness +
                                ",contrast=" + contrast, function() {
  callback();
 });

```

Serveurperso, 2020-09-02:
Bitrate adjustment are very very frequent and never "crash the kernel".
Light adjustment is a visitor button and can be stressed, never crash the kernel

The complete configuration v4l2-ctl cmdline I give you always make the crash.

6by9, 2020-09-02:
If you're setting the width and height then that must be being executed with streaming stopped. I suspect the previous stop_streaming call has failed and wedged the GPU. It completes after a timeout, but your next one is wedging.

Adjusting the bitrate whilst streaming isn't the problem, it is when stopping streaming and the last frame of the stream is dropped by the bit rate control algorithm.

Serveurperso, 2020-09-02:
The TCP recovery H.264 BITRATE we use is (50kbps) 50000
User can configure between
```
          "BITRATE": {
            "type": "integer",
            "minimum": 150000,
            "maximum": 3000000
          },
```

You message arrived when I make this one.

Serveurperso, 2020-09-02:
> If you're setting the width and height then that must be being executed with streaming stopped. I suspect the previous stop_streaming call has failed and wedged the GPU. It completes after a timeout, but your next one is wedging.

I never launch v4l2-ctl complete configuration when the streaming is enabled.
There is a process kill callback.
Can you a protection timeout to never get this again ?
I must give more temporal space when I stop a stream before launch the complete configuration (width/height) ?

6by9, 2020-09-02:
Add ```start_debug=1``` to config.txt, and provide the output from ```sudo vcdbg log assert``` and ```sudo vcdbg log msg``` when it has gone wrong.

Serveurperso, 2020-09-02:
I must do this on many robot to get a chance of crash:(

Serveurperso, 2020-09-02:
The problem occur only when there is also PIGPIO initialization with the v4l2-ctl, it use DMA trick to make precise PWM.

Serveurperso, 2020-09-02:
I make a working stress tester, first time I can reproduce the bug !!!

Serveurperso, 2020-09-02:
```
2988287.376: venc: venc_rc: frame 244 aborted bits 43424 > -57415 (HRD), QP=51
2988322.911: venc: venc_rc: frame 245 aborted bits 43624 > -55948 (HRD), QP=51
2988353.883: venc: venc_rc: frame 246 aborted bits 43912 > -54480 (HRD), QP=51
2988387.229: venc: venc_rc: frame 247 aborted bits 43392 > -53012 (HRD), QP=51
2988424.195: venc: venc_rc: frame 248 aborted bits 42784 > -51545 (HRD), QP=51
2988457.542: venc: venc_rc: frame 249 aborted bits 43784 > -50077 (HRD), QP=51
2988493.707: venc: venc_rc: frame 250 aborted bits 43656 > -48609 (HRD), QP=51
2988522.437: venc: venc_rc: frame 251 aborted bits 43656 > -47142 (HRD), QP=51
2988555.753: venc: venc_rc: frame 252 aborted bits 43536 > -45674 (HRD), QP=51
2988589.071: venc: venc_rc: frame 253 aborted bits 42720 > -44206 (HRD), QP=51
2988622.354: venc: venc_rc: frame 254 aborted bits 43640 > -42739 (HRD), QP=51
2988658.800: venc: venc_rc: frame 255 aborted bits 43040 > -41271 (HRD), QP=51
2988689.034: venc: venc_rc: frame 256 aborted bits 43992 > -39804 (HRD), QP=51
2988724.211: venc: venc_rc: frame 257 aborted bits 43432 > -38336 (HRD), QP=51
2988757.363: venc: venc_rc: frame 258 aborted bits 43504 > -36868 (HRD), QP=51
2988790.711: venc: venc_rc: frame 259 aborted bits 43752 > -35401 (HRD), QP=51
2988826.744: venc: venc_rc: frame 260 aborted bits 43256 > -33933 (HRD), QP=51
2988855.572: venc: venc_rc: frame 261 aborted bits 43544 > -32465 (HRD), QP=51
2988888.933: venc: venc_rc: frame 262 aborted bits 43416 > -30998 (HRD), QP=51
2988926.465: venc: venc_rc: frame 263 aborted bits 43344 > -29530 (HRD), QP=51
2988957.459: venc: venc_rc: frame 264 aborted bits 43480 > -28062 (HRD), QP=51
2988957.705: venc: venc_exit: cnt != out_cnt (127, 128)
2518864.499: mmalsrv: mmal_server_create_control_service: failed to create control service - error 1
```

Serveurperso, 2020-09-02:
I must manual stress for about 30 minutes to get theses information :

Only the last line appear specific to the bug
```
(root|~) vcdbg log msg
1952515.424: venc: venc_rc: frame 1 aborted bits 6440 > -25298 (HRD), QP=35
1953261.371: Using RPI AWB
1953730.066: venc: venc_rc: frame 1 aborted bits 6912 > -25514 (HRD), QP=35
1954099.104: Using RPI AWB
1954570.715: venc: venc_rc: frame 1 aborted bits 6568 > -25794 (HRD), QP=35
1954837.473: Using RPI AWB
1955362.372: Using RPI AWB
1955951.379: Using RPI AWB
1957083.752: Using RPI AWB
1957555.150: venc: venc_rc: frame 1 aborted bits 6304 > -25450 (HRD), QP=35
1958134.265: Using RPI AWB
1958605.590: venc: venc_rc: frame 1 aborted bits 6472 > -25682 (HRD), QP=35
1958638.955: venc: venc_rc: frame 2 aborted bits 13712 > -24015 (HRD), QP=35
1958642.665: venc: venc_exit: cnt != out_cnt (1, 2)
```

I get a lof of related error here :
```
(root|~) vcdbg log assert
1852702.030: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev 6379679
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1858375.794: assert( temps->last_tone_mapping_params_valid ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_agc.c::isp_tuner_brcm_agc_switch_mode line 2745 rev 6379679
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1858585.645: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev 6379679
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1858630.354: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev 6379679
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1906523.039: assert( temps->last_tone_mapping_params_valid ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_agc.c::isp_tuner_brcm_agc_switch_mode line 2745 rev 6379679
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1906732.797: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev 6379679
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1906777.473: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev 6379679
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1912462.260: assert( temps->last_tone_mapping_params_valid ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_agc.c::isp_tuner_brcm_agc_switch_mode line 2745 rev 6379679
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1912672.092: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev 6379679
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1912716.708: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev 6379679
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1958642.474: assert( enc->cnt.frame == enc->out.outframecnt ) failed; ../../../../../codecs/video/hw/enc/venc.c::venc_exit line 2224 rev 6379679
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1958642.776: assert( vstatus == VIDEO_SUCCESS ) failed; ../../../../../middleware/openmaxil/components/ve_worker.c::ve_worker_thread line 168 rev 6379679
vcdbg_ctx_get_dump_stack: dump_stack failed
```

Serveurperso, 2020-09-05:
I just get a crash without entire configuration applying (after 1 day uptime),
Only v4l2-ctl with brighness and/or bitrate

6by9, 2020-09-07:
```
1958642.474: assert( enc->cnt.frame == enc->out.outframecnt ) failed; ../../../../../codecs/video/hw/enc/venc.c::venc_exit line 2224 rev 6379679
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1958642.776: assert( vstatus == VIDEO_SUCCESS ) failed; ../../../../../middleware/openmaxil/components/ve_worker.c::ve_worker_thread line 168 rev 6379679
vcdbg_ctx_get_dump_stack: dump_stack failed
```
```
1958638.955: venc: venc_rc: frame 2 aborted bits 13712 > -24015 (HRD), QP=35
1958642.665: venc: venc_exit: cnt != out_cnt (1, 2)
```
That confirms that it is the codec failing to close down properly due to bitrate control dropping the last frame.
It's already on the list.

Serveurperso, 2020-09-07:
Thanks for you feedback 6by9.
In the meantime I will set up an automatic restart on the robots with statistics on crashes, this will serve as a test once the correction is applied ...

HughWarrington, 2020-09-07:
I think btw #1428 is a dupe of this ticket -- and may be what 6by9 means by 'on the list'.

Serveurperso, 2020-09-17:
Now we use a WatchDog to detect the "venc: venc_exit: cnt != out_cnt" here is frequency occurrence of this problem :

```
14/09 08:36:07:922 |                 |              | 176.179.178.138 | raspA        | Client | Following this uncaught exception, the Node.js process will be terminated automatically
14/09 08:42:14:668 |                 |              | 176.179.178.138 | raspA        | Client | Following this uncaught exception, the Node.js process will be terminated automatically
14/09 18:12:25:688 |                 |              | 82.65.70.88     | WebServer    | Client | Following this uncaught exception, the Node.js process will be terminated automatically
14/09 19:05:38:492 |                 |              | 92.184.117.184  | CrawLard     | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
14/09 19:07:28:364 | 80.12.83.251    | Pascal       | 92.184.108.34   | CrawLard     | Yes 14/09 19:05:38:492 | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
14/09 20:25:47:005 |                 |              | 80.12.83.251    | GPS          | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
14/09 20:34:59:815 |                 |              | 188.154.254.253 | briecam      | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
15/09 06:54:20:859 |                 |              | 80.215.178.56   | Robot4G      | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
15/09 11:03:29:245 |                 |              | 92.169.207.145  | FredLab      | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
15/09 12:35:08:378 |                 |              | 80.215.178.205  | Robot4G      | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
15/09 13:51:27:959 |                 |              | 92.169.207.145  | FredLab      | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
15/09 17:36:32:153 |                 |              | 80.215.210.40   | TRX4         | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
15/09 18:52:04:267 |                 |              | 92.169.207.145  | FredLab      | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
15/09 18:53:34:440 |                 |              | 83.194.64.153   | BigBang      | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
15/09 20:32:03:023 |                 |              | 92.169.207.145  | FredLab      | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
16/09 13:06:55:768 |                 |              | 92.169.207.145  | FredLab      | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
16/09 13:31:13:880 |                 |              | 80.215.178.229  | Robot4G      | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
16/09 14:29:00:628 |                 |              | 176.130.146.217 | Explorer130  | Client | Following this uncaught exception, the Node.js process will be terminated automatically
16/09 17:59:36:566 |                 |              | 82.64.30.121    | TiTiLab      | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
16/09 19:10:13:211 |                 |              | 82.64.30.121    | TiTiLab      | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
17/09 09:25:03:120 |                 |              | 82.64.30.121    | TiTiLab      | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
17/09 11:18:54:758 |                 |              | 82.64.30.121    | TiTiLab      | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
17/09 16:42:48:187 |                 |              | 80.12.83.251    | GPS          | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
17/09 16:51:45:636 |                 |              | 188.154.254.253 | pancake      | Client | Following a Raspberry PI VideoCore crash, the system will be restarted automatically
```

But we get 1 time another strange videocore crash :

vcdbg msg log :
```
3299299.178: Using RPI AWB
3305036.607: Using RPI AWB
3330363.889: Using RPI AWB
3411092.513: mmal: mmal_ril_set_port_settings: ril.video_encode:in:0(OPQV): failed to set port definition (4) buffers 0/0/0
3411092.549: mmal: mmal_ril_set_port_settings: ril.video_encode:in:0(OPQV): fail: color format 0x0, encoding 0x0 stride 0
3411092.596: mmalsrv: mmal_server_do_port_info_set: ril.video_encode:in:0(OPQV): failed (handle 1, status 3)
3709968.155: Using RPI AWB
3715676.225: Using RPI AWB
3721441.400: Using RPI AWB
```

Syslog part for the another strange videocore crash:
```
Sep 17 17:00:02 Box systemd[1]: Started Time & Date Service.
Sep 17 17:00:18 Box kernel: [16210.442399] bcm2835_mmal_vchiq: buffer_from_host: msg_context not allocated, buf a952d818
Sep 17 17:00:18 Box kernel: [16210.442423] bcm2835-v4l2: Failed to enable capture port - error -22. Disabling camera port again
[... cut ...]
Sep 17 17:00:18 Box kernel: [16210.460605] Exception stack(0xae3effa8 to 0xae3efff0)
Sep 17 17:00:18 Box kernel: [16210.460616] ffa0:                   00020000 7ff00000 00000003 76925000 00020000 00000000
Sep 17 17:00:18 Box kernel: [16210.460626] ffc0: 00020000 7ff00000 00000003 00000003 00000003 00000000 00020000 00000000
Sep 17 17:00:18 Box kernel: [16210.460633] ffe0: 0000006c 7ea14b40 0001438c 76e85880
Sep 17 17:00:18 Box kernel: [16210.460643] ---[ end trace c981337579f78dad ]---
Sep 17 17:00:32 Box systemd[1]: systemd-timedated.service: Succeeded.
```

Serveurperso, 2020-09-29:
Another rare videocore crash.
(where v4l2-ctl --all freeze also)

```
Sep 29 18:58:52 raspberrypi kernel: [196478.158463] ------------[ cut here ]------------
Sep 29 18:58:52 raspberrypi kernel: [196478.158488] WARNING: CPU: 3 PID: 5793 at drivers/firmware/raspberrypi.c:63 rpi_firmware_transaction+0x108/0x128
Sep 29 18:58:52 raspberrypi kernel: [196478.158492] Firmware transaction timeout
Sep 29 18:58:52 raspberrypi kernel: [196478.158494] Modules linked in: 8021q garp stp llc brcmfmac brcmutil evdev snd_usb_audio sha256_generic libsha256 snd_hwdep snd_usbmidi_lib snd_rawmidi snd_seq_device cfg80211 rfkill raspberrypi_hwmon snd_bcm2835(C) i2c_bcm2835 snd_pcm bcm2835_codec(C) snd_timer v4l2_mem2mem snd bcm2835_isp(C) videobuf2_dma_contig uio_pdrv_genirq uio fixed bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) vc_sm_cma(C) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc i2c_dev ip_tables x_tables ipv6 nf_defrag_ipv6
Sep 29 18:58:52 raspberrypi kernel: [196478.158593] CPU: 3 PID: 5793 Comm: kworker/3:1 Tainted: G         C        5.4.51-v7+ #1333
Sep 29 18:58:52 raspberrypi kernel: [196478.158595] Hardware name: BCM2835
Sep 29 18:58:52 raspberrypi kernel: [196478.158605] Workqueue: events get_values_poll [raspberrypi_hwmon]
Sep 29 18:58:52 raspberrypi kernel: [196478.158608] Backtrace: 
Sep 29 18:58:52 raspberrypi kernel: [196478.158619] [<8010d480>] (dump_backtrace) from [<8010d778>] (show_stack+0x20/0x24)
Sep 29 18:58:52 raspberrypi kernel: [196478.158625]  r6:902d6000 r5:00000000 r4:80d95cb8 r3:3e5e7a8d
Sep 29 18:58:52 raspberrypi kernel: [196478.158633] [<8010d758>] (show_stack) from [<808cdbe4>] (dump_stack+0xe0/0x124)
Sep 29 18:58:52 raspberrypi kernel: [196478.158640] [<808cdb04>] (dump_stack) from [<8012002c>] (__warn+0xec/0x104)
Sep 29 18:58:52 raspberrypi kernel: [196478.158646]  r8:0000003f r7:00000009 r6:80b02c40 r5:00000000 r4:902d7dfc r3:3e5e7a8d
Sep 29 18:58:52 raspberrypi kernel: [196478.158652] [<8011ff40>] (__warn) from [<801200c8>] (warn_slowpath_fmt+0x84/0xc0)
Sep 29 18:58:52 raspberrypi kernel: [196478.158658]  r9:80b02c40 r8:0000003f r7:80738398 r6:00000009 r5:80b02c60 r4:80d04f48
Sep 29 18:58:52 raspberrypi kernel: [196478.158664] [<80120048>] (warn_slowpath_fmt) from [<80738398>] (rpi_firmware_transaction+0x108/0x128)
Sep 29 18:58:52 raspberrypi kernel: [196478.158670]  r9:9364e240 r8:00000010 r7:00000000 r6:ffffff92 r5:9364e240 r4:80d04f48
Sep 29 18:58:52 raspberrypi kernel: [196478.158676] [<80738290>] (rpi_firmware_transaction) from [<80738484>] (rpi_firmware_property_list+0xcc/0x180)
Sep 29 18:58:52 raspberrypi kernel: [196478.158681]  r7:80d04f48 r6:93c04000 r5:00001000 r4:40000006
Sep 29 18:58:52 raspberrypi kernel: [196478.158687] [<807383b8>] (rpi_firmware_property_list) from [<807385b0>] (rpi_firmware_property+0x78/0xf8)
Sep 29 18:58:52 raspberrypi kernel: [196478.158693]  r10:00000000 r9:9364e240 r8:00030046 r7:902d7ee0 r6:00000010 r5:00000004
Sep 29 18:58:52 raspberrypi kernel: [196478.158695]  r4:903fee80
Sep 29 18:58:52 raspberrypi kernel: [196478.158703] [<80738538>] (rpi_firmware_property) from [<7f13e0c0>] (get_values_poll+0x4c/0x15c [raspberrypi_hwmon])
Sep 29 18:58:52 raspberrypi kernel: [196478.158709]  r10:00000000 r9:000000c0 r8:00000000 r7:97c58500 r6:97c55340 r5:90b23c4c
Sep 29 18:58:52 raspberrypi kernel: [196478.158711]  r4:80d04f48
Sep 29 18:58:52 raspberrypi kernel: [196478.158720] [<7f13e074>] (get_values_poll [raspberrypi_hwmon]) from [<8013c2e4>] (process_one_work+0x17c/0x4b4)
Sep 29 18:58:52 raspberrypi kernel: [196478.158723]  r5:8fbb3380 r4:90b23c4c
Sep 29 18:58:52 raspberrypi kernel: [196478.158730] [<8013c168>] (process_one_work) from [<8013cb4c>] (worker_thread+0x54/0x5b4)
Sep 29 18:58:52 raspberrypi kernel: [196478.158736]  r10:8fbb3380 r9:902d6038 r8:80d03d00 r7:97c55358 r6:00000008 r5:8fbb3394
Sep 29 18:58:52 raspberrypi kernel: [196478.158738]  r4:97c55340
Sep 29 18:58:52 raspberrypi kernel: [196478.158745] [<8013caf8>] (worker_thread) from [<80142f30>] (kthread+0x13c/0x168)
Sep 29 18:58:52 raspberrypi kernel: [196478.158751]  r10:92f8701c r9:8fb51e74 r8:8013caf8 r7:8fbb3380 r6:00000000 r5:903fe600
Sep 29 18:58:52 raspberrypi kernel: [196478.158754]  r4:92f87000
Sep 29 18:58:52 raspberrypi kernel: [196478.158760] [<80142df4>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
Sep 29 18:58:52 raspberrypi kernel: [196478.158763] Exception stack(0x902d7fb0 to 0x902d7ff8)
Sep 29 18:58:52 raspberrypi kernel: [196478.158768] 7fa0:                                     00000000 00000000 00000000 00000000
Sep 29 18:58:52 raspberrypi kernel: [196478.158773] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Sep 29 18:58:52 raspberrypi kernel: [196478.158778] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Sep 29 18:58:52 raspberrypi kernel: [196478.158784]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80142df4
Sep 29 18:58:52 raspberrypi kernel: [196478.158787]  r4:903fe600 r3:80104648
Sep 29 18:58:52 raspberrypi kernel: [196478.158799] ---[ end trace 3e8e251901764094 ]---
Sep 29 18:58:52 raspberrypi kernel: [196478.158827] hwmon hwmon1: Failed to get throttled (-110)
Sep 29 18:58:55 raspberrypi kernel: [196480.878528] cpu cpu0: dev_pm_opp_set_rate: failed to find current OPP for freq 4294967186 (-34)
Sep 29 18:58:57 raspberrypi kernel: [196482.958520] raspberrypi-clk soc:firmware:clocks: Failed to change fw-clk-arm frequency: -110
[... cut ...]
Sep 29 19:19:04 raspberrypi kernel: [197689.377807] Exception stack(0x903b7fa8 to 0x903b7ff0)
Sep 29 19:19:04 raspberrypi kernel: [197689.377812] 7fa0:                   00020000 7ff00000 00000003 76941000 00020000 00000000
Sep 29 19:19:04 raspberrypi kernel: [197689.377818] 7fc0: 00020000 7ff00000 00000003 00000003 00000003 00000000 00020000 00000000
Sep 29 19:19:04 raspberrypi kernel: [197689.377822] 7fe0: 0000006c 7efd7b40 0001438c 76ea1880
Sep 29 19:19:04 raspberrypi kernel: [197689.377827] ---[ end trace 3e8e251901764098 ]---
```

Serveurperso, 2020-09-30:
Another one (a PI 3 A+ this time) only few hours after clean reboot :

All robots run the current stable Kernel (apt update/upgrade without rpi-update)
The current stable Kernel is pretty unstable:(

```
Sep 30 00:00:01 raspberrypi rsyslogd:  [origin software="rsyslogd" swVersion="8.1901.0" x-pid="334" x-info="https://www.rsyslog.com"] rsyslogd was HUPed
Sep 30 00:00:01 raspberrypi systemd[1]: logrotate.service: Succeeded.
Sep 30 00:00:01 raspberrypi systemd[1]: Started Rotate log files.
Sep 30 00:00:01 raspberrypi CRON[9679]: (root) CMD (/usr/local/vigiclient/vigiupdate.sh >> /var/log/vigiupdate.log 2>&1)
Sep 30 00:00:02 raspberrypi systemd[1]: man-db.service: Succeeded.
Sep 30 00:00:02 raspberrypi systemd[1]: Started Daily man-db regeneration.
Sep 30 00:00:02 raspberrypi dbus-daemon[327]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service' requested by ':1.106' (uid=0 pid=9697 comm="timedatectl status ")
Sep 30 00:00:02 raspberrypi systemd[1]: Starting Time & Date Service...
Sep 30 00:00:02 raspberrypi dbus-daemon[327]: [system] Successfully activated service 'org.freedesktop.timedate1'
Sep 30 00:00:02 raspberrypi systemd[1]: Started Time & Date Service.
Sep 30 00:00:32 raspberrypi systemd[1]: systemd-timedated.service: Succeeded.
Sep 30 00:05:01 raspberrypi CRON[9858]: (root) CMD (/usr/local/vigiclient/vigiupdate.sh >> /var/log/vigiupdate.log 2>&1)
Sep 30 00:05:01 raspberrypi dbus-daemon[327]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service' requested by ':1.108' (uid=0 pid=9876 comm="timedatectl status ")
Sep 30 00:05:01 raspberrypi systemd[1]: Starting Time & Date Service...
Sep 30 00:05:02 raspberrypi dbus-daemon[327]: [system] Successfully activated service 'org.freedesktop.timedate1'
Sep 30 00:05:02 raspberrypi systemd[1]: Started Time & Date Service.
Sep 30 00:05:32 raspberrypi systemd[1]: systemd-timedated.service: Succeeded.
Sep 30 00:10:01 raspberrypi CRON[10036]: (root) CMD (/usr/local/vigiclient/vigiupdate.sh >> /var/log/vigiupdate.log 2>&1)
Sep 30 00:10:02 raspberrypi dbus-daemon[327]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service' requested by ':1.110' (uid=0 pid=10054 comm="timedatectl status ")
Sep 30 00:10:02 raspberrypi systemd[1]: Starting Time & Date Service...
Sep 30 00:10:02 raspberrypi dbus-daemon[327]: [system] Successfully activated service 'org.freedesktop.timedate1'
Sep 30 00:10:02 raspberrypi systemd[1]: Started Time & Date Service.
Sep 30 00:10:32 raspberrypi systemd[1]: systemd-timedated.service: Succeeded.
Sep 30 00:15:01 raspberrypi CRON[10216]: (root) CMD (/usr/local/vigiclient/vigiupdate.sh >> /var/log/vigiupdate.log 2>&1)
Sep 30 00:15:01 raspberrypi dbus-daemon[327]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service' requested by ':1.112' (uid=0 pid=10234 comm="timedatectl status ")
Sep 30 00:15:01 raspberrypi systemd[1]: Starting Time & Date Service...
Sep 30 00:15:02 raspberrypi dbus-daemon[327]: [system] Successfully activated service 'org.freedesktop.timedate1'
Sep 30 00:15:02 raspberrypi systemd[1]: Started Time & Date Service.
Sep 30 00:15:32 raspberrypi systemd[1]: systemd-timedated.service: Succeeded.
Sep 30 00:17:01 raspberrypi CRON[10320]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Sep 30 00:20:01 raspberrypi CRON[10407]: (root) CMD (/usr/local/vigiclient/vigiupdate.sh >> /var/log/vigiupdate.log 2>&1)
Sep 30 00:20:02 raspberrypi dbus-daemon[327]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service' requested by ':1.114' (uid=0 pid=10425 comm="timedatectl status ")
Sep 30 00:20:02 raspberrypi systemd[1]: Starting Time & Date Service...
Sep 30 00:20:02 raspberrypi dbus-daemon[327]: [system] Successfully activated service 'org.freedesktop.timedate1'
Sep 30 00:20:02 raspberrypi systemd[1]: Started Time & Date Service.
Sep 30 00:20:32 raspberrypi systemd[1]: systemd-timedated.service: Succeeded.
Sep 30 00:25:01 raspberrypi CRON[10587]: (root) CMD (/usr/local/vigiclient/vigiupdate.sh >> /var/log/vigiupdate.log 2>&1)
Sep 30 00:25:01 raspberrypi dbus-daemon[327]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service' requested by ':1.116' (uid=0 pid=10605 comm="timedatectl status ")
Sep 30 00:25:01 raspberrypi systemd[1]: Starting Time & Date Service...
Sep 30 00:25:02 raspberrypi dbus-daemon[327]: [system] Successfully activated service 'org.freedesktop.timedate1'
Sep 30 00:25:02 raspberrypi systemd[1]: Started Time & Date Service.
Sep 30 00:25:32 raspberrypi systemd[1]: systemd-timedated.service: Succeeded.
Sep 30 00:30:01 raspberrypi CRON[10766]: (root) CMD (/usr/local/vigiclient/vigiupdate.sh >> /var/log/vigiupdate.log 2>&1)
Sep 30 00:30:02 raspberrypi dbus-daemon[327]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service' requested by ':1.118' (uid=0 pid=10785 comm="timedatectl status ")
Sep 30 00:30:02 raspberrypi systemd[1]: Starting Time & Date Service...
Sep 30 00:30:02 raspberrypi dbus-daemon[327]: [system] Successfully activated service 'org.freedesktop.timedate1'
Sep 30 00:30:02 raspberrypi systemd[1]: Started Time & Date Service.
[... cut ...]
Sep 30 06:46:34 raspberrypi systemd[24666]: Listening on GnuPG cryptographic agent and passphrase cache (access for web browsers).
Sep 30 06:46:34 raspberrypi systemd[24666]: Reached target Sockets.
Sep 30 06:46:34 raspberrypi systemd[24666]: Reached target Paths.
Sep 30 06:46:34 raspberrypi systemd[24666]: Reached target Basic System.
Sep 30 06:46:34 raspberrypi systemd[24666]: Reached target Default.
Sep 30 06:46:34 raspberrypi systemd[24666]: Startup finished in 144ms.
Sep 30 06:46:34 raspberrypi systemd[1]: Started User Manager for UID 1000.
Sep 30 06:46:34 raspberrypi systemd[1]: Started Session c2 of user pi.
```

6by9, 2020-09-30:
Please stop posting kernel backtraces - we know the underlying problem is when using low bitrates and the final frame gets dropped by bitrate control. It's on the list of things to investigate.

pelwell, 2022-11-04:
The current kernel in `apt` fixes a bug in the handling of the CSI interface that could cause the unwanted overwriting of some adjacent memory. When did you last update/upgrade?

GrubbyHalo, 2022-11-04:
> The current kernel in `apt` fixes a bug in the handling of the CSI interface that could cause the unwanted overwriting of some adjacent memory. When did you last update/upgrade?

I guess I should have tried a firmware update first. That seemed to work. Thanks

6by9, 2022-11-04:
If you're really experiencing the same issue, then it's a firmware bug in the encoder. Getting ```venc: venc_exit: cnt != out_cnt (1, 2)``` was the tell-tale sign.
No, I haven't looked into it in great detail.

I had hoped that an ex-colleague from the codecs team at Broadcom was coming on as a contractor, but that hasn't happened for various reasons.

GrubbyHalo, 2022-11-04:
> No, I haven't looked into it in great detail.

Something must have changed between the 2 firmware version. Although the device is now updated, i think the old firmware was Feb 2021.  I'll write another sdcard with that old firmmware on it again get you some more info/logs on what I was seeing

6by9, 2022-11-04:
The original issue is chance.
The lockup occurs when bitrate control has to drop the frame immediately before shutdown due to exceeding the bitrate target by too great a margin, which will generally happen if running at a very low bitrate.

For reasons I don't fully understand the codec code refuses to close down, and that leaves the hardware in a bad state. There is a reason given in a comment, but it seems slightly spurious, so simply removing the condition may create other failures.

GrubbyHalo, 2022-11-04:
Well for what its worth, I can duplicate the problem reasonably successfully. If I start and stop the app about 5 or 6 times, never after the first to 5th, regardless of the interval between stop and starts, i get a crash with a log entry `venc: venc_rc: frame ### aborted bits #### > -#### (HRD), QP=##`  . And as far as i remember, it hangs on a call to `mmal_port_disable` ( think that's the call) forcing me to reboot.

6by9, 2022-11-04:
It will.
As per https://github.com/raspberrypi/firmware/issues/1464#issuecomment-685955797 you'll get 
```
1958638.955: venc: venc_rc: frame 2 aborted bits 13712 > -24015 (HRD), QP=35
```
will be logged whenever a frame is aborted due to rate control, but it is 
```
1958642.665: venc: venc_exit: cnt != out_cnt (1, 2)
```
that is the fatal case when that was on the last frame encoded, and it will result in bad things happening.
I could just drop that check and shutdown anyway, but I don't fully understand the consequences of that.