raspberrypi/firmware #1465: GPT SD card booting fails with 5.4 kernel

reraikes, 2020-09-03:
Booting an SD card which has a GPT partition table fails with the 5.4 kernel (2020-08-20-raspios-buster-armhf-lite.img).  A Hybrid MBR is present for the boot partition to allow booting on the Raspberry Pi 3B+.

The same installation boots properly with the 4.19 kernel (2020-05-27-raspios-buster-lite-armhf.img) until you do an apt update/full-upgrade which upgrades the kernel to 5.4.

The problem is equally reproducible on a Raspberry Pi 4 or a Raspberry Pi 3B+.

With the 4.19 kernel, booting completes successfully.

With the 5.4 kernel, the appropriate start.elf is found, loaded, and executed without error, but shortly thereafter, booting halts with the green led repeatedly blinking 7 times.

The same installation on USB devices boots properly with either the 4.19 kernel or the 5.4 kernel.

Simply replacing the 5.4 start.elf files with these 4.19 start.elf files makes the GPT SD card boot properly:

bootcode.bin      52,480   06/06/2020 22:38
fixup.dat          6,746   06/06/2020 22:38
fixup4.dat         5,405   06/06/2020 22:38
fixup4cd.dat       3,146   06/06/2020 22:38
fixup4db.dat       8,417   06/06/2020 22:38
fixup4x.dat        8,419   06/06/2020 22:38
fixup_cd.dat       2,663   06/06/2020 22:38
fixup_db.dat       9,820   06/06/2020 22:38
fixup_x.dat        9,818   06/06/2020 22:38
start.elf      2,884,708   06/06/2020 22:38
start4.elf     2,272,992   06/06/2020 22:38
start4cd.elf     816,124   06/06/2020 22:38
start4db.elf   3,774,532   06/06/2020 22:38
start4x.elf    3,031,652   06/06/2020 22:38
start_cd.elf     694,052   06/06/2020 22:38
start_db.elf   4,861,512   06/06/2020 22:38
start_x.elf    3,799,144   06/06/2020 22:38

It appears something got broken in the 5.4 start.elf files that only affects GPT SD card booting.

timg236, 2020-09-03:
Thanks, can you confirm that you just changed the kernel - please specify exact rpi-update command?

I missread your initial post about  06/06/2020 **working** which does actually line up with when SD/GPT support was added. It also looks like you are using regular GPT with both EFI boot and root inside that

reraikes, 2020-09-03:
5.4.42 boots fine.
5.4.51 does not.
There's a dozen or so commits in between.  I need to get to bed, but I can try to narrow it down to the exact one in the morning.
I'm simply doing 'rpi-update <hash>' at the moment trying to find the breaking point.

GPT w/Hybrid MBR has been working (booting) on SD cards for at least a year (that's when I first tried it).

timg236, 2020-09-03:
Ok, thanks for the info, if you can specify the hash that would be useful. 

This is booting for me from SD card on Pi4. I created a GPT partition table on the SD card then used dd to copy the partitions from the Raspberry Pi OS image and fixed up the PARTUUID in cmdline.txt and config.txt

pi@raspberrypi:~$ sudo uname -a
Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
pi@raspberrypi:~$ sudo fdisk -l /dev/mmcblk0
Disk /dev/mmcblk0: 14.4 GiB, 15476981760 bytes, 30228480 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 10DF60F9-4F6D-4A26-8C70-23D4E70911E0

Device          Start      End  Sectors  Size Type
/dev/mmcblk0p1   2048   821247   819200  400M Microsoft basic data
/dev/mmcblk0p2 821248 30226431 29405184   14G Linux filesystem
pi@raspberrypi:~$ vcgencmd version
Aug 19 2020 17:38:16 
Copyright (c) 2012 Broadcom
version e90cba19a98a0d1f2ef086b9cafcbca00778f094 (clean) (release) (start)

reraikes, 2020-09-03:
I'm still searching.  I'm doing this on a RPi 4 also.

5.4.49 is also bad.

timg236, 2020-09-03:
I would have thought 5.4.47 would be fine, so this would appear to be related to the Hybrid MBR since GPT boots ok

reraikes, 2020-09-03:
5.4.47 (666a0aee1d782fa51069ba6921fc2bd9e740cda1) doesn't run (appears to crash), but I see there is a commit right above it stating 'firmware: Fix Pi4 regression in previous build'.  I'm getting closer.

Hybrid MBR needs to be acceptable (even if not required on a RPi 4) as it's necessary for a Raspberry Pi 3B+ to boot GPT.

timg236, 2020-09-03:
It would be useful to see the output of gdisk -l to verify the partition types

reraikes, 2020-09-03:
Ok, the 'fixed' 5.4.47 (f794477c7c48061a720d6b9331d27f8d0300009c) is good, but 5.4.49 (da3752a358a86014cdcce5fc3be5b18d7ec074c4) is not.  There's one commit in between that wasn't assigned a 5.4.xx that I haven't tested.

The sun is about to come up here, so I really need to bail for a few hours.

Here's your gdisk -l:

root@raspberrypi:~# gdisk -l /dev/mmcblk0
GPT fdisk (gdisk) version 1.0.3

Partition table scan:
  MBR: hybrid
  BSD: not present
  APM: not present
  GPT: present

Found valid GPT with hybrid MBR; using GPT.
Disk /dev/mmcblk0: 60751872 sectors, 29.0 GiB
Sector size (logical/physical): 512/512 bytes
Disk identifier (GUID): 3864BC3D-2CE2-4EDA-821A-C21C60699167
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 33
First usable sector is 34, last usable sector is 60751838
Partitions will be aligned on 2048-sector boundaries
Total free space is 18284477 sectors (8.7 GiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1            8192          532479   256.0 MiB   0700  Microsoft basic data
   2          532480        21503999   10.0 GiB    8300  Linux filesystem

reraikes, 2020-09-03:
The last 5.4.47 (634e380a4d041492f859712bd2c81112a535b515) boots fine.
5.4.49 (da3752a358a86014cdcce5fc3be5b18d7ec074c4) does not, so it's the culprit.
I'll leave the detective work with you from here.

Back in about 8 hours...

pelwell, 2020-09-03:
Well, the 5.4.49 release includes the commit titled
`filesystem: Add support for SD cards with GPT partition table`, which looks to be a likely candidate.

timg236, 2020-09-03:
I was able to reproduce the issue after faffing around with gdisk and it's as I suspected here https://github.com/raspberrypi/firmware/issues/1465#issuecomment-686312228

Hybrid MBR has never been supported, it's worked by luck because the start.elf SD card file-system used to ignore GPT and skip to the next partition. It now switches to GPT, sees no boot partition and stops.

It should be possible to add explicit support for Hybrid MBR by failing over to the next primary record if nothing was found in the GPT. That code is pretty ancient and grim so it's going to be a bit fiddly.

timg236, 2020-09-03:
I've shared a zip of the start4.elf firmware with the experimental fix which resolved the problem that I observed. 
https://drive.google.com/drive/folders/15S6j0h7TM6GZ-KHg7woROhh3DrlI5e-C?usp=sharing

popcornmix, 2020-09-03:
The fix should also now be in latest rpi-update firmware

reraikes, 2020-09-03:
I just updated to 5.4.61 using rpi-update and the RPi 4 booted right up.  I moved the SD card to an RPi 3 and it also booted.  I want to do some more exhaustive testing, including on USB devices, before I open the champagne, but all initial indications are  it's healthy.

Thank you Tim and all for the quick fix.

timg236, 2020-09-03:
That was quick, thanks @popcornmix and @reraikes 

USB *should* work but due to evolution of the code there are multiple places which decode the MBR depending on the stage of the boot and boot medium so I'd keep the champagne on ice for now, although I think this small victory deserves a beer.

reraikes, 2020-10-12:
This problem is back (even in 5.4.61).  Replacing the 5.4 start.elf files with the 4.19 start.elf files (as described in the initial report) makes the GPT SD card boot properly.

uSD voltage 1.8V
GLOBAL_RESET: 01 0

PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:c305221a DATE: Sep  3 2020 TIME: 13:11:46 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1599135103 0xcbb4b4cf 0x00c03111
uSD voltage 3.3V
Initialising SDRAM 'Micron' 16Gb x2 total-size: 32 Gbit 3200
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Reset USB port-power 1000 ms
Boot mode: USB-MSD (04) order f1
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
USB3 rport 2 status 00000000 -> 000002a0
USB3 rport 3 status 00000000 -> 000002a0
USB3 rport 4 status 00000000 -> 000002a0
USB3 rport 5 status 00000000 -> 000002a0
USB2 rport 1 status 00000000 -> 400002e1
USB2 rport 1 status 400002f1 -> 40200e03
root HUB port 1 init
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
USB3 rport 3 status 000002a0 -> 00021203
root HUB port 3 init
DEV [02:00] 3.16 000000:03 class 9 VID 0bda PID 0411
HUB init [02:00] 3.16 000000:03
HUB [01:00] 2.16 000000:01 init port 2 speed 3
DEV [03:01] 2.16 000002:01 class 9 VID 0bda PID 5411
HUB init [03:01] 2.16 000002:01
HUB [01:00] 2.16 000000:01 init port 4 speed 3
DEV [04:01] 2.00 000004:01 class 9 VID 1a40 PID 0201
HUB init [04:01] 2.00 000004:01
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USB MSD timed out after 20 seconds
Boot mode: SD (01) order f
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
CID: 00275048534433324730da4c6a2c00fa
CSD: 400e00325b590000e7bf7f800a400000
SD: bus-width: 4 spec: 2 SCR: 0x02358002 0x01000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00000001,    8191 type: 0xee
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1 r-dir 2 r-sec 0
PM_RSTS: 0x00001000
Partition: 0
lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1 r-dir 2 r-sec 0
Read config.txt bytes     1852 hnd 0x0000038d hash 'b2d26265a688bac2'
recover4.elf not found (6)
recovery.elf not found (6)
Read start4.elf bytes  2283936 hnd 0x00012235 hash '3da0c1afcaebe124'
Read fixup4.dat bytes     5422 hnd 0x0001aa51 hash 'fdfbb390f4a2c93c'
0x00c03111 0x00000000 0x0000001f
MEM GPU: 32 ARM: 992 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0

MESS:00:00:25.702026:0: arasan: arasan_emmc_open
MESS:00:00:26.012257:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
MESS:00:00:26.120540:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
[... 51 lines cut ...]
MESS:00:00:28.545139:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:28.653063:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:28.693004:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:28.718501:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:28.723719:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:28.831643:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:28.870994:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:28.896484:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:28.901698:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:29.009622:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:29.049377:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:29.074884:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:29.080101:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:29.188025:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:29.227377:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:29.252880:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:29.258094:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:29.366018:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:29.405551:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:29.431056:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:29.436276:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:29.544196:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:29.583952:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:29.609441:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:29.614598:0: No compatible kernel found

sallevan, 2021-04-06:
Very strange, but it didn't work for me even with latest firmware.
Secondary stage loader chokes up with my SD card with **any** placement of EFI GPT placeholder (partition number 0xEE) unless I remove it completely.

`Read start4.elf bytes  2228800 hnd 0x00001e0c hash 'dce1545e7133835b'
Read fixup4.dat bytes     5448 hnd 0x0000008c hash '4109cea6ad32e5fe'
0x00c03111 0x00000000 0x000000ff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0

MESS:00:00:05.297358:0: arasan: arasan_emmc_open
MESS:00:00:05.626163:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
MESS:00:00:05.734445:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
MESS:00:00:05.786798:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_WRITE_PROTECTED before we're ready
MESS:00:00:05.814716:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
MESS:00:00:05.914353:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:06.019510:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:06.068891:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:06.093670:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:06.098890:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:06.206811:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:06.256374:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:06.281152:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:06.748855:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:06.756145:0: *** Restart logging
MESS:00:00:06.757647:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:06.865562:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:06.915023:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:06.939803:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:06.948398:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:06.991089:0: HDMI0: hdmi_pixel_encoding: 300000000
MESS:00:00:06.993718:0: HDMI1: hdmi_pixel_encoding: 300000000
MESS:00:00:07.004109:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:07.006208:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:07.114143:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:07.163546:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:07.188565:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:07.193761:0: Failed to load Device Tree file '?'
MESS:00:00:07.199154:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:07.307048:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:07.356451:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:07.381464:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:07.386666:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:07.393658:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:07.500916:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:07.550478:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:07.575497:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:07.580794:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:07.688698:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:07.738264:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:07.763334:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:07.768630:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:07.876536:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:07.925947:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:07.950960:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:07.956253:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.064162:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.113579:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:08.138586:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.143881:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.251788:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.301382:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:08.326401:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:08.331692:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
[... cut ...]
MESS:00:00:09.265406:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:09.270705:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:09.378607:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:09.427974:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:09.452981:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:09.458207:0: No compatible kernel found
`

sallevan, 2021-04-06:
> Very strange, but it didn't work for me even with latest firmware.
> Secondary stage loader chokes up with my SD card with **any** placement of EFI GPT placeholder (partition number 0xEE)
> unless I remove it completely.

I'm sorry, I was wrong.

For the very-very long time, one has to choose where to place EFI placeholder from two choices: partition type 0xEE has to be either first (Linux/BSD standard) or last (Windows standard, but fine to Linux/BSD too).

Secondary stage loader chokes if you will proceed with either of two choices above. Partition type 0xEE has to be placed second.

Why?

sallevan, 2021-04-06:
> Partition type 0xEE has to be placed second.

For openSUSE v15.3 Beta (Linux kernel v5.3.18) it is fine while Windows does not recognise SD card as GPT-partitioned. :(

timg236, 2021-09-27:
I've attached a possible fix for this, if you can very that this works for you then we can try to get this into rpi-update. The problem is that Hybrid / GPT is not really supported by the old firmware-file-system. It just relied on things being not recognized. Improving support for native GPT caused it to always match ANY GPT partition if present.
[hybrid-gpt-experimental-fix-1465.zip](https://github.com/raspberrypi/firmware/files/7235660/hybrid-gpt-experimental-fix-1465.zip)

reraikes, 2021-09-27:
As noted in my comment above (https://github.com/raspberrypi/firmware/issues/1465#issuecomment-686665226), this problem was fixed in 5.4.61 for me.  I just tested hybrid-gpt-experimental-fix-1465.zip and it doesn't appear to break anything here.

sallevan, 2021-09-27:
> I've attached a possible fix for this, if you can very that this works for you then we can try to get this into rpi-update.

**U r a man!** It works!

Thank you very much, @timg236, now I am able to boot Raspberry Pi 4B with "Windows-style" hybrid MBR/GPT partitioning scheme.

sallevan, 2021-09-27:
> As noted in my comment above ([#1465 (comment)](https://github.com/raspberrypi/firmware/issues/1465#issuecomment-686665226)), this problem was fixed in 5.4.61 for me.

Could you please publish both of your partitioning schemes -- GPT and PMBR?

Because in my case @timg236's fix really has fixed what was broken.

timg236, 2021-09-27:
> > I've attached a possible fix for this, if you can very that this works for you then we can try to get this into rpi-update.
> 
> **U r a man!** It works!
> 
> Thank you very much, @timg236, now I am able to boot Raspberry Pi 4B with "Windows-style" hybrid MBR/GPT partitioning scheme.

Glad to hear it. It should be officially available in the next rpi-update release. Please shout if somethings goes wrong!

reraikes, 2021-09-27:
@sallevan

>Could you please publish both of your partitioning schemes -- GPT and PMBR?
>Because in my case @timg236's fix really has fixed what was broken.

root@raspberrypi:~# fdisk /dev/sda

Welcome to fdisk (util-linux 2.33.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

A hybrid GPT was detected. You have to sync the hybrid MBR manually (expert command 'M').

Command (m for help): p

Disk /dev/sda: 29.9 GiB, 32080200192 bytes, 62656641 sectors
Disk model: Flash Drive
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 7BAD5352-3601-40DB-9308-AA6E9FC1BCD1

Device      Start      End  Sectors  Size Type
/dev/sda1    8192   532479   524288  256M Microsoft basic data
/dev/sda2  532480 62656607 62124128 29.6G Linux filesystem

Command (m for help): M
Entering protective/hybrid MBR disklabel.

Command (m for help): p
Disk /dev/sda: 29.9 GiB, 32080200192 bytes, 62656641 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device     Boot Start    End Sectors  Size Id Type
/dev/sda1        8192 532479  524288  256M  c W95 FAT32 (LBA)
/dev/sda2           1   8191    8191    4M ee GPT

Partition table entries are not in disk order.

sallevan, 2021-09-27:
@reraikes, :)))

You have SD card with your own, @reraikes-standard-hybrid partition layout. ;)

To be compatible with UEFI specification to allow your SD card to be read properly with UEFI payload, for example, GRUB version 2 (has been loaded by U-Boot/aarch64), you have to:

1. have your EFI system partition identifying itself with correct GPT UUID -- "Microsoft basic data" is wrong, "EFI system" is right;
2. have 0xEE entry in your PMBR (GPT partition) **covering not only gap between primary GPT and EFI system partition, but the latter also!**

That was mandated by standard, and _start4.elf_ obviously rejected all your GPT information because it is non-standard, and you're luckily booting your Pi.

reraikes, 2021-09-27:
@sallevan

My GPT approach has been in use for many years on the Raspberry Pi 3B and 4B with Raspberry Pi OS.  The only target was Raspberry Pi OS and the limited boot capability of the Raspberry Pi 3B.  No support of EFI/UEFI/GRUB was intended.  With the exception of the short-lived problem introduced in 5.4.49 and fixed in 5.4.61, it has booted and run perfectly with both 32- and 64-bit versions of Raspberry Pi OS.

reraikes, 2021-09-27:
@sallevan

> You have SD card with your own, @reraikes-standard-hybrid partition layout. ;)
>
>To be compatible with UEFI specification to allow your SD card to be read properly with UEFI payload, for example, GRUB version 2 (has been loaded by U-Boot/aarch64), you have to:
>
>have your EFI system partition identifying itself with correct GPT UUID -- "Microsoft basic data" is wrong, "EFI system" is right;
>have 0xEE entry in your PMBR (GPT partition) covering not only gap between primary GPT and EFI system partition, but the latter also!
>
>That was mandated by standard, and start4.elf obviously rejected all your GPT information because it is non-standard, and you're luckily booting your Pi.

Upon further review, I find I'm not creating a 'reraikes-standard-hybrid partition layout'.  Both items you noted are simply what gdisk creates by default.  I find no evidence that either a Raspberry Pi 3 or Raspberry Pi 4 bootloader/start4.elf is rejecting anything.