raspberrypi/firmware #1467: Unable to boot with latest firmware on RPI4

Nopel2020, 2020-09-15:
On a RPI4 with 4GB with the latest kernel (5.4.65-V8+) and the latest firmware (086d9d9) my RPI4 freezes on the rainbow screen.

The only files from the firmware I'm using are fixup4cd.dat and start4cd.elf. After replacing just these two files with the ones from the latest commit before that (a490197), the RPI4 is booting normally again.

With the latest firmware files, the freeze is reproduceable.

* Which model of Raspberry Pi?
RPI4, 4GB

* Which OS and version (`cat /etc/rpi-issue`)?
64bit Debian Buster with self-compiled 64bit 5.4.65 kernel (but I don't think this is relevant, the Pi freezes before it loads anything from the OS)

* Which firmware version (`vcgencmd version`)?
Sep 11 2020 14:23:56 
Copyright (c) 2012 Broadcom
version ecd571f7947fb751a266d8591b0573d4d975fbe6 (clean) (release) (start_cd)

pelwell, 2020-09-15:
> I don't think this is relevant, the Pi freezes before it loads anything from the OS

What makes you say that? Are you capturing the output from the firmware over the UART?

Nopel2020, 2020-09-15:
Because it freezes on the rainbow screen. Does it load anything from the OS while the rainbow screen is still visible?

pelwell, 2020-09-15:
Yes - it's hard for the kernel to display raspberries and start to output messages until its been loaded.

pelwell, 2020-09-15:
Do you have any custom settings in config.txt?

mikey1963, 2020-09-15:
Mine frozen too, latest rpi-update, running 64bit kernel.

---
Regards, Mr Michael A Smith
My apologies in advance for any errors or inaccuracies, mobile email and my brain are both prone to them.

Nopel2020, 2020-09-15:
Ah ok, sorry, then I'm not sure if it's loading something from the OS.

This is my full config.txt:
gpu_mem=16
hdmi_ignore_cec=1
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=4
initramfs initrd8.img followkernel

pelwell, 2020-09-15:
@mikey1963 are you also an initramfs user?

mikey1963, 2020-09-15:
No,
Only changes to standard are;

gpu_mem=256
dtoverlay=i2c-rtc,ds3231
enable_uart=1

arm_64bit=1

Did the rpi-update this evening and it hangs on boot.  I’m headless so can’t give you much more info, sorry.

---
Regards, Mr Michael A Smith
My apologies in advance for any errors or inaccuracies, mobile email and my brain are both prone to them.

pelwell, 2020-09-15:
Oh well - we'll investigate tomorrow.

Nopel2020, 2020-09-15:
Using the next-to-last firmware files is a workaround, so no rush. Thanks for looking into this!

pelwell, 2020-09-15:
I have to - that commit has my fingerprints all over it.

mikey1963, 2020-09-15:
That’s what makes it fun :)

---
Regards, Mr Michael A Smith
My apologies in advance for any errors or inaccuracies, mobile email and my brain are both prone to them.

Nopel2020, 2020-09-15:
FWIW, I'm running the exact same setup on an RPI3B, and almost identical (but 32bit) on an old RPI1 and some RPI0Ws. They are all booting fine, only the RPI4 is affected.

pelwell, 2020-09-15:
If it's the change I think it is then the code path is only visited when arm_64bit=1, but the 64-bit 3B non-failure is an interesting data point.

yvoy, 2020-09-16:
Same issue on RPI4 , need to temporaly disable arm_64bit=1

pelwell, 2020-09-16:
Problem identified - sorry folks. Luckily, in the process of fixing this I spotted a corner case that also wouldn't have worked. I'll comment again when the fix is available.

popcornmix, 2020-09-16:
Fix should be in latest rpi-update firmware - please test and report if issue is resolved.

pelwell, 2020-09-16:
Thanks, @popcornmix.

by, 2020-09-16:
> Fix should be in latest rpi-update firmware - please test and report if issue is resolved.

Confirmed: it does work now.

mikey1963, 2020-09-16:
Confirmed now back fully working, great work guys.  Thanks.

—-
Regards, Mr Michael A Smith.
My apologies in advance for any errors or inaccuracies, mobile email and my brain are both prone to them.

pelwell, 2020-09-16:
If you're feeling brave you can now gzip your kernel8.img:
```
$ sudo gzip -k /boot/kernel8.img
$ sudo mv /boot/kernel8.img{,.orig}
$ sudo mv /boot/kernel8.img{.gz,}
```

tomty89, 2020-09-16:
@pelwell It still fails to load u-boot in ArchLinuxARM (aarch64)

vbauerster, 2020-09-16:
Can confirm that RPI3B+ affected as well.

pelwell, 2020-09-16:
> Can confirm that RPI3B+ affected as well.

With U-boot or a Linux kernel?

vbauerster, 2020-09-16:
There is just a rainbow screen no any kernel boot messages, that's all I know.

Nopel2020, 2020-09-16:
Original poster here. I can confirm that it's working on my RPI4 (and also on my RPI3B, (not 3B+)). Thanks!

pelwell, 2020-09-16:
@vbauerster When did you install the update?

pelwell, 2020-09-16:
@vbauerster A few more questions:

2. Are you using a Raspberry Pi OS image?
3. Have you customised config.txt in any way?
4. If you can put the SD card in another device, how large (in bytes) is start.elf?

pelwell, 2020-09-16:
@tomty89 Are you definitely running the latest firmware? What does `strings start.elf | grep VC_BUILD` show?

What do you have in config.txt? (You ignore comments and blank lines)

graysky2, 2020-09-16:
@pelwell - The [raspberrypi-firmware](https://github.com/archlinuxarm/PKGBUILDs/blob/82538eeee30308ecb3944a02ed44eb3454065c66/alarm/raspberrypi-firmware/PKGBUILD) package for Arch ARM is currently supplying https://github.com/raspberrypi/firmware/commit/525f1eb4d264777323c2489d8a1b753fd77f237a

It is my understanding that the only time it needs to be updated is if there are changes to anything under https://github.com/raspberrypi/firmware/tree/master/hardfp/opt/vc ... Please correct me if I am mistaken.

pelwell, 2020-09-16:
That's correct - if it works for you, stick with it. Only the brave and the foolish take every update that comes along.

graysky2, 2020-09-16:
For Arch ARCH as of today:
* [raspberrypi-firmware](https://github.com/archlinuxarm/PKGBUILDs/blob/82538eeee30308ecb3944a02ed44eb3454065c66/alarm/raspberrypi-firmware/PKGBUILD) is currently supplying https://github.com/raspberrypi/firmware/commit/525f1eb4d264777323c2489d8a1b753fd77f237a
* [raspberrypi-bootloader](https://github.com/archlinuxarm/PKGBUILDs/blob/b8249f646833a1bca97c638d86df1d771312ba21/alarm/raspberrypi-bootloader/PKGBUILD) is currently supplying https://github.com/raspberrypi/firmware/commit/28f031eb3a06b94ce9219973fad72e4a647bb690

On Arch ARM armv7h, RPi4 boots fine.
On Arch ARM aarch64, RPi4 does NOT boot.  I too only see the rainbow screen which does not go away.

```
% strings /boot/start.elf | grep VC_BUILD
VC_BUILD_ID_USER: dom
VC_BUILD_ID_TIME: 11:12:26
VC_BUILD_ID_VARIANT: start
VC_BUILD_ID_TIME: Sep 16 2020
VC_BUILD_ID_BRANCH: bcm2711_2
VC_BUILD_ID_HOSTNAME: buildbot
VC_BUILD_ID_PLATFORM: raspberrypi_linux
VC_BUILD_ID_VERSION: 0368c5cfa664e0534e5428dc45ac2a972802159c (clean)
```

pelwell, 2020-09-16:
What does Arch put in its config.txt? Until we understand why you and some others are still affected I suggest you revert your raspberrypi-bootloader package to https://github.com/raspberrypi/firmware/commit/a490197f0672d948860b2b807884ae65eabc4d4f.

thisisnotev, 2020-09-16:
On a fresh install just "enable_uart=1". 
A user on the forum is reporting that it hangs after a message from systemd saying "A start job is running for /dev/mmcblk0".

pelwell, 2020-09-16:
@thisisnotev A fresh install of what?

thisisnotev, 2020-09-16:
@pelwell ArchLinux ARM

graysky2, 2020-09-16:
> What does Arch put in its config.txt?

@pelwell - To be clear, Arch ARM offers an armv7h (32-bit) and an aarch64 (64-bit) option for RPi4.

For armv7h (which currently works), [config.txt](https://github.com/archlinuxarm/PKGBUILDs/blob/master/core/linux-raspberrypi4/config.txt)
For aarch64 (which currently does not boot), as @thisisnotev mentioned, simply `enable_uart=1`

graysky2, 2020-09-16:
> Until we understand why you and some others are still affected I suggest you revert your raspberrypi-bootloader package to a490197.

For Arch ARM users of aarch64, the corresponding package is [raspberrypi-bootloader-20200911-1](http://tardis.tiny-vps.com/aarm/packages/r/raspberrypi-bootloader/raspberrypi-bootloader-20200911-1-any.pkg.tar.xz
) which boots just fine.

pelwell, 2020-09-16:
What does Arch call its 64-bit kernel image?
Does including `arm_64bit=1` in config.txt make it work?

tomty89, 2020-09-17:
@pelwell Image and Image.gz (the former is used AFAIK). kernel8.img is u-boot.

vbauerster, 2020-09-17:
@pelwell 
> When did you install the update?

Yesterday.
> Are you using a Raspberry Pi OS image?

No, I use Arch linux.
> Have you customised config.txt in any way?

No

> If you can put the SD card in another device, how large (in bytes) is start.elf?
I already restored from backup, so cannot check.

vbauerster, 2020-09-17:
Still "bricking" device with bootloader 20200916-1
[image]

pelwell, 2020-09-17:
There's clearly something different about Arch that is triggering the boot failure. I'm installing the 64-bit Arch to test it.

pelwell, 2020-09-17:
I think I've got this: Arch uses U-boot as its kernel8.img, and U-boot doesn't include the ARM64 header used by the Linux kernel. An unintentional result of this was to change the load address of the image, which U-boot doesn't like - it's either not position-independent code or it ended up clashing with other resources. The fix is in the internal repo and will be in the next rpi-update release. In the meantime you can work around the issue (for testing purposes) by adding `kernel_address=0x80000` to config.txt.

ncguk, 2020-09-17:
> I think I've got this: Arch uses U-boot as its kernel8.img, and U-boot doesn't include the ARM64 header used by the Linux kernel. An unintentional result of this was to change the load address of [...]

I can confirm this workaround enables a previously non-booting aarch64 install to boot (Raspberry Pi 4 1GB RAM).

popcornmix, 2020-09-17:
Latest rpi-update firmware has the fix for kernel_address.

graysky2, 2020-09-17:
@pelwell @popcornmix - Thank you both for for the engagement to detect and fix this.  I can confirm that https://github.com/raspberrypi/firmware/commit/7b99da75f55a5ad7d572ec4ebe4e8f9573deaee7 fixes the issue; I just updated Arch ARM's [bootloader package](https://github.com/archlinuxarm/PKGBUILDs/commit/35d608d4e250a783c4193990bc8f0b276ee62fda).  Safe to close this ticket from my prespective.

pelwell, 2020-09-17:
Thanks - I don't plan to make any further changes in this area for a while...

closed 2020-09-17 by pelwell