raspberrypi/firmware #1468: Raspberry pi 4 8GB can't boot using usb, recover4.elf and recovery.elf not found

darksponge, 2020-09-15:
Hello,
I'm having trouble booting my raspberry pi 4. I've installed raspberry os recently on a ssd drive and everything was working fine, but last week it suddenly stopped booting. I can see something on my screen for one or two seconds, with these two errors:  
recover4.elf not found (6)
recovery.elf not found (6)
and then everything seems to crash and my screen turns black.

Which model of Raspberry Pi?
Raspberry pi 4 model B 8gb ram
Which OS and version?
Which firmware version (`vcgencmd version`)?
Which kernel version (`uname -a`)?

^^ I don't know how to access these information since I can't boot I have no idea how to find out.

I've tried to recover the eeprom, I got a green screen but it is still not working, I've tried with another usb stick, didn't work, with an sd card, didn't work either. I couldn't find anything about the two files mentioned above, so I don't really understand where the problem come from.

What could cause this issue ?
Thank you very much for your help !
[image]

timg236, 2020-09-15:
The significant error is the XHCI cmd error which might indicate a power issue e.g. try connecting to the SSD via a powered hub.

The default bootloader in the Raspberry Pi Imager 2020-09-03 but I wouldn't have thought that would make much difference.

darksponge, 2020-09-16:
I'm already using a powered hub, so maybe it's not working correctly but if that's the case I don't get why it cannot boot on the sd card either. Could it be the power cable ?

andrum99, 2020-09-17:
> I'm already using a powered hub, so maybe it's not working correctly but if that's the case I don't get why it cannot boot on the sd card either. Could it be the power cable ?

Try the SSD directly into the Pi, with no other USB devices connected. Also, upgrade the bootloader firmware to the latest version. You can also try a different USB hub, if you have one available.

I take it you've tried the obvious, i.e. connecting the SSD to another computer?

mikey1963, 2020-09-17:
I have an ssd connected via a powered hub with a 64 bit kernel on a pi 4 8gb latest rpi-update and boot rom booting nicely.

---
Regards, Mr Michael A Smith
My apologies in advance for any errors or inaccuracies, mobile email and my brain are both prone to them.

darksponge, 2020-09-21:
> Try the SSD directly into the Pi, with no other USB devices connected. 

I tried but since the did an eeprom recovery it shouldn't be able to boot on usb anymore. I've tried to boot on an sd card but it didin't work.
> Also, upgrade the bootloader firmware to the latest version. You can also try a different USB hub, if you have one available.

How can I upgrade the bootloader if I can't boot the pi ?
> I take it you've tried the obvious, i.e. connecting the SSD to another computer?

I have done this and I can see all the files without any issue
> The messages about the bootloader not finding recovery4.elf and recovery.elf are entirely normal - it will only find those when you are using the EEPROM recovery mode.

Ok thank you.
I've also tried another power cable, but made no difference.

darksponge, 2020-09-23:
Thanks for your help but since I really need my raspberry pi for a project I sent it back for inspection, so I think that we can close this issue since there probably won't be anything new.

closed 2020-09-24 by darksponge

Hoggaan, 2020-12-30:
Can you identify the problem right now as I am facing the same problem?

darksponge, 2020-12-30:
> Can you identify the problem right now as I am facing the same problem?

Hello,
I'm sorry, I couldnt fix that problem. I brought it back to store and they gave me a new one.
It's a pity that I couldnt understand what was wrong, maybe someone has the answer elsewhere...

Ph4Lc0N, 2025-02-01:
I had the same problem. My Pi di not want to boot and I had been using it for many years before and everything was right.

I FIXED it...!

I took the SD card off and mounted it on my Mac OS. I found the file _config.txt_ and since I had previously Overclocked the Pi, I edited the file and commented the lines for the overclock. I then booted the Pi again and everything worked perfectly.

I found out that the Pi does not like to be overclocked beyond 2100 Mhz and 6 volts.

After rebooting the Pi I edited the config.txt file again and overclocked it like this and everything works perfectly now.

Here are my settings:

over_voltage=6
arm_boost=1
arm_freq=2100
gpu_freq=750
force_turbo=1