raspberrypi/firmware #1472: Pi 4: Netconsole set but no udp packet seen

RamaSpaceShip, 2020-09-21:
**Describe the bug**
I have a P4 4GB with RaspiOS Desktop on it. USB storage is via a X835 board and a Sandisk SSD.
I can boot with /boot on the SD card and / on the SSD, but full USB boot doesn't work. :(

To investigate this issue, I tried to use netconsole, but I don't get any UDP message with Wireshark when booting without the SD card. :(

My bootloader settings are:
root@raspberrypi:~# vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41

# Enable debug if GPIO 7 is pulled low
[gpio7=0]
NETCONSOLE=6666@192.168.1.14/,6666@192.168.1.2/

Before rebooting without the SD card, I verified the gpio 7:
root@raspberrypi:~# pigs r 7
0

But Wireshark on the machine at address 192.168.1.2 does not receive any UDP packet from 192.168.1.14.

The firmware is:
root@raspberrypi:~# rpi-eeprom-update 
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: jeudi 3 septembre 2020, 12:11:43 (UTC+0000) (1599135103)
 LATEST: jeudi 3 septembre 2020, 12:11:43 (UTC+0000) (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1

**To reproduce**
Set the netconsole as indicated above
check if messages are received on the target machine

**Expected behaviour**
I should see boot messages on UDP port 6666 using Wireshark on the target machine

**Actual behaviour**
No message is received

**System**
System Information
------------------

Raspberry Pi 4 Model B Rev 1.2
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage4

Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
Revision        : c03112
Serial          : 100000006baa8fa6
Model           : Raspberry Pi 4 Model B Rev 1.2
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0
[... 212 lines cut ...]
008668.618: dtparam: audio=on
008677.833: brfs: File read: 1784 bytes
008685.172: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d.dtbo
008701.737: Loaded overlay 'vc4-fkms-v3d'
008744.735: brfs: File read: 1238 bytes
008746.034: brfs: File read: /mfs/sd/cmdline.txt
008746.081: Read command line from file 'cmdline.txt':
008746.094: 'console=serial0,115200 console=tty1 root=PARTUUID=03927c2a-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles'
009916.628: brfs: File read: 166 bytes
010415.165: brfs: File read: /mfs/sd/kernel7l.img
010415.204: Loading 'kernel7l.img' to 0x8000 size 0x5b7fc0
010415.251: Device tree loaded to 0x2eff4100 (size 0xbedd)
010419.302: bfs_xhci_stop
010419.314: XHCI-STOP
010419.494: xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
013465.474: vchiq_core: vchiq_init_state: slot_zero = 0xded80000, is_master = 1
013468.946: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
013476.239: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000023
019803.206: TV service:host side not connected, dropping notification 0x00000008, 0x00000004, 0x00000000

**Logs**
none as it is before the kernel is booted

**Additional context**
none

pelwell, 2020-09-21:
NETCONSOLE is a configuration option for the BOOT EEPROM, not config.txt. See https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md and https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md.

timg236, 2020-09-21:
Config keys values are limited to 32 characters,  surprisingly nobody has hit this before since it's been like that since day one!

The default source port is 6665 and default target port is 6666 so it's possible to get the same result with this slightly shorter string.

NETCONSOLE=6666@192.168.1.14/eth0,@192.168.1.2/

RamaSpaceShip, 2020-09-21:
I changed the netconsole string in the bootloader config to the shortest I can:
NETCONSOLE=@192.168.1.8/,192.168.1.2/
But I still didn't get any UDP message.

Did you try the netconsole option, and thus do you have a bootloader config where the netconsole is working?
If so, I would like to know how different it is from mine.

timg236, 2020-09-21:
I used the following config with the debug Pi connected directly to another Pi running WireShark,

NETCONSOLE=6666@192.168.1.14/,@192.168.1.2/

RamaSpaceShip, 2020-09-21:
This is now solved

closed 2020-09-21 by RamaSpaceShip

timg236, 2020-09-22:
Great, thanks for confirming the workaround. The config parser is fixed and will appear in a future beta/stable release but I'll just document it as a limitation for now.