raspberrypi/firmware #1473: Boot issue with X835: boot when connected on USB2 but not on USB3

RamaSpaceShip, 2020-09-22:
**Describe the bug**
I have a P4 4GB with RaspiOS Desktop on it. USB storage is via a X835 board and a Sandisk SSD.
I can boot with /boot on the SD card and / on the SSD, but full USB boot on USB3 port doesn't work. :(
I tried using a USB2 port and it worked???

I setup netconsole to investigate and it boots on USB3 when netconsole is active.

**To reproduce**
Connect a X835 with an SSD on a USB3 port of the Pi4 and try to boot on it.

**Expected behaviour**
The system boots normally

**Actual behaviour**
It does not boot and wait for a bootable device

If netconsole is set, it boots but the network is broken

**System**
Raspberry Pi 4 Model B Rev 1.2
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage4

Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
Revision : c03112
Serial : 100000006baa8fa6
Model : Raspberry Pi 4 Model B Rev 1.2
Throttled flag : throttled=0x0
Camera : supported=0 detected=0
Videocore information

Sep 2 2020 21:13:13
Copyright (c) 2012 Broadcom
version 4439d2aaa6c376a2d1ef4402f142e1cf4de37c43 (clean) (release) (start)

alloc failures: 0
compactions: 0
legacy block fails: 0
Filesystem information

Sys. de fichiers blocs de 1K Utilisé Disponible Uti% Monté sur
/dev/root 30832068 6855768 22387076 24% /
devtmpfs 1827296 0 1827296 0% /dev
tmpfs 1959392 5120 1954272 1% /dev/shm
tmpfs 1959392 952 1958440 1% /run
tmpfs 5120 4 5116 1% /run/lock
tmpfs 1959392 0 1959392 0% /sys/fs/cgroup
/dev/sda1 294736 56032 238704 20% /boot
/dev/sda4 924971004 145904 877769408 1% /home
tmpfs 391876 8 391868 1% /run/user/1001
tmpfs 391876 0 391876 0% /run/user/1000

Filename Type Size Used Priority
/dev/sda3 partition 4194236 0 -2
/var/swap file 102396 0 -3
Package version information

raspberrypi-ui-mods:
Installé : 1.20200611
raspberrypi-sys-mods:
Installé : 20200812
openbox:
Installé : 3.6.1-8+rpt4
lxpanel:
Installé : 0.10.0-2+rpt12
[... 251 lines cut ...]
USB3 rport 5 status 00000000 -> 000002a0
USB2 rport 1 status 00000000 -> 400002e1
USB2 rport 1 status 400002f1 -> 40200e03
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
USB3 rport 3 status 000002b1 -> 00281203
DEV [02:00] 3.32 000000:03 class 9 VID 05e3 PID 0620
HUB init [02:00] 3.32 000000:03
DEV [03:01] 2.00 000001:01 class 0 VID 046d PID c016
DEV [04:02] 3.00 000001:03 class 0 VID 174c PID 0829
MSD device [04:02] 3.00 000001:03 conf 0 iface 0 ep 81#1024 02#1024
PM_RSTS: 0x00001000
MBR: 0x0000ffff,  589815 type: 0x0c
MBR: 0x0009fff6,62913600 type: 0x83
MBR: 0x03c9fc36, 8388480 type: 0x82
MBR: 0x0449fbb6,1881572426 type: 0x83
PM_RSTS: 0x00001000
DEV [05:01] 2.00 000003:01 class 0 VID 046d PID c318
DEV [06:01] 2.16 000002:01 class 9 VID 05e3 PID 0610
HUB init [06:01] 2.16 000002:01
DEV [07:02] 3.00 000002:03 class 0 VID 174c PID 0829
MSD device [07:02] 3.00 000002:03 conf 0 iface 0 ep 81#1024 02#1024

**Additional context**
none

timg236, 2020-09-22:
This should really be on the rpi-eeprom https://github.com/raspberrypi/rpi-eeprom

"Netconsole log (but it boots if netconsole is active)" - netconsole delays booting because it blocks until the Ethernet link is established (or a timeout occurs). This suggests that the X835/SSD is slow to initialise and times out. 

Increasing USB_MSD_DISCOVER_TIMEOUT to 30 seconds and USB_MSD_LUN_TIMEOUT to 30 seconds may help.

RamaSpaceShip, 2020-09-22:
I set the timeouts as suggested:
root@raspberrypi:~# vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41

USB_MSD_DISCOVER_TIMEOUT=30000
USB_MSD_LUN_TIMEOUT=30000

But it does not boot and I have the message on the screen:
USB MSD timed out after 30 seconds

It eventually booted after many retries. I cannot say precisely, but the last time I looked at the screen the 'Restart' count was 16, and about five minutes later the system was booted

pelwell, 2020-09-22:
How are you powering the SSD? What is the supply rated at?

timg236, 2020-09-22:
It might be a good idea to verify the SSD with a regular USB SATA adapter and HUB instead of the X835. Assuming you are referring to one of these
https://geekworm.com/products/raspberry-pi-4-x835-3-5-inch-sata-hdd-storage-expansion-board

RamaSpaceShip, 2020-09-22:
The power adapter is that one: https://geekworm.com/collections/raspberry-pi/products/dc-5v-4a-power-adapter-with-eu-us-plug-for-x820-x800-x4000-x6000-expansion-board

It is rated 5V 4A

RamaSpaceShip, 2020-09-22:
Sorry I made a mistake: the SATA board is the X829:
https://geekworm.com/products/for-raspberry-pi-4-x829-dual-2-5-sata-ssd-hdd-storage-expansion-board?_pos=2&_sid=c830af888&_ss=r

pelwell, 2020-09-22:
4A should be sufficient. I asked because intermittent faults at boot time are often power-related.

timg236, 2020-09-22:
XHCI_DEBUG=0x3 or XHCI_DEBUG=0xf with BOOT_UART=1 would give more information to identify when the device starts responding to requests

RamaSpaceShip, 2020-09-22:
I tried with another USB SATA adapter, and it booted immediatly on the first try.
On the second and third tries, it booted after one retry.

I will try with the XHCI debug settings.

RamaSpaceShip, 2020-09-22:
I tried with the following settings:
root@raspberrypi:~# vcgencmd bootloader_config
[all]
BOOT_UART=1
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41

XHCI_DEBUG=0xf

And it boots after one retry???

timg236, 2020-09-22:
It sounds rather timing depending, do you have the logs from the first retry or perhaps try less verbose XHCI_DEBUG e.g. 0x1 or 0x3.

RamaSpaceShip, 2020-10-05:
Sorry for the delay, it took me some time to get a cable to read the Pi serial console.

So with XHCI_DEBUG=0xf, it boots, see log file RPI-boot.log.
[RPI-boot.log](https://github.com/raspberrypi/firmware/files/5327379/RPI-boot.log)

So with XHCI_DEBUG=0x3, it does not boot, see log file RPI-noboot.log.
[RPI-noboot.log](https://github.com/raspberrypi/firmware/files/5327380/RPI-noboot.log)

Hope this helps.

timg236, 2020-10-16:
Thanks for the info. Just to let you know that I've got an X835 on order. If you get chance it would be useful to know if this bootloader is better/different. There are some improvements to diagnostics and it avoids som unnecessary timeouts

[rpi-eeprom-recovery.zip](https://github.com/raspberrypi/firmware/files/5393544/rpi-eeprom-recovery.zip)

890311b6c995ea0ac47bca1de8b3cf3675fbf14f511f4dd3a82ddbe302e724e2 (sha256)