raspberrypi/firmware #1475: program_usb_boot_mode=1 might brick any pi4

cleverca22, 2020-09-23:
**Describe the bug**
if `program_usb_boot_mode=1` still sets OTP17 bit29, as it did on previous models, it will mess with some unknown field (bits 8-10 and 28-30 of OTP17 are compared), causing OTP66 and OTP67 to not be loaded
due to 66/67 not being loaded, any attempt to boot from SPI or SD is disabled, i suspect usb-device boot may still function

**Expected behaviour**
`program_usb_boot_mode=1` should either do nothing on a pi4, or it should re-flash the bootconf.txt in the SPI for the user

**Additional context**
discovered the potential issue on https://www.raspberrypi.org/forums/viewtopic.php?p=1731266#p1731266 where a user may have already bricked their pi4 via this route
cross-referencing the old OTP docs (reg17 bit 29 is usb-host boot) and the disassembled pi4 maskrom, i can see how setting bit29 may cause major problems

due to the risk of bricking my pi4, i have not tested it on my own hardware

pelwell, 2020-09-23:
Thank you for your concern, but `program_usb_boot_mode` does nothing on a Pi 4.

closed 2020-09-23 by pelwell

cleverca22, 2020-09-23:
have you confirmed in the source that it is disabled, and/or tested that on an actual pi4?

pelwell, 2020-09-23:
Yes.

cleverca22, 2020-09-23:
yeah, digging into start4.elf, i can see the check
it looks at the processor code in the revision, and will only act on a BCM2837, and no other model
looks like my theory is a bust, and that user has some other issue

pelwell, 2020-09-23:
It's a shame you didn't check that before spreading the fear, uncertainty and doubt.

cleverca22, 2020-09-23:
i thought that somebody with proper access to source could check things faster then i could via decompiling, but the function turned out to be a lot simpler then i expected

pelwell, 2020-09-23:
But you left it to me to tell the user.

cleverca22, 2020-09-23:
was also updating that thread with info, as i confirmed it, and didnt notice that PhilE is also pelwell