raspberrypi/firmware #1476: Rotate HDMI display does not work in raspberrypi

arashsoftco, 2020-09-24:
i  tried to rotate my waveshare screen 180deg by editing config in boot 
```

pi@raspberrypi:~ $ vcgencmd get_config int
arm_freq=1500
audio_pwm_mode=514
config_hdmi_boost=5
core_freq=500
core_freq_min=200
disable_commandline_tags=2
disable_l2cache=1
display_hdmi_rotate=1
display_lcd_rotate=1
display_rotate=1
enable_gic=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=500
gpu_freq_min=250
init_uart_clock=0x2dc6c00
lcd_framerate=60
lcd_rotate=1
mask_gpu_interrupt0=1024
mask_gpu_interrupt1=0x10000
max_framebuffers=2
over_voltage_avs=-36250
pause_burst_frames=1
program_serial_random=1
total_mem=2048
hdmi_force_cec_address:0=65535
hdmi_force_cec_address:1=65535
hdmi_pixel_freq_limit:0=0x11e1a300
hdmi_pixel_freq_limit:1=0x11e1a300
```
i also tried 2 insted of 1 but nothing changed
display_hdmi_rotate=2
display_lcd_rotate=2
display_rotate=2

i guess the problem is Initial value is -1 !

cmitu, 2020-09-25:
If your system is a Pi4 (which looks like it, based on the `total_mem` reported), then `display_rotate` is not supported for it - see [this forum post](https://www.raspberrypi.org/forums/viewtopic.php?t=247619) for alternatives.

Swap-File, 2021-01-12:
I know rotation doesn't work on the Pi4, but is there any reason to not support flipping on the Pi4?

`display_rotate=0x10000` and  `display_rotate=0x20000` (Vertical and horizontal flipping) used to work on the RPI4.

It looks like this was disabled as part of this update: https://github.com/raspberrypi/firmware/commit/5cb5ddd076ac30b4f5a2af867b7c4f268c99728f

I've downgraded with the following command and it seems to work fine.
`sudo rpi-update f0236cc587a58532fbe1267c1db7a0dd7b22d01e`

JamesH65, 2021-01-12:
You can easily rotate and flip on the Pi4 - if using the desktop use the screen configuration utility. If console, use the kernel command line. See https://www.raspberrypi.org/documentation/configuration/display_rotation.md

Ruffio, 2021-01-12:
@JamesH65 will rotating/flipping by config.txt not the supported for Pi4+ in the future?

JamesH65, 2021-01-12:
Not the lcd/hdmi_rotate commands, no, I don't think so. Our aim is more standardised ways of doing things, moving away from custom stuff like config.txt entries.

Ruffio, 2021-01-12:
Ok, I thought that the standard way was to set it in config.txt - That is how it is done in all the other pies.

JamesH65, 2021-01-12:
By standard I mean conforming to the industry standard (Linux standard tools etc) not what we have used in the past.

Swap-File, 2021-01-12:
Using cmdline.txt to flip my display seems to work, but it seems to take effect late in the boot process.

Using config.txt with the older firmware works through the entire bootup process.

Is there any way to have the flip take effect earlier?  Or if I want that behavior do I just need to stay on the older firmware?

JamesH65, 2021-01-12:
Try leaving the config entries in as well. They may work until the FKMS driver takes over.

Swap-File, 2021-01-12:
Thanks for the suggestion, I tried that, but from what I can tell any firmware past https://github.com/raspberrypi/firmware/commit/5cb5ddd076ac30b4f5a2af867b7c4f268c99728f completely ignores `display_rotate` when using the KMS driver.

I think I'll just stay on the older firmware for now.  It's rather ugly having an incorrectly flipped boot screen.

JamesH65, 2021-01-12:
@popcornmix @6by9 Any thoughts?