raspberrypi/firmware #1477: Issue with Module8.symvers with kernel 5.4.69-v8+

MrEngman, 2020-10-03:
Compiling out of tree wifi drivers failed when compiling for 64bit kernel 5.4.69-v8+ #1344. Compiling for +, -v7+ and -v7l+ kernels compile OK.

I use the following commands to prepare the kernel for compiling the drivers
```
        make mrproper
        make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcm2711_defconfig
        make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- modules_prepare
        wget -q https://github.com/Hexxeh/rpi-firmware/raw/$rpi_firmware_commit_id/Module8.symvers -O Module.symvers
```

Typical error which appears for all the modules I try compiling for kernel version -v8+ is:-
```
  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "__stack_chk_guard" [/home/pi/wifi-rtl8822bu/rtl8822bu/8822bu.ko] undefined!
scripts/Makefile.modpost:93: recipe for target '__modpost' failed
make[2]: *** [__modpost] Error 1
Makefile:1645: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
make[1]: Leaving directory '/home/pi/src/linux'
Makefile:2002: recipe for target 'modules' failed
make: *** [modules] Error 2
```

**__stack_chk_guard** is not in the Module8.symvers file downloaded from repo Hexxeh/rpi-firmware commit 64d94937a4 or repo raspberrypi/firmware commit 11e3c314bc, however, doing a full compile of the **5.4.69-v8+** kernel and the Module.symvers file created includes the value **__stack_chk_guard**.

Would be helpful to fix this issue to save having to do a full compile of the kernel before compiling the wifi drivers.

MrEngman

pelwell, 2020-10-06:
This seems to be due to a difference in newer gcc versions. Looking at .config files generated with systems running gcc 5.2 and 9.3 respectively:
```diff
7c7
< # Compiler: aarch64-unknown-linux-gnueabi-gcc (crosstool-NG crosstool-ng-1.22.0-115-g785dbce - aarch64-linux-gnueabi) 5.2.1 20151005
---
> # Compiler: aarch64-linux-gnu-gcc (Ubuntu 9.3.0-10ubuntu1) 9.3.0
10c10
< CONFIG_GCC_VERSION=50201
---
> CONFIG_GCC_VERSION=90300
13a14
> CONFIG_CC_HAS_ASM_INLINE=y
417a419,420
> CONFIG_CC_HAVE_STACKPROTECTOR_SYSREG=y
> CONFIG_STACKPROTECTOR_PER_TASK=y
7437a7441,7442
> CONFIG_CC_HAS_SANCOV_TRACE_PC=y
> # CONFIG_KCOV is not set
```
gcc 9.3 is shipping as standard with Ubuntu 20.04, both native and as cross-compilers, and that has become our standard toolchain. I suggest you upgrade.

MrEngman, 2020-10-07:
Thanks for the response @pelwell.

Doesn't make too much sense to me but then I essentially worked as a hardware engineer and my programming work was primarily with assemblers for the Motorola M6809 and Zilog Z80180 microprocessors, although I did do some work on the linux NCR53c8xx scsi driver back in the mid 1990's.

I compile the wifi drivers on a Pi3B+ and usually use the gcc version that matches the one used to compile the kernel as detailed in the ModuleXX.symvers file. I did notice that the latest v8+ kernel used gcc 9.1 so I'm now looking to update the 64bit compiler to match that and move to using a Pi4B.

Unfortunately I cannot find the link to the site that I used originally showing how to install the 64 bit compiler but have found another that is not quite as specific so hopefully I will get things working today.

Once again thanks for your response

MrEngman

pelwell, 2020-10-07:
I think a rough translation would be "newer GCC compilers support a fancy implementation of the stack checking feature, and that difference is visible in the module symbols". It's unfortunate that it breaks compatibility in this way, but we have to upgrade at some point.

MrEngman, 2020-10-07:
Well, I have been able to install 64bit gcc 9.1 on a Pi4B 4GB and it appears to be working OK. Initially I got an error but it appeared that I needed to install libssl-dev. Once done the kernel appeared to compile OK although there were some warnings while compiling a few drivers.

I have now noticed however with the latest commit on the repo that you are now using gcc 9.3 to compile all versions of the kernel, +, -v7+, -v7l+ and -v8+. Makes life rather awkward as gcc 9 does not appear to be available for the raspberrypi.

Will update gcc-9.1 to gcc-9.3 for 64bit kernels but I'm not sure how to get hold of a version to compile the 32bit kernels. Still it'll give me something to keep me busy.

MrEngman

popcornmix, 2020-10-08:
gcc-9 looks to be available in debian testing which can be got by adding testing to sources.list (e.g. [here](https://stackoverflow.com/questions/61590804/how-to-install-gcc-9-on-debianlatest-docker)).

But before trying that does gcc-8 fail? While we know there was a problematic change somewhere between ggc-5 and gcc-9, it may not occur between gcc-8 and gcc-9.

MrEngman, 2020-10-12:
I'm now able to compile the wifi drivers using gcc-8 for the 32bit kernels and gcc-9 for the 64bit kernel, although I'm getting a variety of warnings that I didn't see using the previous gcc 4.9 and gcc 5.4.

Tried your link to updating gcc and can make no sence out of it. Just get a load of errors.

MrEngman

pelwell, 2020-10-28:
This is just to warn you that for various reasons we've reverted our kernel builds to use gcc 8, so for now the `__stack_chk_guard` is back as it used to be, and is likely to remain so as long as we're on buster.