raspberrypi/firmware #1479: Zero W no usable X with dtoverlay=vc4-kms-v3d

pepa65, 2020-10-07:
I hope this is the right place for this report. It seems that this got added to /boot/config.txt:
```
[all]
dtoverlay=vc4-kms-v3d
```
This made the display output for X unusable (consoles were OK) and caused a hang once trying to display X.

**To reproduce**
I did the Raspbian upgrade on the Zero W early this month.

**Expected behaviour**
It would still work.

**Actual behaviour**
Whenever X got display, it would cause the system to hang. If I switched to console at the right point, the system was still usable otherwise.

raspinfo:
```
System Information
------------------

Raspberry Pi Zero W Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage2

Linux spider 5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux
Revision	: 9000c1
Serial		: 00000000b314b5d6
Model		: Raspberry Pi Zero W Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Sep  2 2020 21:15:18 
Copyright (c) 2012 Broadcom
version 4439d2aaa6c376a2d1ef4402f142e1cf4de37c43 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30393948 2982508  26145172  11% /
devtmpfs          188256       0    188256   0% /dev
tmpfs             221272   40972    180300  19% /dev/shm
tmpfs             221272    3200    218072   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             221272       0    221272   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   54617    203479  22% /boot
tmpfs              44252       8     44244   1% /run/user/1000

Filename				Type		Size	Used	Priority
/var/swap                               file		102396	40704	-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
  Installed: 20200812
openbox:
[... 445 lines cut ...]
[   28.823168] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   28.824831] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd
[   29.781259] random: crng init done
[   29.781318] random: 7 urandom warning(s) missed due to ratelimiting
[   40.720711] 8021q: 802.1Q VLAN Support v1.8
[   40.921736] uart-pl011 20201000.serial: no DMA platform data
[   42.709829] Adding 102396k swap on /var/swap.  Priority:-2 extents:2 across:307200k SSFS
[   45.884613] Bluetooth: Core ver 2.22
[   45.884909] NET: Registered protocol family 31
[   45.884932] Bluetooth: HCI device and connection manager initialized
[   45.885086] Bluetooth: HCI socket layer initialized
[   45.885122] Bluetooth: L2CAP socket layer initialized
[   45.885199] Bluetooth: SCO socket layer initialized
[   46.046482] Bluetooth: HCI UART driver ver 2.3
[   46.046523] Bluetooth: HCI UART protocol H4 registered
[   46.046699] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   46.047170] Bluetooth: HCI UART protocol Broadcom registered
[   50.971763] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   50.971784] Bluetooth: BNEP filters: protocol multicast
[   50.971820] Bluetooth: BNEP socket layer initialized
[   55.146137] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   57.638426] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   66.234608] fuse: init (API version 7.31)
[   92.556381] usb 1-1: USB disconnect, device number 2
```

popcornmix, 2020-10-07:
Pi0/Pi1 do not have enough ram or cpu power to run the kms driver.

pepa65, 2020-10-07:
I am not sure what went wrong in the upgrade, but it would be good to prevent problems for other people with a Zero. Make sure those lines don't get added to config.txt. Where should I go with this?

pelwell, 2020-10-07:
I don't think the upgrade would have added or modified that line - we make a point of not touching config.txt except in response to a specific user request such as using the Configuration utility.

pepa65, 2020-10-07:
OK, that is it. When I was trying to get the `gldriver-test` (which I purged later) to work, I chose an option in the Configuration utility that added the line. PEBKAC...

closed 2020-10-07 by pepa65