raspberrypi/firmware #1480: brcmf_sdio_hostmail: mailbox indicates firmware halted (not running as AP, can reproduce, need help debugging)

tayl, 2020-10-09:
Hello,

**Describe the bug**
Raspberry Pi 3B+ wireless connection hangs or fails to connect initially to wireless network. We have reason to believe the issue is caused by environmental factors. Specifically, this is only happening at one site, although the firmware and versions involved are the same.

**To reproduce**
We have thousands of deployed devices, a specific subset of which in the same physical space experience this issue regularly. I have remote access to the devices, and can typically reproduce the issue in less than 30 minutes on at least one device. I'm just not sure how to grab relevant/useful information about what's going on.

**Expected behaviour**
Raspberry Pi connects to wireless network and remains connected.

**Actual behaviour**
Raspberry Pi frequently fails to connect to wireless network and does not remain connected once a connection is established.

**System**

* Which model of Raspberry Pi?
Raspberry Pi 3B+

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-07-10

* Which firmware version (`vcgencmd version`)?
Aug 15 2019 12:06:42 
Copyright (c) 2012 Broadcom
version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux ampxd-B827EBEB39A0 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

**Logs**
Two scenarios: scenario (1) is when the device starts up but is not able to connect to wifi right off the bat, scenario (2) is where the device has started up and is connected to wifi but some random time later experiences an issue that causes it to disconnect.

scenario (1):
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.66-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1253 SMP Thu Aug 15 11:49:46 BST 2019
... startup ...
[   21.113271] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted
[   30.391838] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   30.392228] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[   30.392243] brcmfmac: brcmf_cfg80211_disconnect: error (-110)
[   32.951849] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   32.952253] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[   32.952267] brcmfmac: brcmf_cfg80211_reg_notifier: Country code iovar returned err = -110
[   35.511841] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   35.512226] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[   35.512240] brcmfmac: brcmf_run_escan: error (-110)
[   35.512250] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[   39.031855] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   39.032273] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
... repeats ...
[  156.872283] brcmfmac: brcmf_run_escan: error (-110)
[  156.872293] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[  160.391888] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[  160.392286] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[  160.392302] brcmfmac: brcmf_run_escan: error (-110)
[  160.392312] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[  163.911885] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[  163.912282] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[  163.912297] brcmfmac: brcmf_run_escan: error (-110)
[  163.912307] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[  167.431888] brcmfmac: brcmf_run_escan: error (-110)
[  167.431905] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
... repeats ...
[  649.671927] brcmfmac: brcmf_run_escan: error (-110)
[  649.671945] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[  653.191915] brcmfmac: brcmf_run_escan: error (-110)
[  653.191931] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
... device reboots intentionally
```

scenario (2):

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.66-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1253 SMP Thu Aug 15 11:49:46 BST 2019
... startup ...
[86412.014522] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted
[86420.334042] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[86420.334450] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[86420.334466] brcmfmac: brcmf_cfg80211_disconnect: error (-110)
[86422.894019] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[86422.894365] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[86422.894380] brcmfmac: brcmf_cfg80211_reg_notifier: Country code iovar returned err = -110
[86425.454051] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[86425.454469] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[86425.454484] brcmfmac: brcmf_run_escan: error (-110)
[86425.454494] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
... repeats ...
[86548.334564] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[86548.334959] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[86548.334974] brcmfmac: brcmf_run_escan: error (-110)
[86548.334984] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[86551.854583] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[86551.855037] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[86551.855052] brcmfmac: brcmf_run_escan: error (-110)
[86551.855062] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[86555.374613] brcmfmac: brcmf_run_escan: error (-110)
[86555.374628] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[86558.894623] brcmfmac: brcmf_run_escan: error (-110)
[86558.894640] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[86562.414630] brcmfmac: brcmf_run_escan: error (-110)
[86562.414645] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[86565.934642] brcmfmac: brcmf_run_escan: error (-110)
[86565.934658] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[86569.454675] brcmfmac: brcmf_run_escan: error (-110)
[86569.454691] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[86572.094669] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[86572.094687] brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
[86574.654657] brcmfmac: brcmf_run_escan: error (-110)
[86574.654666] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[86578.174721] brcmfmac: brcmf_run_escan: error (-110)
[86578.174737] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
... repeats ...
[86627.454926] brcmfmac: brcmf_run_escan: error (-110)
[86627.454944] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[86630.974950] brcmfmac: brcmf_run_escan: error (-110)
[86630.974967] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[86633.534958] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[86633.534991] brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
[86636.094960] brcmfmac: brcmf_run_escan: error (-110)
[86636.094976] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[86639.614998] brcmfmac: brcmf_run_escan: error (-110)
[86639.615014] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
... repeats ...
[87227.457787] brcmfmac: brcmf_run_escan: error (-110)
[87227.457802] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[87230.977790] brcmfmac: brcmf_run_escan: error (-110)
[87230.977807] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
... device reboots intentionally
```

**Additional context**
Possibly relevant is that the devices are using Cisco 3602 APs. The area is somewhat wirelessly congested, but no more than we see elsewhere so I don't believe that to be relevant.

tayl, 2020-10-09:
Just noting.. I've seen similar issues reported, but all seem to be related to using the Raspberry Pi as an access point of some sort. In this scenario, we're only connecting to wireless networks. I believe for that reason that this is a fundamentally different issue although the generic output seen is the same or similar.

pelwell, 2020-10-09:
Can you retest with `brcmfmac.debug=0x100000` in /boot/cmdline.txt (keep everything in a single line)? The firmware should log debug into the kernel log, including details of the crash.

tayl, 2020-10-09:
pelwell,

I put that in the cmdline.txt file and see it reflected in the dmesg output, but I don't see any additional debug information here. Should I be looking elsewhere?

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.66-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1253 SMP Thu Aug 15 11:49:46 BST 2019
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xac/0x4b4 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s39488 r8192 d21952 u69632
[    0.000000] pcpu-alloc: s39488 r8192 d21952 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=89c0edae-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait brcmfmac.debug=0x100000
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 939084K/970752K available (8192K kernel code, 629K rwdata, 2176K rodata, 1024K init, 821K bss, 23476K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                   lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 630 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 822 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26394 entries in 78 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000023] Switching to timer-based delay loop, resolution 52ns
[    0.000291] Console: colour dummy device 80x30
[    0.000917] console [tty1] enabled
[    0.000975] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001020] pid_max: default: 32768 minimum: 301
[    0.001377] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001412] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002345] CPU: Testing write buffer coherency: ok
[    0.002831] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003499] Setting up static identity map for 0x100000 - 0x10003c
[    0.003666] rcu: Hierarchical SRCU implementation.
[    0.004482] smp: Bringing up secondary CPUs ...
[    0.005344] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.006273] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.007137] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.007258] smp: Brought up 1 node, 4 CPUs
[    0.007340] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.007364] CPU: All CPU(s) started in HYP mode.
[    0.007384] CPU: Virtualization extensions available.
[    0.008377] devtmpfs: initialized
[    0.021693] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.021969] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.022018] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.022626] pinctrl core: initialized pinctrl subsystem
[    0.023501] NET: Registered protocol family 16
[    0.026530] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.032277] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[... 217 lines cut ...]
[   14.129813] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   14.130053] Bluetooth: HCI UART protocol Broadcom registered
[   14.387429] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   14.387435] Bluetooth: BNEP filters: protocol multicast
[   14.387450] Bluetooth: BNEP socket layer initialized
[   23.082157] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted
[   31.271902] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[... cut ...]
[   36.392318] brcmfmac: brcmf_run_escan: error (-110)
[   36.392328] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
...
[  164.711933] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[  164.712359] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[  164.712375] brcmfmac: brcmf_run_escan: error (-110)
[  164.712385] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
...
[  653.992014] brcmfmac: brcmf_run_escan: error (-110)
[  653.992033] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
```

tayl, 2020-10-11:
New development, I just realized that all the scenario 2 cases happen at or very soon after the 24 hour uptime mark. Still not sure where to see debug logs though. They don't show in dmesg for any of the logs I'm getting.