raspberrypi/firmware #1482: Pi4: Problem with mesa 20.2.1 on vc4-kms-v3d and 5.9.y kernel

vanfanel, 2020-10-16:
Hi there,

I have been trying to make latest stable MESA 20.2.1 work with my Pi4.
I am using latest kernel in the NEXT branch:

```
pi@raspberrypi:~/src/mesa-20.2.1/b4 $ uname -a
Linux raspberrypi 5.9.0-v8+ #1351 SMP PREEMPT Wed Oct 14 12:50:37 BST 2020 aarch64 GNU/Linux
```

I am using the vc4-kms-v3d overlay, and run full 64bit Raspberry Pi OS with a 64bit kernel and libs.

**libDRM's modeset test works well, and I can see the color pattern** with
`modetest -s 32:1360x768`
(I am using latest libDRM, which is libdrm-2.4.102)

I build MESA 20.2.1 configuring it like this:
`meson -Dglx=disabled -Dplatforms=drm -Dllvm=disabled -Dvulkan-drivers='' -Ddri-drivers='' -Dgallium-drivers=vc4,v3d,kmsro ..`

MESA configuration report with these options is as follows:
```
pi@raspberrypi:~/src/mesa-20.2.1/b4 $ meson -Dglx=disabled -Dplatforms=drm -Dllvm=disabled -Dvulkan-drivers='' -Ddri-drivers='' -Dgallium-drivers=vc4,v3d,kmsro ..
The Meson build system
Version: 0.52.1
Source dir: /home/pi/src/mesa-20.2.1
Build dir: /home/pi/src/mesa-20.2.1/b4
Build type: native build
Program python found: YES (/usr/bin/python)
Project name: mesa
Project version: 20.2.1
C compiler for the host machine: cc (gcc 8.3.0 "cc (Debian 8.3.0-6) 8.3.0")
C linker for the host machine: GNU ld.bfd 2.31.1
C++ compiler for the host machine: c++ (gcc 8.3.0 "c++ (Debian 8.3.0-6) 8.3.0")
C++ linker for the host machine: GNU ld.bfd 2.31.1
Host machine cpu family: aarch64
Host machine cpu: aarch64
meson.build:324: WARNING: Platform `drm` is now automatically selected; setting this option will be an error in Mesa 20.3
Program python3 found: YES (/usr/bin/python3)
Fetching value of define "ETIME" : 62 
Checking for function "bswap32" : YES 
Checking for function "bswap64" : YES 
Checking for function "clz" : YES 
Checking for function "clzll" : YES 
Checking for function "ctz" : YES 
Checking for function "expect" : YES 
Checking for function "ffs" : YES 
Checking for function "ffsll" : YES 
Checking for function "popcount" : YES 
Checking for function "popcountll" : YES 
Checking for function "unreachable" : YES 
Compiler for C supports function attribute const: YES 
Compiler for C supports function attribute flatten: YES 
Compiler for C supports function attribute malloc: YES 
Compiler for C supports function attribute pure: YES 
Compiler for C supports function attribute unused: YES 
Compiler for C supports function attribute warn_unused_result: YES 
Compiler for C supports function attribute weak: YES 
Compiler for C supports function attribute format: YES 
Compiler for C supports function attribute packed: YES 
Compiler for C supports function attribute returns_nonnull: YES 
Compiler for C supports function attribute alias: YES 
Compiler for C supports function attribute noreturn: YES 
Compiler for C supports function attribute visibility:hidden: YES 
Checking if "__uint128_t" compiles: YES 
Compiler for C supports arguments -Werror=implicit-function-declaration: YES 
Compiler for C supports arguments -Werror=missing-prototypes: YES 
Compiler for C supports arguments -Werror=return-type: YES 
Compiler for C supports arguments -Werror=empty-body: YES 
Compiler for C supports arguments -Werror=incompatible-pointer-types: YES 
Compiler for C supports arguments -Werror=int-conversion: YES 
[... 128 lines cut ...]
KMSCUBE results in this:
```

pi@raspberrypi:~/src/kmscube/b4 $ ./kmscube
Using display 0x55639ac770 with EGL version 1.4
===================================
EGL information:
  version: "1.4"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_client_extensions EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_KHR_debug EGL_EXT_platform_device EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_image_dma_buf_export EGL_MESA_query_driver "
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 3.1 Mesa 20.2.1"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Mesa/X.org"
  renderer: "softpipe"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_float_linear GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_NV_conditional_render GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_OES_viewport_array GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_copy_image GL_EXT_draw_buffers_indexed GL_EXT_draw_elements_base_vertex GL_EXT_gpu_shader5 GL_EXT_primitive_bounding_box GL_EXT_render_snorm GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_buffer GL_EXT_texture_cube_map_array GL_EXT_texture_norm16 GL_EXT_texture_view GL_KHR_context_flush_control GL_NV_image_formats GL_OES_copy_image GL_OES_draw_buffers_indexed GL_OES_draw_elements_base_vertex GL_OES_gpu_shader5 GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_buffer GL_OES_texture_cube_map_array GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_OES_texture_view GL_EXT_blend_func_extended GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_clip_cull_distance GL_EXT_disjoint_timer_query GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_EXT_clip_control GL_EXT_texture_compression_bptc GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_EXT_texture_sRGB_R8 GL_MESA_framebuffer_flip_y GL_EXT_depth_clamp GL_EXT_texture_query_lod "
===================================
Using modifier ffffffffffffff
failed to set mode: Invalid argument
```

I need this to work (I have to test on the PI4 my own KMSDRM backend for SDL2, which is available here: https://hg.libsdl.org/SDL/file/c524405e6e65/src/video/kmsdrm), and **I suspect the problem is somewhere in MESA since libDRM tests work.**
Any idea on what's going on, please?

popcornmix, 2020-10-16:
On Pi4 I believe you want
```
kmscube -D /dev/dri/card1
```

vanfanel, 2020-10-16:
@popcornmix 

`kmscube -D /dev/dri/card1` results in:

```
pi@raspberrypi:~/src/kmscube/b4 $ ./kmscube -D /dev/dri/card1
Using display 0x55a9a2d470 with EGL version 1.4
===================================
EGL information:
  version: "1.4"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_client_extensions EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_KHR_debug EGL_EXT_platform_device EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_image_dma_buf_export EGL_MESA_query_driver "
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 3.1 Mesa 20.2.1"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Mesa/X.org"
  renderer: "softpipe"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_float_linear GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_NV_conditional_render GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_OES_viewport_array GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_copy_image GL_EXT_draw_buffers_indexed GL_EXT_draw_elements_base_vertex GL_EXT_gpu_shader5 GL_EXT_primitive_bounding_box GL_EXT_render_snorm GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_buffer GL_EXT_texture_cube_map_array GL_EXT_texture_norm16 GL_EXT_texture_view GL_KHR_context_flush_control GL_NV_image_formats GL_OES_copy_image GL_OES_draw_buffers_indexed GL_OES_draw_elements_base_vertex GL_OES_gpu_shader5 GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_buffer GL_OES_texture_cube_map_array GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_OES_texture_view GL_EXT_blend_func_extended GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_clip_cull_distance GL_EXT_disjoint_timer_query GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_EXT_clip_control GL_EXT_texture_compression_bptc GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_EXT_texture_sRGB_R8 GL_MESA_framebuffer_flip_y GL_EXT_depth_clamp GL_EXT_texture_query_lod "
===================================
Using modifier ffffffffffffff
failed to set mode: Invalid argument
```

Also, I have found out that RetroArch works but VERY slowly.
Now, looking closely at the KMSCUBE information, I see:
```

OpenGL ES 2.x information:
  version: "OpenGL ES 3.1 Mesa 20.2.1"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Mesa/X.org"
  renderer: "softpipe"
```

So it's probably trying to use the **softpipe** software rasterizer. As you can see, I have built MESA with v3d, vc4 and kmsro, as it seems to be recommended everywhere. And it succeeds on both configuration and building.
So still there's something missing here. Why no MESA VC4/V3D?

vanfanel, 2020-10-16:
@popcornmix Also, forcing the V3D driver results on a segfault on libGBM when it tries to create a GBM surface:
```

pi@raspberrypi:~/src/kmscube/b4 $ GALLIUM_DRIVER=v3d gdb ./kmscube
GNU gdb (Debian 8.2.1-2+b3) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./kmscube...done.
(gdb) r
Starting program: /home/pi/src/kmscube/b4/kmscube 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
gbm_surface_create_with_modifiers (gbm=0x0, width=1920, height=1080, format=875713112, modifiers=0x7ffffff398, count=1)
    at ../src/gbm/main/gbm.c:623
623        return gbm->surface_create(gbm, width, height, format, 0,
(gdb) 
(gdb) bt
#0  gbm_surface_create_with_modifiers (gbm=0x0, width=1920, height=1080, format=875713112, modifiers=0x7ffffff398, count=1)
    at ../src/gbm/main/gbm.c:623
#1  0x0000005555558b28 in init_surface (modifier=0) at ../common.c:96
#2  0x0000005555558ca8 in init_gbm (drm_fd=3, w=1920, h=1080, format=875713112, modifier=0, surfaceless=false) at ../common.c:136
#3  0x0000005555560054 in main (argc=1, argv=0x7ffffff5b8) at ../kmscube.c:202
```

What's happening with V3D on the Pi4?
Let me repeat that libDRM tests work.

joolswills, 2020-10-17:
I'm running Mesa 20.3.0-devel (packages built on 12th October - not sure exact Mesa commit), and it's working fine on 5.9.0 kernel on my RPI4 buster (32bit).

When building newer Mesa, it did require a newer libdrm than is available in buster. It built against the buster version but I had an issue which may have been related to your problem (no details, but no output but I didn't debug, as I had noticed upstream Debian Mesa packaging had bumped the required libdrm so just went to try that and it worked afair). 

I'm running 2.4.102-1 (built from Debian unstable repo).

I'm also running libglvnd 1.3.2 again from Debian unstable sources.

Worth updating libdrm to see if it's a similar issue.

joolswills, 2020-10-17:
Btw (wrong place really but as it was mentioned) your sdl2 krmdrm backend has been broken for some time. Since you made the atomic stuff optional.and removed all the asserts it sort of works. With loads of tearing. I plan to open an SDL issue on that btw. Back on topic...

joolswills, 2020-10-17:
Sorry. I missed that you were already running the latest libdrm. I can try the code mentioned and post a log from my system if that helps. Perhaps it's something affecting 64bit only though.

joolswills, 2020-10-17:
Just built kmscube. Here's a log. Works ok on my config.

```
Using display 0x828710 with EGL version 1.4
===================================
EGL information:
  version: "1.4"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 3.1 Mesa 20.3.0-devel (git-181d5f59b8)"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_EXT_buffer_storage GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
===================================
Rendered 121 frames in 2.016267 sec (60.011882 fps)
Rendered 242 frames in 4.032510 sec (60.012249 fps)
user interrupted!
```

joolswills, 2020-10-17:
Sorry for ticket spam. Maybe something useful in the log above. I can try on 64bit if needed. Perhaps it's just something missing from your Mesa build? 

If I can help further testing anything to compare please let me know.

vanfanel, 2020-10-17:
@joolswills Thanks for trying to help, really.
I have a couple of questions for you.
1) Do you have a /usr/local/lib/aarch64-linux-gnu/dri/libgallium_dri.so file? Or a libgallium_dri.so file living wherever in your system, installed by your own MESA build.
2) Can you ldd to that file, please? Maybe I am missing something that MESA needs and it's not building libgallium because of that.
3) Can you try on a 64bit system? Kernel and libs. Use the kms overlay, not the fkms one.

About SDL2 KMSDRM backend, the ATOMIC version works very well here on the Pi3, so I hope it does the same on the PI4 unless it's still missing some extension, there should be no tearing at all. But first I have to get it running here.

vanfanel, 2020-10-17:
Also, what are these dependencies?

```
Run-time dependency v3dv3 found: NO (tried pkgconfig and cmake)
Run-time dependency simpenrose found: NO (tried pkgconfig and cmake)
Run-time dependency v3dv3 found: NO (tried pkgconfig and cmake)
```
Are they needed for the v3d driver or related to the Pi at all?

vanfanel, 2020-10-17:
@joolswills Also, what HDMI connector are you using?

vanfanel, 2020-10-18:
OK, problem solved. 
It seems that the "pi" user has to be in the "render" group.
`sudo usermod -a -G render pi`

It really should be documented somewhere...

closed 2020-10-18 by vanfanel

joolswills, 2020-10-20:
Glad it's sorted. The reason I updated glvnd was just due to the fact it was a bumped dependency upstream and I was building packages based on that.

v3dv3 is the new recently merged vulkan driver afaik. Or related to.

Not sure the permissions issue can be blamed on documentation here. Raspbian afaik ships with the correct permissions for the pi user - actually on my 32bit Raspbian the video group is used. What OS did you start with?

joolswills, 2020-10-20:
Sorry. you're right about v3dv3. I had seen it in reference to vulkan changes recently.