raspberrypi/firmware #1485: [Pi4] [Pi OS 64] Kernel errors with vc4-hdmi audio device.

vanfanel, 2020-10-18:
Hi,

I am running latest NEXT branch kernel
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.9.0-v8+ #1351 SMP PREEMPT Wed Oct 14 12:50:37 BST 2020 aarch64 GNU/Linux
```
I am using the vc4-hdmi audio device, so **dtparam=audio=on** is commented in my config.txt.

Everything looks fine, except I get this everytime a program using ALSA audio quits:

```
pi@raspberrypi:~ $ dmesg
[  275.293236] BUG: scheduling while atomic: retroarch/444/0x00000002
[  275.294654] Modules linked in: hid_logitech_hidpp joydev vc4 cec drm_kms_helper v3d gpu_sched hid_logitech_dj raspberrypi_hwmon i2c_brcmstb drm drm_panel_orientation_quirks bcm2835_isp(C) bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem snd_soc_core videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 snd_compress videobuf2_common snd_pcm_dmaengine videodev snd_pcm mc vc_sm_cma(C) snd_timer snd syscopyarea sysfillrect sysimgblt rpivid_mem fb_sys_fops backlight uio_pdrv_genirq uio i2c_dev
[  275.304664] Preemption disabled at:
[  275.304685] [<ffffffc008b3242c>] snd_pcm_stream_lock_irq+0x3c/0x48 [snd_pcm]
[  275.308288] CPU: 2 PID: 444 Comm: retroarch Tainted: G        WC        5.9.0-v8+ #1351
[  275.310145] Hardware name: Raspberry Pi 4 Model B Rev 1.2 (DT)
[  275.312021] Call trace:
[  275.313886]  dump_backtrace+0x0/0x1b0
[  275.315751]  show_stack+0x20/0x30
[  275.317547]  dump_stack+0xe0/0x14c
[  275.319301]  __schedule_bug+0xb0/0xc0
[  275.321033]  __schedule+0x75c/0x878
[  275.322705]  schedule+0x50/0x110
[  275.324300]  schedule_hrtimeout_range_clock+0x94/0x118
[  275.325881]  schedule_hrtimeout_range+0x1c/0x28
[  275.327381]  usleep_range+0x7c/0xb0
[  275.328829]  vc4_hdmi_write_infoframe+0x2f0/0x690 [vc4]
[  275.330394]  vc4_hdmi_set_audio_infoframe+0x58/0x80 [vc4]
[  275.331977]  vc4_hdmi_audio_trigger+0x3c/0x288 [vc4]
[  275.333541]  snd_soc_pcm_dai_trigger+0x5c/0x108 [snd_soc_core]
[  275.335234]  soc_pcm_trigger+0xc0/0xd8 [snd_soc_core]
[  275.336953]  snd_pcm_do_start+0x3c/0x48 [snd_pcm]
[  275.338577]  snd_pcm_action_single+0x50/0xb0 [snd_pcm]
[  275.340148]  snd_pcm_action+0x84/0xa0 [snd_pcm]
[  275.341645]  snd_pcm_action_lock_irq+0x40/0x68 [snd_pcm]
[  275.343079]  snd_pcm_common_ioctl+0xf8c/0x12a0 [snd_pcm]
[  275.344440]  snd_pcm_ioctl+0x38/0x50 [snd_pcm]
[  275.345805]  __arm64_sys_ioctl+0xb0/0xf8
[  275.347043]  el0_svc_common.constprop.0+0x7c/0x1b0
[  275.348273]  do_el0_svc+0x2c/0x98
[  275.349459]  el0_sync_handler+0x9c/0x1b8
[  275.350674]  el0_sync+0x158/0x180

```
Is this expected with the experimental 5.9.y kernel?

6by9, 2020-10-18:
Duplicated at https://www.raspberrypi.org/forums/viewtopic.php?p=1744139#p1744139
NB This is a Linux kernel issue, not firmware - https://github.com/raspberrypi/linux/issues next time please.

vanfanel, 2020-10-18:
@6by9 Oh, sorry :)

closed 2020-10-18 by vanfanel