raspberrypi/firmware #1486: No output over HDMI with cut down firmware

IronOxidizer, 2020-10-18:
**Is this the right place for my bug report?**
This repository contains the GPU firmware used on the Raspberry Pi. This software is the closed source part of the Raspberry Pi system, it includes booting (including network booting and USB booting), low-level power and clock control, FKMS and legacy HDMI control (not full KMS that is in the linux kernel), hardware legacy codecs (MPEG2, H264 and VC1), encode hardware including the ISP (image sensor pipeline) and camera control, audio output (analogue and HDMI audio).

If you believe that the issue you are seeing is within this area, this is the right place. If not, we have other repositories for the linux kernel at [github.com/raspberrypi/linux](https://github.com/raspberrypi/linux) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
I am unable to boot to tty over HDMI using cut down firmware.

**To reproduce**
Start with RPi OS Minimal or Buildroot RPi defconf. (boots to tty, no X or any other graphical interface).

Replace `start.elf` with `start_cd.efl` and `fixup.dat` with `fixup_cd.dat`. Append the following lines to `config.txt`:
```
gpu_mem=16
start_file=start_cd.elf
fixup_file=fixup_cd.dat
```
Attempt to boot. 

**Expected behaviour**
HDMI output to tty with kernel loading messages.

**Actual behaviour**
Device receives power but there is no output over HDMI (not even black screen, monitor detects no signal). Also does not work with `hdmi_safe=1`.

**System**
- Raspbery Pi 1B and Raspberry Pi 0 W
- Buildroot 2020.08.1
- https://github.com/raspberrypi/firmware/tree/stable (Sept 2)
- raspberrypi-kernel_1.20200902-1

timg236, 2020-10-19:
Please can you try to reproduce on Raspberry Pi OS lite + rpi-update

IronOxidizer, 2020-10-19:
Unable to reproduce on RPi OS Lite.

IronOxidizer, 2020-10-19:
Was able to figure out the issue is related to `stable` firmware (`start_cd.elf` and `fixup_cd.dat`).

For now I will use `master` branch but I'll keep the issue open for now.

closed 2020-10-19 by IronOxidizer

whitslack, 2020-12-15:
> For some reason, if the boot partition is smaller than 8208KB (even if all the firmware fits), the system fails to boot.

@IronOxidizer: That's not true. I have a Raspberry Pi 1B that boots from a 7104-KiB boot partition using the latest (Dec 2020) firmware files. The key is that the boot code doesn't support FAT12, so you need to ensure that your boot partition contains a FAT16 file system. FAT16 requires at least 4085 data clusters. If you're using 2048-byte clusters, then you need at least 8170 KiB of data area to ensure a FAT16 file system and not a FAT12. You can have a smaller FAT16 file system, though, simply by decreasing the cluster size. My boot partition has a FAT16 file system using 1024-byte clusters, which gives me 7040 data clusters, well above the 4085-cluster lower bound.

IronOxidizer, 2020-12-15:
Thanks for all the insight, I would have never guessed to look at block size.

IronOxidizer, 2021-02-01:
I just did some testing and I wasn't able to replicate @whitslack 's success. I was able to set the FS to FAT12 and the cluster size to 512 using the following line in genimage

```
extraargs = "-F 16 -S 512"
```

And verified the output using `parted`

```
>>> sudo fdisk -l /dev/sdc1
Disk /dev/sdc1: 7.62 MiB, 7987200 bytes, 15600 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos

>>> sudo file -s /dev/sdc1
/dev/sdc1: DOS/MBR boot sector, code offset 0x3c+2, OEM-ID "mkfs.fat", sectors/cluster 4, reserved sectors 4, root entries 512, sectors 15600 (volumes <=32 MB), Media descriptor 0xf8, sectors/FAT 16, sectors/track 32, heads 64, serial number 0xe79ba1e8, unlabeled, FAT (16 bit)
```

However I was not able to boot the device with this setup, even when setting the partition size 1kB smaller than my previously stated limit (8208kB).

For reference, this partition shouldn't need to be any bigger than 7M

```
du -sh /mnt                                                                                                
6.9M	/mnt
```

whitslack, 2021-02-01:
@IronOxidizer: For your reference, here are my partition and file system parameters.

```
# fdisk -l /dev/mmcblk0
Disk /dev/mmcblk0: 121.25 MiB, 127139840 bytes, 248320 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xdbd70e86

Device         Boot Start    End Sectors   Size Id Type
/dev/mmcblk0p1        128  14335   14208   6.9M  e W95 FAT16 (LBA)
/dev/mmcblk0p2      14336 248319  233984 114.3M 83 Linux

# fsck.fat -nv /dev/mmcblk0p1
fsck.fat 4.1 (2017-01-24)
Checking we can access the last sector of the filesystem
0x25: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Boot sector contents:
System ID "mkfs.fat"
Media byte 0xf8 (hard disk)
       512 bytes per logical sector
      1024 bytes per cluster
        40 reserved sectors
First FAT starts at byte 20480 (sector 40)
         2 FATs, 16 bit entries
     14336 bytes per FAT (= 28 sectors)
Root directory starts at byte 49152 (sector 96)
       512 root directory entries
Data area starts at byte 65536 (sector 128)
      7040 data clusters (7208960 bytes)
16 sectors/track, 4 heads
       128 hidden sectors
     14208 sectors total
Checking for unused clusters.
Leaving filesystem unchanged.
/dev/mmcblk0p1: 10 files, 6732/7040 clusters
```

IronOxidizer, 2021-02-01:
That was super helpful. I forgot to set the partition type from `0xC` (FAT32) to `0xE` (FAT16). Everything worked as intended after that. Thanks, issue resolved!