raspberrypi/firmware #1487: Raspberry Pi 4 won't boot anymore

daniframinan, 2020-10-23:
[image]
Hi, I've been using this Raspberry for a while, and have had it running octopi for at almost a year. The other day, after installing PiVPN on it, and then rebooting it(I had already rebooted it before so I don't think that's the problem), it stopped working completely, it didn't even show up on the network. I plugged it onto a monitor(by the way the HDMI_1 isn't working), and it displayed this message and then the screen turned blank. The green light is only on some times, but it's not blinking. After some research I believe it's a firmware problem, but I don't know what to do. The reset USB port-power 1000 ms message stopped after I disconnected the keyboard and mouse. I've already tried using another SD card, with octopi, with the same result. I have also tried the SD card in a model B+ and worked fine. I have also tried using EEPROM recovery tool with pretty much the same results. I am currently looking into installing the firmware completely (I have unzipped this [Files](https://github.com/raspberrypi/firmware/tree/master/boot) into an sd, but it said it didn't detect any firmware, I'm not sure how i'm supposed to do it since I'm a bit of a newbie with the RPi and here at github) and I also seeked help [here](https://github.com/raspberrypi/firmware/issues/1445) and [here](https://superuser.com/questions/1595904/raspberry-pi-4-booting-displaying-recover4-elf-and-recovery-elf-not-found-then?noredirect=1#comment2433175_1595904) without much success.

I'm using a Raspberry Pi 4 - 4GB, with LAN, power and mini-HDMI(through an adapter to HDMI and the to DVI-D, I have also tried with hdmi directly to mini hdmi) connected.

I am stuck and I don't know what to do, would really appreciate help.

timg236, 2020-10-23:
"and it displayed this message and then the screen turned blank"

The screenshot shows that the EEPROM is running correctly, and the hashes of the firmware indicate that the files were loaded into memory correctly. If the display is going blank then I think it's probably started the firmware.

If you have a USB UART cable then you could try adding uart_2ndstage=1 to config.txt which should show progress in the firmware.

The failure might be due to damage to the HDMI I2C EDID which could lockup the I2C detection, there's update firmware workaround for that. Please can you try grabbing the latest start4x.elf and fixup4.dat files from
https://github.com/raspberrypi/firmware/tree/master/boot to see if that resolves the issue

daniframinan, 2020-10-23:
@timg236  How do i install the firmware you've provided me (I know I sound stupid, I just don't know much about RPis), do I just unzip the files onto an sd card and put it into the rpi?

timg236, 2020-10-23:
Make a backup of fixupx.dat and startx.dat on the sd-card and download just those two files from https://github.com/raspberrypi/firmware/tree/master/boot to the boot partition on the sdcard

daniframinan, 2020-10-23:
@timg236 Thanks a lot, I will try that, by the way I do have an USB UART cable (an FTDI I don't know if that's the same), but I can't get it until Tuesday, is it really necessary?

timg236, 2020-10-23:
You don't need the USB UART cable to try the updated firmware, it's just the next stage of debug which hopefully you won't need

daniframinan, 2020-10-23:
@timg236 Same result

daniframinan, 2020-10-23:
@timg236 This is probably stupid, but i do have an Arduino, is it possible to use it as the USB UART?

daniframinan, 2020-10-23:
@timg236 I am very sorry, but I completely forgot to inform you that the raspberry pi is also no connecting to the internet, the LAN port lights up but nowhere in the router have i been able to see its MAC, so I'm guessing it's not working. I also said the green light lighted up sometimes, but lately it's been like this: Lights up and blinks while the screen is displaying the message, and then turns off at the same time as the screen.

timg236, 2020-10-23:
Thanks that’s useful information. The change in the network light indicates that’s it’s at least reached a certain point in the process.
If you were able to use the Raspberry Pi Imager to try this alpha release of the bootloader then it might give more information 
Flash it as a custom image and use the eeprom rescue to revert back to default once you are done
https://github.com/raspberrypi/rpi-eeprom/issues/242#issuecomment-715419076

You can use

daniframinan, 2020-10-23:
@timg236 Ok, I've done that, my screen is now green. I have now tried the raspbian(octopi) card, and the message is pretty  much the same.
[image]

timg236, 2020-10-23:
The green screen indicates that the EEPROM update completed successfully. 
Does the screen still go blank shortly after displaying MEM GPU?

It might be worth trying
* Flashing the default Raspberry Pi OS release to the SD-CARD that you used for the EEPROM update - known good SD. 
* Trying older bootloader release from https://github.com/raspberrypi/rpi-eeprom/releases although it looks as though the firmware is loading.
* Adding the following settings to config.txt

hdmi_safe=1
over_voltage=1
hdmi_force_hotplug=1

daniframinan, 2020-10-23:
@timg236 I can try an older one, do you want me to try one in particular?

daniframinan, 2020-10-23:
@timg236 I tried the [vl805](https://github.com/raspberrypi/rpi-eeprom/releases/tag/v2020.07.31-138a1) one, with basically the same results. 

[image]

By the way about "Does the screen still go blank shortly after displaying MEM GPU?" yes, it goes blank almost immidieatly, in fact, i had never been able to read that meassage until i recorded it and went frame by frame. 
By the way about the arduino idea, do you think it's possible that it would work?

timg236, 2020-10-24:
If you can get a UART working with the arduino then that's a good idea . The Raspberry Pi side of the UART console interface is documented here
https://www.raspberrypi.org/documentation/configuration/uart.md

The nightly Raspberry Pi OS builds are also a useful resource if you want to try different releases. It's basically latest Raspbian + rpi-update release.
It might be worth trying the build from yesterday
http://downloads.raspberrypi.org/nightlies/

daniframinan, 2020-10-24:
@timg236 I connected it to the arduino and putty outputed this : 
▒▒S
   ▒5▒5'3▒#!▒3!#7▒▒▒▒▒▒▒S
5▒▒▒▒▒▒▒▒▒▒/#7▒▒▒▒▒▒▒▒;▒▒/=/▒▒▒▒▒7▒▒;▒55▒▒▒▒▒▒▒▒▒▒▒9▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒3▒▒e                                                                                                                                                             ue▒q_U▒▒▒▒▒▒}[e▒▒▒▒▒▒WaW}g▒▒▒▒▒▒▒▒VQ▒-#1▒=▒5'3▒▒359▒▒▒▒▒▒=▒▒▒▒▒Y]VV▒▒▒▒▒▒▒▒▒▒▒▒▒                                                                                                                                                             ▒▒▒==#▒▒==#A5%%9A!5#▒▒Y]VV▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒;▒▒s-'5▒=7▒▒▒%3▒7▒9!#3-1▒▒▒euYY▒▒▒▒▒▒▒                                                                                                                                                             5▒▒Y]VV▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒owem▒▒uwmw▒5=7-#1▒uwmw▒;'!9)▒▒▒=5%▒▒▒▒Y]VV▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒                                                                                                                                                             ▒owem▒▒uwmw▒5=7-#1▒uwmw▒;'!9)▒▒▒=5%▒▒▒▒Y]VV▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒owem▒▒uwmw▒5=7-#1▒uwm                                                                                                                                                             w▒;'!9)▒▒▒=5%▒▒▒▒Y]VV▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒owem▒▒uwmw▒5=7-#1▒uwmw▒;'!9)▒▒▒=5%▒▒▒▒Y]VV▒                                                                                                                                                             ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒owem▒▒uwmw▒5=7-#1▒uwmw▒;'!9)▒▒▒=5%▒▒▒▒Y]VV▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒owem▒▒u                                                                                                                                                             wmw▒5=7-#1▒uwmw▒;'!9)▒▒▒=5%▒▒▒▒Y]VV▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒owem▒▒uwmw▒5=7-#1▒uwmw▒;'!9)▒                                                                                                                                                             ▒▒=5%▒▒▒▒Y]VV▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒owem▒▒uwmw▒5=7-#1▒uwmw▒;'!9)▒▒▒=5%▒▒▒▒Y]VV▒▒▒▒▒▒▒▒▒                                                                                                                                                             ▒▒▒▒▒▒▒owem▒▒uwmw▒5=7-#1▒uwmw▒;'!9)▒▒▒=5%▒▒▒▒Y]VV▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒owem▒▒uwmw▒5=7-                                                                                                                                                             #1▒uwmw▒;'!9)▒▒▒=5%▒▒▒▒euYY▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒owem▒▒uwmw▒1--#1▒▒!#▒=7-#1▒uwmw▒;                                                                                                                                                             '!9)▒▒▒▒euYY▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒;▒▒s-'5▒=7▒▒▒%3▒7▒9!#3-1▒▒▒Y]VV▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒1                                                                                                                                                             -!%=#▒▒1-!%=#A15A-#A#%▒▒-#▒wmY_g}MAwYmA_a[W▒#!▒753-#57▒▒euYY▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒                                                                                                                                                             5▒▒euYY▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒59▒▒59A%-77'5=A!5!#▒▒59A;=5▒▒▒▒59▒▒▒▒▒▒▒-7▒▒▒▒▒▒▒▒▒▒▒                                                                                                                                                             59▒▒▒▒59▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒5#9▒▒▒▒59▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒91%=5▒▒▒▒59▒▒▒▒9▒▒▒▒▒▒▒▒▒▒▒▒Y]V                                                                                                                                                             V▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒7;A3-'5▒▒;9%▒▒▒▒▒▒▒;▒7;▒▒▒Y]VV▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒;▒▒s-'5▒=7▒▒▒%3▒7                                                                                                                                                             ▒;9%▒▒▒▒▒▒▒;▒7;▒▒euYY▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒g!=7-#1▒▒;9%▒▒▒▒▒▒▒;▒7;▒▒!▒▒▒▒▒▒-
                                                                          5▒▒;▒▒                                                                                                                                                             5▒▒euYY▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒;▒▒s-'5▒=7▒▒▒%3▒7▒9!#3-1▒▒▒euYY▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒7=                                                                                                                                                             5▒▒euYY▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒;▒▒s-'5▒=7▒▒▒%3▒7▒9%7'-#5▒▒▒euYY▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒[                                                                                                                                                             5=7▒9!%%=#7▒'-#5▒33-'5▒▒9%7'-#5▒▒▒▒▒euYY▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒
5▒5▒▒5'5=!9▒▒▒5▒▒▒▒▒▒
75=7'-#5▒39)▒=-
5▒▒Y]VV▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒;▒▒s-'5▒=7▒▒▒%3▒7▒)5'▒'▒-%1▒▒euYY▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒g!=7                                                                                                                                                             -#1▒▒)5'▒'▒-%1▒▒!▒▒▒▒▒▒▒-
                         5▒▒▒;▒39▒▒▒euYY▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒w5-95▒5▒'!=757▒!▒▒▒5                                                                                                                                                             33▒▒▒▒▒▒-
         5▒▒;597▒▒▒euYY▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒=▒Y5▒_g▒▒▒▒;=7▒5▒!▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒o
                                                                             ▒▒Y                                                                                                                                                             ]VV▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒=▒{=7▒5▒9/=#15▒7!#5▒▒▒▒▒euYY▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒

Is that something or just an error?

pelwell, 2020-10-24:
That's garbage. Don't forget to attach the GND as well, and set the Arduino to 115200 baud.

daniframinan, 2020-10-24:
ok, i think i forgot to connect some resistors, i'm using this website now https://radiostud.io/understanding-raspberrypi-uart-communication/

daniframinan, 2020-10-24:
Ok, I got the arduino working, this is the message:

recover4.elf not found (6)
recovery.elf not found (6)
Read start4x.elf bytes  2985256 hnd 0x00033166 hash '6f5d6cc05fc28383'
Read fixup4x.dat bytes     8430 hnd 0x000348b7 hash '524d74b1ee853157'
0x00c03111 0x00000000 0x0000001f
MEM GPU: 128 ARM: 896 TOTAL: 1024
Starting start4x.elf @ 0xfec00200 partition 0

MESS:00:00:05.591545:0: arasan: arasan_emmc_open
MESS:00:00:05.741707:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.744973:0: brfs: File read: 1889 bytes
MESS:00:00:05.854891:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.879406:0: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:05.903923:0: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:05.928433:0: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:05.952950:0: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:05.977462:0: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:06.001979:0: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:06.026489:0: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:06.051006:0: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:06.075519:0: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:06.080022:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:06.086938:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:07.193609:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not d                                                                             efined
MESS:00:00:07.200726:0: *** Restart logging
MESS:00:00:07.202332:0: brfs: File read: 1889 bytes
MESS:00:00:07.207577:0: vec: vec_middleware_power_on: vec_base: 0x7ec13000 rev-i                                                                             d 0x00002708 @ vec: 0x7ec13100 @ 0x00000420 enc: 0x7ec13060 @ 0x00000220 cgmsae:                                                                              0x7ec1305c @ 0x00000000
MESS:00:00:07.227761:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:07.235079:0: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
MESS:00:00:07.238326:0: Loading 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xb96f
MESS:00:00:07.256319:0: brfs: File read: 47471 bytes
MESS:00:00:07.265944:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:07.320423:0: brfs: File read: 1423 bytes
MESS:00:00:07.324153:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:07.327949:0: dtparam: audio=on
MESS:00:00:07.339867:0: brfs: File read: 1889 bytes
MESS:00:00:07.350931:0: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d.dtbo
MESS:00:00:07.370217:0: Loaded overlay 'vc4-fkms-v3d'
MESS:00:00:07.408464:0: brfs: File read: 1238 bytes
MESS:00:00:07.411499:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:07.415624:0: Read command line from file 'cmdline.txt':
MESS:00:00:07.421507:0: 'root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=dead                                                                             line fsck.repair=yes rootwait'
MESS:00:00:08.468749:0: brfs: File read: 85 bytes
MESS:00:00:08.916565:0: brfs: File read: /mfs/sd/kernel7l.img
MESS:00:00:08.919205:0: Loading 'kernel7l.img' to 0x8000 size 0x5b7fc0
MESS:00:00:08.925462:0: Device tree loaded to 0x2eff4100 (size 0xbecd)
MESS:00:00:08.934464:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:08.940829:0: uart: Baud rate change done...
MESS:00:00:08.942889:0:▒

timg236, 2020-10-24:
That's more promising, you can see that firmware has successfully loaded the kernel. Enabling enable_uart=1 to config.txt will show the Linux boot races

This line indicates that there is a problem with the HDMI EDID I2C interface and the updated firmware is no longer stuck here.
HDMI1:EDID error reading EDID block 0 attempt 7

If you can take a high res photo of the top side of the board showing the HDMI connectors and the surrounding components then that would be useful

Adding hdmi_safe=1 to config.txt should cause something to appear on HDMI0

Or to force 1080p60
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=16 

For other mode choices see https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

daniframinan, 2020-10-24:
@timg236 i think it's the same

recover4.elf not found (6)
recovery.elf not found (6)
Read start4x.elf bytes  2985256 hnd 0x00033166 hash '6f5d6cc05fc28383'
Read fixup4x.dat bytes     8430 hnd 0x000348b7 hash '524d74b1ee853157'
0x00c03111 0x00000000 0x0000001f
MEM GPU: 128 ARM: 896 TOTAL: 1024
Starting start4x.elf @ 0xfec00200 partition 0

MESS:00:00:05.577380:0: arasan: arasan_emmc_open
MESS:00:00:05.727485:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.730797:0: brfs: File read: 1929 bytes
MESS:00:00:05.840663:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.865178:0: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:05.889695:0: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:05.914205:0: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:05.938722:0: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:05.963234:0: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:05.987751:0: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:06.012261:0: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:06.036778:0: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:06.061291:0: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:06.065794:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:06.072693:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:07.179417:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:07.186530:0: *** Restart logging
MESS:00:00:07.188140:0: brfs: File read: 1929 bytes
MESS:00:00:07.193384:0: vec: vec_middleware_power_on: vec_base: 0x7ec13000 rev-id 0x00002708 @ vec: 0x7ec13100 @ 0x00000420 enc: 0x7ec13060 @ 0x00000220 cgmsae: 0x7ec1305c @ 0x00000000
MESS:00:00:07.213563:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:07.220877:0: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
MESS:00:00:07.224123:0: Loading 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xb96f
MESS:00:00:07.242117:0: brfs: File read: 47471 bytes
MESS:00:00:07.251854:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:07.306331:0: brfs: File read: 1423 bytes
MESS:00:00:07.310059:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:07.313874:0: dtparam: audio=on
MESS:00:00:07.325768:0: brfs: File read: 1929 bytes
MESS:00:00:07.336800:0: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d.dtbo
MESS:00:00:07.356065:0: Loaded overlay 'vc4-fkms-v3d'
MESS:00:00:07.394336:0: brfs: File read: 1238 bytes
MESS:00:00:07.397368:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:07.401496:0: Read command line from file 'cmdline.txt':
MESS:00:00:07.407379:0: 'root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait'
MESS:00:00:08.489827:0: brfs: File read: 85 bytes
MESS:00:00:09.144109:0: brfs: File read: /mfs/sd/kernel7l.img
MESS:00:00:09.146766:0: Loading 'kernel7l.img' to 0x8000 size 0x5b7fc0
MESS:00:00:09.153024:0: Device tree loaded to 0x2eff4100 (size 0xbec9)
MESS:00:00:09.161575:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:09.168346:0: uart: Baud rate change done...
MESS:00:00:09.170378:0:$<$$8▒▒$▒#8▒$ǃ$<$$8▒▒$▒#8▒$ǃ$<$$8▒▒$▒#8▒$ǃ▒#▒%▒▒Ai▒▒▒#▒%▒▒Ai▒▒▒+▒▒▒▒Ai▒▒▒3!1%!ũ!V▒3!5%!ũ!V▒7!1%!ŭ!V`▒`▒▒|▒▒▒▒▒`▒▒▒▒▒▒`▒`▒▒|▒▒▒▒▒`▒▒▒▒▒▒`▒`▒▒|▒▒▒▒▒`▒▒▒▒▒▒▒x▒▒▒▒x▒▒▒x▒▒▒▒▒x▒▒▒▒▒▒x▒x▒▒x▒▒x▒▒▒x▒x▒▒▒▒x▒x▒x▒▒▒x▒▒x▒▒▒x▒x▒▒▒▒x▒x▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒

daniframinan, 2020-10-24:
@timg236 
[image]
[image]

daniframinan, 2020-10-25:
@timg236 This issue seems to have similar problems, should  I try the solution given by @pelwell?

daniframinan, 2020-10-27:
@timg236 What should I do?, I need the raspberry Pi working

JamesH65, 2020-10-27:
Have you ever tried a new SD card with an up to date installation of Raspberry Pi OS, to ensure the problem isn't Octopi? Just use Raspberry pi Imager and let it install the latest version.

wsegatto, 2020-10-30:
> @timg236 This issue seems to have similar problems, should I try the solution given by @pelwell?
> #1266

That solution did not work for me. I'm having the same issue as you are. I'm completely stuck and have no idea whether I should get a new Pi or not.

daniframinan, 2020-10-30:
@timg236 https://www.raspberrypi.org/forums/viewtopic.php?t=210129 how about I try this?

daniframinan, 2020-10-30:
@timg236 by the way what does the "Set PL011 baud rate to 103448.300000 Hz" command do? If tried connecting to that baud rate but the result is just garbage...

daniframinan, 2020-10-30:
@JamesH65 I have, it doesn't work

daniframinan, 2020-10-30:
@wsegatto Glad to know I'm not the only one, I was worried I'd be the only one with the problem and I'd never get help.

wsegatto, 2020-10-31:
You are not alone. The main thing I see is that I don't get the 4 green flashes when I have only the power plugged. Not sure the bootloader versions these days do not have the 4 flashes anymore so I don't know for sure. 

I got an UART today. I can only see the following, and then it stops and does nothing:

`recover4.elf not found (6)
recovery.elf not found (6)
Read start4.elf bytes  2278336 hnd 0x0001128e hash '78e6729ebb0d2e84'
Read fixup4.dat bytes     5418 hnd 0x00000399 hash '6f078b9acde7f5e1'
0x00c03112 0x00000000 0x0000001f
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0

MESS:00:00:05.368627:0: arasan: arasan_emmc_open
MESS:00:00:05.807544:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.810750:0: brfs: File read: 1813 bytes`

It doesn't matter the bootloader version, the files in the SD card, the many configurations to config.txt one can use, no joy. I think I'll have to buy another one and toss this one out.

wsegatto, 2020-10-31:
OMG OMG OMG!
I got somewhere!

I just copied over the files in my boot FAT32 partition with the ones I got from this website:
[https://github.com/raspberrypi/firmware/tree/master/](url) (download the ZIP, grab the entire contents of the boot folder and paste it over the current ones you got.

The boot went all the way, just got no HDMI yet, since config.txt has no HDMI settings now. Just the basics.
But I got here:
```
Debian GNU/Linux 10 raspberrypi ttyAMA0

raspberrypi login: pi
pi

Password: raspberry

Last login: Thu Aug 20 12:59:23 BST 2020 on tty1
Linux raspberrypi 5.4.73-v8+ #1360 SMP PREEMPT Thu Oct 29 16:00:37 GMT 2020 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.
```

wsegatto, 2020-10-31:
SOLVED!

I was able to rescue my OS by replacing the files in the boot partition as I explained before.
Extracting the start*.elf files got me to a certain point, but no "full boot" to login screen.
When I put the fix*.dat files to the boot partition and tried again, login screen!

They should definitely update this page, since I did not have the 4 green flashes with no SD card on but my pi is not "blown" as they say.
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=58151

daniframinan, 2020-10-31:
> OMG OMG OMG!
> I got somewhere!
>
> I just copied over the files in my boot FAT32 partition with the ones I got from this website:
> [https://github.com/raspberrypi/firmware/tree/master/](url) (download [...]

That website has no files, it says o results matched your search

daniframinan, 2020-10-31:
@JamesH65 Thank you so much, I have got it up and running (except the hdmi but I don't need it anyways). I haven't closed this issue in case anyone fixes the hdmi problem.

wsegatto, 2020-10-31:
Great!!! I’m so glad a newbie like me could help. 

Here’s config.txt in relation to HDMI. 

```
disable_overscan=1

[HDMI:0]
  hdmi_group=1
  hdmi_mode=16
  hdmi_drive=1
[HDMI:1]
  hdmi_group=1
  hdmi_mode=16
  hdmi_force_hotplug=1
  hdmi_force_mode=1
  hdmi_drive=1
```

3iljw, 2020-11-03:
I have a same issue with you.
when I use rpi-update then I can't enter my system any more.

wsegatto, 2020-11-03:
Backup your /boot files and try what I described above. Replace those files with the latest one from that site and check if the issue is solved.

petitsurfeur, 2020-11-19:
Hi
I have the same issue :( 

I received my RPI4B 3 weeks ago, 

Using a 16Gb sdcard (I did what was explained to boot from a SSD). I copy the content from https://github.com/raspberrypi/firmware to the sd partition (it's the Ubuntu Server 20.04_RPI-USB-BOOT.zip image but I also tried with the raspberry OS).

[image]

Here is my config.txt, I also tried adding your comment. What can I do

[pi4]
max_framebuffers=2
dtoverlay=vc4-fkms-v3d
boot_delay
kernel=vmlinux
initramfs initrd.img followkernel

[pi2]
kernel=uboot_rpi_2.bin

[pi3]
kernel=uboot_rpi_3.bin

[all]
arm_64bit=1
device_tree_address=0x03000000

enable_uart=1

disable_overscan=1

[HDMI:0]
  hdmi_group=1
  hdmi_mode=16
  hdmi_drive=1
[HDMI:1]
  hdmi_group=1
  hdmi_mode=16
  hdmi_force_hotplug=1
  hdmi_force_mode=1
  hdmi_drive=1

cmdline=cmdline.txt

include syscfg.txt
include usercfg.txt

Before it stop starting, I had  

$ vcgencmd bootloader_version
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
update-time 0
capabilities 0x00000000

$ vcgencmd bootloader_config
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
 
[all]
BOOT_ORDER=0xf41

What can I do ?

ret394, 2020-11-23:
> Using a 16Gb sdcard (I did what was explained to boot from a SSD).

🙀 I can guess. But who explained

vrescobar, 2020-12-17:
Hi @petitsurfeur, did you try copying the firmware onto the boot partition? (or sdcard)

These are the files you find in the boot folder from this repository:
https://github.com/raspberrypi/firmware

Amal5187, 2021-04-13:
hey i have the same issue can u please help

wsegatto, 2021-04-14:
> hey i have the same issue can u please help

https://github.com/raspberrypi/firmware/issues/1487#issuecomment-719867480

elgreg75, 2021-10-03:
Hello I just had the same problem tonight....

Amal5187, 2021-10-04:
> Hello I just had the same problem tonight....
Try this
https://github.com/raspberrypi/firmware/issues/1487#issuecomment-719867480

Raspi-Pi-4B-8GB, 2022-02-16:
I am using the Raspberry PI Imager as well as Etcher but had this black screen not booting condition on my Raspberry PI 4B no matter which distribution I burned. Luckily I still had an SD card with a working distro on it. So I began comparing the entries in the config.txt on the new burns to the working burn. And one by one I changed the settings until I found the one problem entry. I dont have the why its this way for you, only something you might look at and try.

The Raspberry PI Imager is putting the dtoverlay parameter in the config.txt file like this:

dtoverlay=vc4-kms-V3d

However, I noticed in the working build I had this entry had the letter “f” in front of the “kms” part. So I edited the config.txt and changed the dtoverlay entry as follows and the boot process continued and I was up and running on the new images:

dtoverlay=vc4-fkms-v3d

Maybe someone can explain this parameter and why the imagers are putting out a code that appears to hang up the Raspberry Pi 4B boot process. I just know this is how I finally got these images to boot. Hope this helps someone.

pelwell, 2022-02-16:
The `vc4-fkms-v3d` overlay enables the old firmware-based display driver; `vc4-kms-v3d` is for the new kernel-based display driver.

What display do you have attached? That is likely to be the trigger for the problem.

Raspi-Pi-4B-8GB, 2022-02-16:
Thank you for the reply and information.  The monitor is a Sansui TV/Monitor 19” combo.  I’m plugged into the HDMI port with my raspberry pi.    The tv/monitor has the rs232 port for traditional computer to monitor operation and it has setup options using the remote to adjust the monitor settings/format.  The hdmi port on the Tv/monitor doesnt have the same adjustment options.  

[image]

[image]

pelwell, 2022-02-16:
Using the vc4-fkms-v3d overlay, boot up the Pi 4 with the display attached, then run the following commands:
```
$ tvservice -d edid.dat
$ base64 edid.dat
```
Then post the output - it will look something like this:
```
AP///////wAQrF9BV1dHQRceAQOAPCJ47u6Vo1RMmSYPUFSlSwBxT4GAqUDRwAEBAQEBAQEBVl4A
oKCgKVAwIDUAVVAhAAAaAAAA/wBIRFA0WTEzCiAgICAgAAAA/ABERUxMIFUyNzE5RAogAAAA/QA4
TB5aGQAKICAgICAgAfQCAyLxT5AFBAMCBxYBBhESFRMUHyMJfweDAQAAZQMMABAAAjqAGHE4LUBY
LEUAVVAhAAAefjkAoIA4H0AwIDoAVVAhAAAaAR0AclHQHiBuKFUAVVAhAAAevxYAoIA4E0AwIDoA
VVAhAAAaAAAAAAAAAAAAAAAAAAAAAAAAAAAAiA==
```

Raspi-Pi-4B-8GB, 2022-02-16:
I'm getting this error message on the tvservice command:

[E] Failed to initialize VCHI (ret=-1)

I tried running as sudo and got this message:

vchi_msg_dequeue -> -1(90)
Nothing written!

pelwell, 2022-02-16:
Which kernel are you using (`uname -a`)? That should work on our kernels, even when running the cut-down firmware.

pelwell, 2022-02-16:
If you can ssh into your 4B then you may be able to grab the EDID (while using vc4-kms-v3d) with:
```
$ cp /sys/devices/platform/soc/soc:gpu/drm/card0/card0-HDMI-A-1/edid edid3
```
If it complains that the file doesn't exist, have a look within `/sys/devices/platform/soc/soc:gpu/drm/card0` to see if a similarly named directory does exist.

Raspi-Pi-4B-8GB, 2022-02-16:
5.13.0-1016-raspi #18-Ubuntu SMP PREEMPT Thu Jan 20 08:53:01 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux

pelwell, 2022-02-16:
Any luck with grabbing an EDID?

Raspi-Pi-4B-8GB, 2022-02-16:
just tried the first one and it did say no such file exists.

pelwell, 2022-02-16:
If you can't grab an EDID (and perhaps even if you can), the fastest way to a solution is probably going to be via one of our kernel builds. If you have spare SD card, consider installing the latest RPiOS - the newly out-of-beta 64-bit version in your case.

kimssj3918, 2022-08-12:
The same thing happens in my raspberry pi4. I put the Ubuntu 18.04 server image, but it doesn't run. how can i do??

JamesH65, 2022-08-12:
> The same thing happens in my raspberry pi4. I put the Ubuntu 18.04 server image, but it doesn't run. how can i do??

Does it work with Raspberry Pi OS? What display are you using (make, resolution etc)?

Raspi-Pi-4B-8GB, 2022-08-12:
I edited the config.txt and set the two parameters below to the values shown below.  

The dtoverlay parameter change made mine continue the boot process:

dtoverlay=vc4-kms-V3d    (dtoverlay was set to vc4-fkms-v3d  - remove the “f”)

I also altered the disable_overscan parameter to force a less than full screen display because in full screen mode some of the display image was out of view on the sides of the display.

disable_overscan=0      (disable_overscan was set to 1)

These changes solved the problem for both Ubuntu and Raspberry Pi images that just would not boot.

Chrisx19, 2022-09-01:
Sighhhhhh is there a proper way to fix this? I also have a Rpi4 -4gb with image ubuntu 18.04. I have the same problem as yall but when I try replacing the firmware with the repo provided above, it booted but its all messed up. I cant connect to my network at all.

popcornmix, 2023-01-24:
I'm going to close this as original reporter has bowed out, and it's been taken over by issues that seem unrelated.

If anyone still has an issue with a Pi not booting, or failing to display an image using a bullseye RPiOS image, then please create a new issue.

closed 2023-01-24 by popcornmix