raspberrypi/firmware #1488: Pi 4 with 2560x1440 monitor running max 30Hz/fps max

xlazom00, 2020-10-26:
**Describe the bug**
I have a monitor 2560x1440x60Hz
I have weston and I am running weston testing app ```/usr/lib/weston/weston-simple-egl```
and app is v-syncing on 30fps

I have running Raspbian OS ( kernel 5.4.72-v7l+)

**To reproduce**
run weston
run terminal
run /usr/lib/weston/weston-simple-egl

**Expected behaviour**
60fps

**Actual behaviour**
v-sync 30fps

**System**
https://pastebin.com/ECJb4DPb
Pi4

Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage2

vcgencmd version
Oct 22 2020 13:59:27
Copyright (c) 2012 Broadcom
version 74e754ff8947c58d2773253f77f6f68a303188f8 (clean) (release) (start)

Linux raspberrypi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux

When I overclock GPU/VPU/CPU nothing changed

MichaIng, 2020-10-26:
Please try to add `hdmi_enable_4kp60=1` to your `config.txt`: https://www.raspberrypi.org/documentation/configuration/config-txt/video.md
It's not a 4k monitor but more than 1080p at least, not sure where the boarder is for this setting to be required.

xlazom00, 2020-10-26:
@MichaIng same result

timg236, 2020-10-26:
The default pixel-frequency filter is <= 300MHz

xlazom00, 2020-10-26:
@timg236 so I will need to setup whole dpi_timings manauly?

alanbork, 2020-12-16:
Was this ever resolved?

popcornmix, 2023-01-24:
Running on bullseye with kms, and `video=HDMI-A-1:2560x1440@60D` in cmdline.txt works as expected. Picture on monitor and:
```
pi@pi4:~ $ kmsprint 
Connector 0 (32) HDMI-A-1 (connected)
  Encoder 0 (31) TMDS
    Crtc 3 (96) 2560x1440@59.95 241.500 2560/48/32/80/+ 1440/3/5/33/- 60 (59.95) U|D 
      Plane 3 (86) fb-id: 339 (crtcs: 3) 0,0 2560x1440 -> 0,0 2560x1440 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU12 YV12 NV12 NV21 NV16 NV61 P030 XR30 AR30 AB30 XB30 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 339 2560x1440 RG16
```
Please report if this is still an issue?

xlazom00, 2023-01-25:
Fixed

closed 2023-01-25 by xlazom00