raspberrypi/firmware #1489: Regression: Something strange in device tree handling - it gets clobbered

andreiw, 2020-11-04:
I work on the TianoCore UEFI that recently resulted in the Pi 4B getting Arm's SystemReady ES certification...

So this is what our config.txt looks like:
arm_64bit=1
enable_uart=1
uart_2ndstage=1
enable_gic=1
armstub=RPI_EFI.fd
disable_commandline_tags=1
disable_overscan=1
device_tree_address=0x1f0000
device_tree_end=0x200000
dtoverlay=miniuart-bt

..e.g. DTB is put right after the RPI_EFI.fd file.

In release 1.20 I noticed that the DTB header at 0x1f0000 was getting clobbered by something. Only UEFI release 1.18 was okay. But then I took RPI_EFI.FD from 1.20 and ran it with the rest of the bits of 1.18 - e.g the VPU firmware. And it was fine.

See https://github.com/pftf/RPi4/issues/107

Here are the MD5 sums for start4.elf. Anything special happen between 808c5e57a5740ab5d7b10e52b0e4dd54 and b85438bf1c23162513f017db385b06f5?

awarkentin-a02:Downloads awarkentin$ md5 RPi4_UEFI_Firmware_v1.20/start4.elf
MD5 (RPi4_UEFI_Firmware_v1.20/start4.elf) = 84deebfa49d19f5202f2d1e753718bfe

awarkentin-a02:Downloads awarkentin$ md5 RPi4_UEFI_Firmware_v1.19/start4.elf
MD5 (RPi4_UEFI_Firmware_v1.19/start4.elf) = b85438bf1c23162513f017db385b06f5

awarkentin-a02:Downloads awarkentin$ md5 RPi4_UEFI_Firmware_v1.18/start4.elf
MD5 (RPi4_UEFI_Firmware_v1.18/start4.elf) = 808c5e57a5740ab5d7b10e52b0e4dd54

pelwell, 2020-11-04:
Without dates it's going to be tedious trying to identify those firmwares - `strings start4.elf | grep VC_BUILD` is much more useful.

There was an issue that affected firmwares built between July 31st and October 14th that would only appear if using an external armstub file; the common symptom was corruption of the DTB.

pbatard, 2020-11-04:
Yes, I am fairly confident that this the same as the issue from #1445, as the problem with the Pi 4 UEFI firmware happened around the time we picked the DTB corruption with the Pi 3, and since we mostly use ACPI on the Pi 4, we haven't updated the core files in our published releases yet.

The corruption is pretty much the same thing as what we observed on the Pi 3, and since my understanding is much of the code is factorized, it's logical that the Pi 4 would be affected in the same manner.

We're going to test with the latest `start4.elf` and confirm that the problem is fixed.

pbatard, 2020-11-06:
Sorry for the delay. I can confirm that, as suspected, this is the same as #1445 and that the latest `start4.elf` fixes the problematic behaviour observed, by going through a full Debian 11 install in Device Tree mode, which wouldn't even start installing with the older version of `start4.elf`.

With the current version of `start4.elf`, Linux happily accepts the Device Tree being provided, and I see no sign of Device Tree corruption. Therefore, I believe this issue can be closed.