raspberrypi/firmware #1495: XHCI_RESET mailbox command appears to assume a particular RC_BAR2 state

andreiw, 2020-11-15:
Dear Raspberry Pi firmware developers,

I work on the SystemReady ES-certified UEFI firmware for the Pi 4 (https://github.com/pftf/RPi4). We have a problem where xHCI controlled fails to be initialized on CM4 and Pi 400, yet works on the Pi 4B. The difference is of course the subtle change to the SoC (3GB DMA bug vs full DMA).

The problem? UEFI xHCI driver would timeout trying to reset xHCI controller, as no firmware would actually be loaded into it.

It helps to understand that we initialize the RC in a particular way - the inbound window has no translation. Bus address == CPU address. This worked well.

And by the way, the actual device I did my experiments on is a Pi 4 with the new SoC and 8GiB (in case anyone wonders why I keep talking about Pi 400 and 8GiB).

So in trying to analyze the Pi 400/CM4 bug, I first looked at the FDT. The FDT has a dma-ranges property that pretty clearly showed an IB with a translation (on my 8GiB Pi - an 8GiB translation, on 4GiB Pi - a 4GiB translation). So this is what I did:

So I changed my code from...
- PCIE_MISC_RC_BAR2_CONFIG_LO = 0x10 /* Size = 2G */
- PCIE_MISC_RC_BAR2_CONFIG_HI = 0x0     /* Offset = 0G */

...to
- PCIE_MISC_RC_BAR2_CONFIG_LO = 0x10 /* Size = 2G */
- PCIE_MISC_RC_BAR2_CONFIG_HI = 0x2     /* Offset = 8G */

...and lo and behold, the xHCI driver in UEFI was able to reset the controller. Of course, DMA didn't work, as I UEFI was still expecting 1:1 cpu<-> bus address translations.

So now I tried the following:
- after XHCI_RESET, wait for 2000us (~ 2 x more than xHCI fw load should take)
- set PCIE_MISC_RC_BAR2_CONFIG_HI to = 0x0

...unfortunately, that didn't help. Perhaps RC doesn't like IB window being reconfigured without a reset? Perhaps firmware load into VL805 leaves some crumbs with the device "knowing" where the memory window is? Hard to tell. What's also curious also is that the same code on a Pi 4B is fine...perhaps due to the PCIe bug (limiting DMA to 3GB) the high bits of the IB window get entirely ignored anyway.

**It seems like the VPU is assuming what the IB window is, instead of reading this from RC_BAR2. That's where "the bug" is. Now in UEFI we don't want to honor the FDT config. Limiting DMA size is much less weirder than a DMA translation! Could you kindly change the XHCI_RESET implementation to honor the real RC_BAR2 configuration?**

timg236, 2020-11-16:
"We have a problem where the xHCI controller fails to be initialized on CM4"

CM4 (https://www.raspberrypi.org/products/compute-module-4/?variant=raspberry-pi-cm4001000&resellerType=home) doesn't have an XHCI controller. If there is an external PCIe controller then firmware knows not to do any special initialisation even if it is a VL805. 

Pi4 B < revision 1.4 or older (1GB, 2GB, 4GB) have a dedicated VLI SPI EEPROM under the control of the VLI. The firmware doesn't get involved in this so the XHCI RESET mailbox call is a no-op.

Pi4 8GB does not have a dedicate VLI SPI EEPROM and so the VLI runs the firmware from SDRAM in a manner similar to SPI cached in SPI RAM. This means that the xHC must be reset and the VLI FW reloaded if the PCI BAR configuration is changed.

"we don't want to honor the FDT config."

The normal way to modify the FTD is via a dtoverlay. Please can you try that or provide more information as to why that approach is not suitable.

andreiw, 2020-11-18:
Hi Tim,

Some clarifications. While I acknowledge what you've said about CM4, this is a problem with the Raspberry 400. I don't have a Raspberry 400, but I have a "special" Pi 4B with the updated SoC  - like a regular 8GB Pi 4, it does not carry a VLI SPI EEPROM. Unlike a Regular Pi 4B, it has the same chip as CM4 or Raspberry 400.

I acknowledge your question around trying a modified FDT with a modified IB window there. I'll let you know how that goes. Does this mean that the VPUs "view" into RC configuration is not hardcoded, but driven by the DT as well?

andreiw, 2020-11-18:
So here's the "stock" DTB from SD card decompiled:

dma-ranges = <0x2000000 0x0 0x0 0x0 0x0 0x0 0xc0000000>; 

... i.e in-bound window PCI 0x0 -> CPU 0x0, size 3GiB

Here's the DTB after booting Raspbian:

dma-range = <0x2000000 0x02 0x00 0x00 0x00 0x02 0x00> 

... i.e in-bound window PCI 0x200000000 -> CPU 0x0, size 8GiB

...so I'm not sure what I'm supposed to do. The original DTB already has the configuration I am interested in. The VPU firmware appears to patch to the shifted IB window values.

vianpl, 2020-11-19:
AFAIK the DT patching by VC4 happens during boot based on the SoC revision. So it will not honor whatever you put in originally. On the other hand, dtoverlays happen after the fact, so you could use an overlay to resize your 'dma-ranges'.

If that's too hacky, why not get the SoC revision and add inbound window size rules based on that.

timg236, 2020-11-19:
I think this updated firmware may help
https://www.raspberrypi.org/forums/viewtopic.php?p=1759627#p1761834

andreiw, 2020-11-19:
> AFAIK the DT patching by VC4 happens during boot based on the SoC revision. So it will not honor whatever you put in originally. On the other hand, dtoverlays happen after the fact, so you could use [...]

Again, the problem isn't what we do, the problem was assumptions by XHCI_RESET around what the inbound configuration was (assumption that the OS has to honor the FDT).

andreiw, 2020-11-19:
> I think this updated firmware may help
> https://www.raspberrypi.org/forums/viewtopic.php?p=1759627#p1761834

Will give it a spin! What did you change? Does this change the inbound window base?

andreiw, 2020-11-19:
> AFAIK the DT patching by VC4 happens during boot based on the SoC revision. So it will not honor whatever you put in originally. On the other hand, dtoverlays happen after the fact, so you could use [...]

That's all fine, but will XHCI_RESET honor the dma-ranges as overridden by my dtbo? If it's still going to assume the inbound window is at 8GB, it's not going to work.

timg236, 2020-11-19:
XHCI_RESET gets the inbound window from the device tree layer which returns the inbound window base. AIUI if this is specified in DT (or via an overlay) it should override the firmware default inbound window base

cleverca22, 2020-11-21:
ive also been documenting the vl805 over in https://github.com/librerpi/rpi-open-firmware/blob/master/docs/vl805.txt

including what the XHCI_RESET mailbox is doing to load `vl805hub.bin` and `vl805mcu.bin` into the controller

andreiw, 2021-03-08:
And it's now fixed (same as #1518

closed 2021-03-08 by andreiw