raspberrypi/firmware #1496: Is there a Pi400 config.txt conditional filter, I don't see it documented

vianpl, 2020-11-16:
I can't seem to find a Pi400 config.txt conditional filter. The documentation I refer to is https://www.raspberrypi.org/documentation/configuration/config-txt/conditional.md. If non-existent, I'd appreciate it could be introduced (and the same goes for CM4).

Thanks for everything :slightly_smiling_face:

JamesH65, 2020-11-16:
There isn't one, and unlikely to be one for a while - there is some technical stuff we need to figure out.

vianpl, 2020-11-16:
Fair enough, should I close the issue or you're OK if I keep it open as a reminder?

popcornmix, 2020-11-16:
Open is fine until we have a decision.

MichaIng, 2020-12-04:
Don't they respect the `[Pi4]` filter, same as `[pi1]` and `[pi3]` include the respective compute modules due to identical SoC? Individual model differentiation of course would be nice as well, but I don't see something missing here, but instead this being an extension/addition to the current conditional filter feature.

pelwell, 2020-12-04:
Yes - currently `[pi4]` is effectively `[bcm2711]` (which doesn't exist), so it includes CM4 and Pi 400.

MichaIng, 2020-12-09:
Now there are [pi400] and [cm4] conditional filters, if I interpret the latest commit correctly? https://github.com/raspberrypi/firmware/commit/919aee0ed75f7db48a38b8b96c13228a7584cfd7
> firmware: config: Add cm4 and pi400 config section filters

👏

timg236, 2020-12-09:
Yes, that's correct. The next Pi 4 bootloader beta will also support those for advanced configuration like gpu_mem / start_file. I think that will also be the next stable bootloader release unless there are major regressions.

pelwell, 2020-12-09:
Note that a `[pi4]` is true for Pi 400 and CM4, but `[pi400]` and `[cm4]` are only true for Pi 400 and CM4 respectively. If you want something to only apply to a Pi 4 you'll currently have to revert the settings on Pi 400/CM4.

MichaIng, 2020-12-09:
That is great, so one can e.g. define different overclocking for the Pi 400 that is running on higher frequency by default (1500 vs 1800 MHz) and skip configuring onboard audio on CM4 and such. Many thanks for implementing this!

timg236, 2020-12-09:
The next rpi-update firmware will also include a change where board-specific changes can be stored in the EEPROM config and automatically appended to config.txt at runtime.

E.g. In the PI400 EEPROM you could have something like

```
# Bootloader will stop here
[none]

# Start.elf will augment config.txt at runtime with this
[config.txt]
arm_freq=2000
over_voltage=4
```

Although, you need to be a bit careful when configuring this and have a recovery image on standby :)

zyga, 2021-11-15:
You can use the board type filter to have pi400-specific conditional filter.

timg236, 2022-11-28:
Fixed

closed 2022-11-28 by timg236