raspberrypi/firmware #1497: RTL8152 new realtek driver make problem with 5.4.75+ firmware/kernel

bogaskilop, 2020-11-16:
**Describe the bug**
I have  120 pcs of raspberry pi zero 1.3   with installed debian strech  all updates done also rpi-update  so now i have kernel 5.4.75+  and for those  rpi zeros i have  lan adapters  140 pcs RTL8152B    and also few asix AX8872A   and  AX8872BFL and CFL versions for test.  but the issue is  that new kernel has also old realtek driver   r8152 1-1:1.0 eth0: v1.10.11

And during boot time on such an amoutn of rpi zeros there are issues with geting  lan connection to those rpi and i need to just power them off and power on to get acces to them . i also using 
4.14.78+  for tests but i am also unable to add  new RTL 8152  driver  to kernel 

so rpi in such moments  boots up lan internface is up as other scripts are runing on it  recognising IP addres etc etc  but there is no network so reboot required again .

here is a dmsg as a result of using a simple script to reset usb and unload  kernel module and reinitialize network :

systemd[1]: Listening on fsck to fsckd communication Socket.
[    4.186894] systemd[1]: Listening on Journal Socket.
[    4.637125] usbcore: registered new interface driver r8152
[    4.653016] usb 1-1.1: reset low-speed USB device number 4 using dwc_otg
[    4.963001] usb 1-1.1: device descriptor read/64, error -71
[    5.373034] usb 1-1.1: device descriptor read/64, error -71
[    5.783054] usb 1-1.1: reset low-speed USB device number 4 using dwc_otg
[    6.123011] usb 1-1.1: device descriptor read/64, error -71
[    6.533012] usb 1-1.1: device descriptor read/64, error -71
[    6.763040] usb 1-1.4: reset high-speed USB device number 5 using dwc_otg
[    6.940559] r8152 1-1.4:1.0 eth0: v1.09.9
[    7.183042] usb 1-1.1: reset low-speed USB device number 4 using dwc_otg
[    7.622985] usb 1-1.1: device not accepting address 4, error -71
[    7.624016] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    7.913057] usb 1-1.1: reset low-speed USB device number 4 using dwc_otg
[    8.353035] usb 1-1.1: device not accepting address 4, error -71
[    8.356404] usb 1-1.1: USB disconnect, device number 4
[    8.452998] usb 1-1.1: new low-speed USB device number 6 using dwc_otg
[    8.509309] systemd-journald[80]: Received request to flush runtime journal from PID 1
[    8.553076] usb 1-1.1: device descriptor read/64, error -71
[    8.773057] usb 1-1.1: device descriptor read/64, error -71
[    8.993039] usb 1-1.1: new low-speed USB device number 7 using dwc_otg
[    9.093021] usb 1-1.1: device descriptor read/64, error -71
[    9.313073] usb 1-1.1: device descriptor read/64, error -71
[    9.433440] usb 1-1-port1: attempt power cycle
[   10.093026] usb 1-1.1: new low-speed USB device number 8 using dwc_otg
[   10.533024] usb 1-1.1: device not accepting address 8, error -71
[   10.633059] usb 1-1.1: new low-speed USB device number 9 using dwc_otg
[   11.073036] usb 1-1.1: device not accepting address 9, error -71
[   11.073512] usb 1-1-port1: unable to enumerate USB device
[   11.316200] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[   11.381952] bcm2835_alsa bcm2835_alsa: card created with 8 channels
[   17.641500] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   46.140745] r8152 1-1.4:1.0 eth0: carrier on
[   46.141047] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   50.575000] random: crng init done
[   50.575024] random: 7 urandom warning(s) missed due to ratelimiting

basicly  booting  rpi zero with usb lan adapter is a lotery to get it work properly  so i wanted to install new  REALTEK driver 
from 
https://www.realtek.com/en/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-usb-3-0-software

i installed  headers but there is no build folder in 5.4.75+ 

so i wanted to try  : 
/home/pi/r8152-2.14.0# make
make -C /lib/modules/5.4.75+/build M=/home/pi/r8152-2.14.0 modules
make[1]: *** /lib/modules/5.4.75+/build: No such file or directory.  Stop.
Makefile:24: recipe for target 'modules' failed
make: *** [modules] Error 2

 vcgencmd version
Nov  6 2020 18:01:30
Copyright (c) 2012 Broadcom
version 9ae94aeb1241adffec586ae451862e3fa6e05e6a (clean) (release) (start_x)

 uname -a
Linux CAM100 5.4.75+ #1367 Mon Nov 9 15:00:41 GMT 2020 armv6l GNU/Linux

 cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

how can i add this new  module to that kernel  ?
or can you just add this  so every one else will not have problem with rtl8152 lan adapters and lan+hub adapters ?

thx Michal

FD-, 2021-02-26:
I reported a very similar problem in 2019: https://github.com/raspberrypi/linux/issues/3310

Averell7, 2021-09-10:
See here :
https://github.com/raspberrypi/firmware/issues/1367
There is a good solution which worked for me. I added 
`usb_modeswitch -v 0BDA -p 8151 --reset-usb`
(adapt to your devices)
in rc.local, added a sleep 10 and then restarted the affected services. 
This is a dirty workaround, but it works.