raspberrypi/firmware #1500: Enable USB by default on the CM4

PTS93, 2020-11-25:
This is causing headache for so many people as this isn't all that obvious that for some reason it was chosen to not enable USB by default on the CM4. Out of all peripherals USB is probably one of the most used ones next to a UART boot console. Disabling that by default but having peripherals like Ethernet and HDMI running at the same time is a bit puzzling.

For the sake of possibly thousandth of cumulative hours spent by people being frustrated that their USB devices aren't working before realising that USB is simply disabled I would advocate that this is either enabled again in the firmware so that the CM4 has the same behaviour as a Pi4 by default or have the DT overlay in config.txt added by default and whoever wants it disabled can comment that out.

ccoenen, 2020-11-26:
Knowing that earlier would have saved me a couple of hours of pulling my hair out. What's the reasoning behind deactivating it in the first place?

pelwell, 2020-11-26:
Ironically, what you are asking for is to make CM4 _less_ like a Pi 4.

Using a default Raspberry Pi OS image, the 2711 on a Pi 4 by has PCIe enabled and both onboard USB interfaces disabled. The same is true for a CM4. If you plug an XHCI PCIe card into the slot on a CM4IO carrier, what you get is very close to a Pi 4 - I know because I've tried it. Like a Pi 4, if you want to enable either of the onboard USB 2 interfaces you will have to use `dtoverlay=dwc2`, adding `otg_mode=1` to select the newer XHCI USB 2 controller.

Your problem is that you are confusing the main USB ports on a Pi 4 with the OTG functionality available on the USB C socket - that's what the CM4IO board gives access to. If we were to enable that by default - something that would need to be done in the CM4 Device Tree since config.txt is common to all Pis - we would also need to add a CM4-specific method to _disable_ that OTG interface, since you can't delete or comment out config settings that aren't there, making it less like a Pi 4.

PTS93, 2020-11-26:
I see, then I misunderstood the USB setup on the Pi4.
Either way, what is currently the default for the CM4 is incredibly frustrating for users. Whatever a possible way is to have it enabled by default would be very welcome.

pelwell, 2020-11-26:
It's on page 5 of the documentation, and by creating this issue you are helping to publicise the fact, but I don't think the default is going to change.

PTS93, 2020-11-29:
I feel that would be short sighted to keep this default for the CM4 if all that is necessary is a change in the device tree and inverting the setting for the config.txt.
Apart from people interacting with the dev kit, this poses a problem for people who wish to create hardware based on the CM4 that is meant to be used in a similar setting as the Pi4 SBC.
Most end users will likely gloss over such an instruction to add some obscure line of text to a text file after flashing.
So for a smooth UX you would have to offer your own Raspbian images just to have USB enabled by default which I think isn't in the interest of anyone.

pelwell, 2020-11-29:
It was always the intention that producers of Compute Module-based products would provide either a custom DTB or an overlay. I'm not expecting many carriers that don't require some kind of bespoke configuration, and this is just another part of that.

ret394, 2020-12-01:
mark https://github.com/raspberrypi/firmware/issues/1500#issuecomment-734013796 as off topic please, it's hard to read the thread.

Completely reasonable request, but please wait a bit while I make an issue here for raspi 4's USB Boot use-case. See  https://github.com/raspberrypi/firmware/issues/1506

d-c-d, 2021-04-05:
In preparation for my Piunora, I started trying to get my CM4 Lite to boot from NVMe.
While I did see that raspi-config gave the advance option to boot from SD then USB, or boot from USB, otherwise boot from SD card.  I get the feeling that there are improvements that could improve the CM4 and CM4 Lite boot experience.

timg236, 2022-11-28:
USB host mode is enabled by default via otg_mode=1 in the default Raspberry Pi OS config.txt

closed 2022-11-28 by timg236

scotcha1, 2022-11-29:
I ran into lots of different challenges with this when i tried to install Home Assistant OS directly 

- on arrival, i was surprised to find out that USB devices were disabled (sure its in the datasheet, but i didnt read it like everyone else)
- i tried to SSH in to the device, but i needed to set it up correctly. since there's no SSD card, i couldn't do it as easily
- i tried to use the micro USB as a way to slave into the device, and the look at the eMMC boot folder. i had to make it so the 2 pins were connected (used some metal wire around the house) I tried this on both Mac & Ubuntu but I was unable to get it to show up as 'boot' partition
- i ended up purchasing a PCIe to USB 3.2 Gen1 Adapter for Raspberry Pi and plugged it in, and fixed the issues with not being able to see my Sonoff USB 3.0 Zigbee device. I imagine the next thing i could do is plug in a keyboard/usb mouse and just update it directly without SSH