raspberrypi/firmware #1501: bcm2710-rpi-3-b-plus.dtb

AradoSKY, 2020-11-26:
Taking backup of bcm2710-rpi-3-b-plus.dtb.
Installing new bcm2710-rpi-3-b-plus.dtb.
ln: failed to create symbolic link '/boot/dtb-': Operation not permitted
dpkg: error processing package flash-kernel (--configure):
 installed flash-kernel package post-installation script subprocess returned error exit status 1

This DTB script does not account for /boot/dtb-1. The script truncates the LN command causing the process to fail. This specifically impacts the Raspberry PI 3B+ running UBUNTU Mate.

pelwell, 2020-11-26:
Which script do you think is failing?

AradoSKY, 2020-11-26:
Hello,

The script which creates the symbolic link to a truncated directory path. What I provided is the actually syntax from this package executing. The package fails.

pelwell, 2020-11-26:
What command did you type?

AradoSKY, 2020-11-26:
Hello,

I ran “Apt-get update” then “apt-get dust-upgrade”. This is part of a firmware package.

AradoSKY, 2020-11-27:
Hello,

I must of accidentally closed this problem.

AradoSKY, 2020-11-30:
Hello,

The RPI3B+ boot directory has two dtb directories.

/boot/dtb
/boot/dtb-1 (the script fails establishing a symbolic link here)

AradoSKY, 2020-12-07:
root@astroberry:/home/astroberry# apt-get dist-upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up flash-kernel (3.98ubuntu12) ...
Using DTB: bcm2710-rpi-3-b-plus.dtb
Installing /etc/flash-kernel/dtbs/bcm2710-rpi-3-b-plus.dtb into /boot/dtbs//./bcm2710-rpi-3-b-plus.dtb
Taking backup of bcm2710-rpi-3-b-plus.dtb.
Installing new bcm2710-rpi-3-b-plus.dtb.
ln: failed to create symbolic link '/boot/dtb-': Operation not permitted
dpkg: error processing package flash-kernel (--configure):
 installed flash-kernel package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 flash-kernel
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@astroberry:/home/astroberry# 

I have added the recent attempt to install this package. The package is not fixed. I am seeking the source files to see which script is failing to see /boot/dtb-1.

XECDesign, 2020-12-07:
> Setting up flash-kernel (3.98ubuntu12) ...

That doesn't seem to be our package or even our distro.

closed 2020-12-07 by pelwell

AradoSKY, 2020-12-08:
Hello,

Hmm... Raspberry PI and ARMFH are the basis of UBUNTU MATE. The source of this package is here. This is an OS need and not a distribution need. Look again.

UBUNTU MATE and Raspberry PI pull from their respective repositories. ARMFH are their targets. This is a boot manager applet to boot an RPI3B+. Just because the included evidence contains a distribution name does not give justification to dump it in the distribution’s lap. An OS patch is a Raspberry PI Patch.

XECDesign, 2020-12-08:
> The source of this package is here.

The error is from the `flash-kernel` package, which is not hosted here.

pelwell, 2020-12-08:
```
pi@raspberrypi:~$ apt info flash-kernel
Package: flash-kernel
Version: 3.99
Priority: optional
Section: utils
Maintainer: Debian Install System Team <debian-boot@lists.debian.org>
Installed-Size: 188 kB
Depends: debconf (>= 0.5) | debconf-2.0, devio, initramfs-tools (>= 0.92f), linux-base (>= 3.2), mtd-utils, ucf
Recommends: u-boot-tools
Download-Size: 50.7 kB
APT-Sources: http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
Description: utility to make certain embedded devices bootable
 flash-kernel is a script which will put the kernel and initramfs in
 the boot location of embedded devices that don't load the kernel and
 initramfs directly from /boot.  flash-kernel supports devices that
 boot from flash memory (hence the name) as well as some devices that
 require a special boot image on the disk.
```

XECDesign, 2020-12-08:
Just a heads up that the package provided by Canonical has additional patches that are meant to handle the kernel install in the way that they prefer and that's what's falling over. So Debian maintainers won't be able to help either. This is specific to Ubuntu.

AradoSKY, 2021-04-25:
Hello,

This is bogus. The kernel supplied is from Raspberry to work with Ubuntu.  Given the to volume method to run Raspberry PI. the FAT vs EXT dance is what Raspberry is supposed to know about. 

However, I looked into this more. The problem is multiple issues. One is syntax. The other is FAT. It is as if Raspberry PI has forgotten its roots. 

ln: failed to create symbolic link '/boot/dtb-/bcm2710-rpi-3-b-plus.dtb': Operation not permitted is the message. '/boot/dtb-/bcm2710-rpi-3-b-plus.dtb' is one continuous string. I have looked at the script used to install. I have created a 'dtb-' directory to see if this resolves the Syntax error. Apparently, DTB is a logical name. Nothing can be copied into a DTB directory. If nothing can be created, then why include this path into the script?  

This says the script is broken. Creating a link is not possible with DTB in the link. Yes. DTB is in the FAT side. The folder is /var/lib/dpkg/info/flash-kernel.  Every file in this folder has been viewed under Pluma. Postinst contains the script. The other files are variable and settings. The following are the directories. 

**astroberry@astroberry:~$ ls -all /boot/dtbs/**  
total 80
drwxr-xr-x  3 root root  4096 Apr 25 18:09 .
drwxr-xr-x 11 root root 16384 Dec 31  1969 ..
drwxr-xr-x  2 root root  4096 Oct 20  2019 1
-rwxr-xr-x  1 root root 27054 Apr 25 18:09 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x  1 root root 27054 Apr 25 17:42 bcm2710-rpi-3-b-plus.dtb.bak
**astroberry@astroberry:~$ ls -all /boot/dtb**    
total 20
drwxr-xr-x  2 root root  4096 Sep  2  2020 .
drwxr-xr-x 11 root root 16384 Dec 31  1969 ..
**astroberry@astroberry:~$ ls -all /boot/dtb-1**
total 20
drwxr-xr-x  2 root root  4096 Oct 20  2019 .
drwxr-xr-x 11 root root 16384 Dec 31  1969 ..
astroberry@astroberry:~$ ls -all /boot/dtb- 
total 20
drwxr-xr-x  2 root root  4096 Apr 22 22:54 .
drwxr-xr-x 11 root root 16384 Dec 31  1969 ..
astroberry@astroberry:~$ ls -all /boot/dtbs//.
total 80
drwxr-xr-x  3 root root  4096 Apr 25 18:09 .
drwxr-xr-x 11 root root 16384 Dec 31  1969 ..
drwxr-xr-x  2 root root  4096 Oct 20  2019 1
-rwxr-xr-x  1 root root 27054 Apr 25 18:09 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x  1 root root 27054 Apr 25 17:42 bcm2710-rpi-3-b-plus.dtb.bak
astroberry@astroberry:~$ ls -all /boot/dtbs/1
total 64
drwxr-xr-x 2 root root  4096 Oct 20  2019 .
drwxr-xr-x 3 root root  4096 Apr 25 18:09 ..
-rwxr-xr-x 1 root root 27054 Oct 20  2019 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x 1 root root 27054 Oct 20  2019 bcm2710-rpi-3-b-plus.dtb.bak

This shows that the Kernel gets backuped to /boot/dtbs/1.  The claim is only DTBS has EXT as its file system. Notice the other DTB folders are not like DTBS.  Nothing can be copied into them. With ls -all, root is the owner for /BOOT/*.* As shown, this script looks broken. With this kernel broken, no upgrades are possible as this psuackage errors out. This stops the move to a new OS. Since November, any update fails because of the FLASH-KERNAL package.

juergh, 2021-04-26:
This is the wrong place for reporting flash-kernel issues. Please use https://launchpad.net/ubuntu/+source/flash-kernel.