raspberrypi/firmware #1505: PI4: fkms/drm not setting refresh correctly or tvservice not detecting it

alanbork, 2020-12-01:
using the official DRM tests package, specifically unmodified modetest.c, I can tell the DRM subsystem to set the display mode to a NTSC refresh rate, ie 59.94, like this:

root@raspberrypi:~/drm-master/builddir/tests/modetest# modetest -s 89:#6
trying to open device 'i915'...failed
trying to open device 'amdgpu'...failed
trying to open device 'radeon'...failed
trying to open device 'nouveau'...failed
trying to open device 'vmwgfx'...failed
trying to open device 'omapdrm'...failed
trying to open device 'exynos'...failed
trying to open device 'tilcdc'...failed
trying to open device 'msm'...failed
trying to open device 'sti'...failed
trying to open device 'tegra'...failed
trying to open device 'imx-drm'...failed
trying to open device 'rockchip'...failed
trying to open device 'atmel-hlcdc'...failed
trying to open device 'fsl-dcu-drm'...failed
trying to open device 'vc4'...done
setting mode 720x480-59.94Hz on connectors 89, crtc 87

while modetest still has control of the screen I can use tvservice to query it's state:

tvservice -s
state 0x6 [DVI CUSTOM RGB full 16:9], 1280x720 @ 60.00Hz, progressive

as you can see it's correct about the resolution but not the refresh rate. I've tested other modes and tvservice is always right about the resolution, but never thinks that it's the NTSC refresh rate. I don't currently have the right equipment to tell what the display is actually outputting,

uname -a
Linux raspberrypi 5.4.61-v7l+ #1339 SMP Tue Sep 1 18:51:27 BST 2020 armv7l GNU/Linux

6by9, 2020-12-02:
tvservice works on the standard modes but with a flag to say rescale the clock for NTSC timings.
If you look at the source code, it even does the munge in the app - https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/tvservice/tvservice.c#L468
The frame rate passed up from the VPU is only an integer, so it can't show any additional precision.

DRM works by duplicating all the modes as a normal and NTSC variant, and will then set the exact mode that is requested. That's why modetest will list for example
```
  #8 1920x1080 60.00 1920 2008 2052 2200 1080 1084 1089 1125 148500 flags: phsync, pvsync; type: driver
  #9 1920x1080 59.94 1920 2008 2052 2200 1080 1084 1089 1125 148352 flags: phsync, pvsync; type: driver
 ```
148500 / 1.001 = 148532.
tvservice gets told the full mode, the internal flag never gets set, 

Tested with an HDMI analyser and it reports Vfreq of 59.9411Hz, clock freq of 148.354MHz for the above mode 9.

tvservice should not be used with DRM as it may give misleading answers.

alanbork, 2020-12-02:
thanks for that info & good to know that the NTSC refresh rates are really getting applied.

Is there a DRM alternative to TV service for querying the floating point value of the refresh rate from the command line or file system? In addition to the NTSC stuff I'm working on stuffing some custom modelines into DRM and it would be nice to double check that I'm getting the refresh rate I'm expecting, without requiring a fancy HDMI analyzer.

popcornmix, 2020-12-03:
There may be better ways but I use this:
```
if sudo test -f "/sys/kernel/debug/dri/1/state"; then
 F=/sys/kernel/debug/dri/1/state
else
 F=/sys/kernel/debug/dri/0/state
fi
sudo cat $F |grep -A 20 "crtc\[" | grep -A 20 enable=1 | grep mode:
```
This gives:
```
mode: "1920x1080": 60 148500 1920 2008 2052 2200 1080 1084 1089 1125 0x40 0x5
```
for 1080p60 and:
```
mode: "1920x1080": 60 148352 1920 2008 2052 2200 1080 1084 1089 1125 0x40 0x5
```
for 1080p59.94. You could probably calculate the frame rate from the ```clock / ((sum horizontal timings) * (sum vertical timings))```

6by9, 2020-12-03:
> You could probably calculate the frame rate from the ```clock / ((sum horizontal timings) * (sum vertical timings))```

You can, and DRM uses cumulative counts for the timings, so that line even gives you the numbers in the mode line. eg:
```(148352 * 1000) / (2200 * 1125) = 59.94```

alanbork, 2020-12-03:
Thanks for the detailed answer, works for me!

closed 2020-12-03 by alanbork