raspberrypi/firmware #1507: Framebuffer is registered regardless of max_framebuffers=0

MichaIng, 2020-12-06:
**Describe the bug**
The [documentation](https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/video.md#max_framebuffers) states that setting `max_framebuffers=0` in `config.txt` "will prevent any framebuffers from being allocated.". But regardless of this setting being set, during boot `bcm2708_fb` loads, finds the attached display and registers a framebuffer for it, resulting in `/dev/fb0` being present.

**To reproduce**
```
echo 'max_framebuffers=0' >> /boot/config.txt
reboot
dmesg | grep bcm2708_fb
ls -l /dev/fb0
```

**Expected behaviour**
No framebuffer is registered.

**Actual behaviour**
A framebuffer is registered.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? **Raspberry Pi 2 Model B Rev 1.1**
* Which OS and version (`cat /etc/rpi-issue`)? **Debian Bullseye with official archive.raspberrypi.org firmware/kernel packages.**
* Which firmware version (`vcgencmd version`)?
```
Nov 30 2020 22:14:05
Copyright (c) 2012 Broadcom
version ab1181cc0cb6df52bfae3b1d3fef0ce7c325166c (clean) (release) (start_cd)
```
* Which kernel version (`uname -a`)?
```
Linux micha.gnoedi.org 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux
```

**Logs**
```
2020-12-06 18:26:47 root@micha:~# grep max_framebuffers /boot/config.txt
max_framebuffers=0
2020-12-06 18:27:12 root@micha:~# dmesg | grep bcm2708_fb
[    0.000000] Kernel command line: coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=16 bcm2708_fb.fbheight=16 bcm2708_fb.fbdepth=8 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  root=PARTUUID=907af7d0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait net.ifnames=0 quiet loglevel=0 vt.global_cursor_default=0 snd_bcm2835.enable_hdmi=0
[    1.011706] bcm2708_fb soc:fb: FB found 1 display(s)
[    1.014545] bcm2708_fb soc:fb: Registered framebuffer for display 0, size 16x16
```

6by9, 2020-12-07:
You are quite right. I'll sort a firmware patch out.

6by9, 2020-12-08:
Firmware patch is merged internally. It should be in the next rpi-update release.

popcornmix, 2021-01-04:
@MichaIng please test latest rpi-update firmware and close if happy.

MichaIng, 2021-01-10:
Many thanks, it works fine now. I guess those two error messages are expected then and cannot be muted when disabling framebuffers firmware-wise, right?
```
[    1.332206] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.
[    1.332242] bcm2708_fb: probe of soc:fb failed with error -2
```

closed 2021-01-10 by MichaIng