raspberrypi/firmware #1508: Raspberry PI4 ne démarre pas sur un SSD

jppfirst, 2020-12-06:
Bonjour,
J'ai un raspberry PI 4 4megs.
J'ai créé une carte SD (32go) à partir de l'image : 2020-08-20-raspios-buster-armhf-full.img.
Le Raspberry démarre correctement sur la carte SD et j'ai atcivé le BOOT sur SSD.
Ensuite, j'ai copié la carte SD sur un SSD 240 go à partir de l'option Accessoires => SD Card Copier.
Ensuite, j'ai arrêté le Raspberry pour enlever la carte SD et je l'ai redémarrer sur le SSD.
Je vous joint une copie de l'écran de démarrage qui boucle sur les 11 dernières lignes.
Merci pour votre aide.
[image]

pelwell, 2020-12-06:
To save everyone else the translation task:

Hello,
I have a raspberry PI 4 4megs.
I created an SD card (32gb) from the image: 2020-08-20-raspios-buster-armhf-full.img.
The Raspberry boots correctly on the SD card and I have activated the BOOT on SSD.
Then I copied the SD card to 240gb SSD from Accessories => SD Card Copier option.
Then I shut down the Raspberry to remove the SD card and restart it on the SSD.
I am enclosing a copy of the start screen which loops over the last 11 lines.
Thanks for your help.

jppfirst, 2020-12-06:
Hello,
I have a PI 4 4megs raspberry.
I created an SD card (32gb) from the image: 2020-08-20-raspios-buster-armhf-full.img.
The Raspberry boots correctly on the SD card and I have activated the BOOT on SSD.
Then I copied the SD card to a 240 SSD going from the Accessories => SD Card Copier offer.
Then I shut down the Raspberry to remove the SD card and restart it on the SSD.
I am enclosing a copy of the start screen which loops over the last 11 lines.
Thanks for your help.

timg236, 2020-12-07:
The bootloader is reporting the firmware is missing from the USB boot device so presumably, the files aren't there.

I think that that the SD Card Copier supports creating USB MSD devices from and SD card.

I recommend using the Raspberry Pi Imager to create a new OS install on the USB SSD device.

andrewufrank, 2021-03-03:
I have tried with raspi4 bullseye daily 2021-02-26 09:58:-0600 and it starts if written on an SD card (and drives dual monitors - great!). It does not start when the image is written (with disk imager) onto an SSD. Copying the initialized image from the SD to the SSD (with gparted, copy paste) does not start either. My raspi4 with 8gb starts other images from SD without problems with the same USB-SATA bridge. 
There seems to be something wrong with accessing the SSD connected to the USB3 port. If more details about errors etc. are helpful, please ask, I can reconstruct the installation easily. Thank you for getting the dual monitors to run!