raspberrypi/firmware #1509: xHC boot fails in Compute Module 4 IO board

trumpetsven, 2020-12-08:
Hi,
I am trying to get xHC Boot to work on Compute Module 4 IO board but I am failing.

Here is what I did:
- Installed Bootlader pieeprom-2020-11-24.bin
```
 $ vcgencmd bootloader_version
Nov 24 2020 15:08:04
version 6fdd3d2be0127f72508c4c2ed48df7b260637ed9 (release)
timestamp 1606230484
update-time 0
capabilities 0x0000001f
```
- with the following config
```
$ vcgencmd bootloader_config
[all]
BOOT_UART=1
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf15
```

- With Win32 Disk Imager wrote "2020-12-02-raspios-buster-armhf-lite.img" on a USB Stick.
- Plugged the USB Disk into the CM4IO Board
- Power On
- From the Log I can see the following
```
PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:6fdd3d2b DATE: Nov 24 2020 TIME: 15:08:08 BOOTMO                                                                                                 DE: 0x00000006 part: 0 BUILD_TIMESTAMP=1606230484 0x74049526 0x00c03140 0x000b31                                                                                                 1c
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200
PCI reset
PCI reset
PCIe timeout: 0x0000008f
USB xHC init failed
Boot mode: BCM-USB-MSD (05) order f1
XHCI-STOP
xHC ver: 272 HCS: 01000140 0c0000f1 07ff000a HCC: 0220fe65
xHC ports 1 slots 64 intrs 1
USB2[1] 000206e1 connected
USB2[1] 00200e03 connected enabled
USB2 root HUB port 1 init
DEV [01:00] 2.00 000000:01 class 9 VID 0424 PID 2514
HUB init [01:00] 2.00 000000:01
HUB [01:00] 2.00 000000:01 init port 1 speed 3
DEV [02:01] 2.16 000001:01 class 0 VID 0781 PID 5583
MSD device [02:01] 2.16 000001:01 conf 0 iface 0 ep 81#512 02#512
MSD [02:01] 2.16 000001:01 register MSD
MSD [02:01] 2.16 000001:01 LUN 0
MSD INQUIRY [02:01] 2.16 000001:01
MSD [02:01] 2.16 000001:01 lun 0 block-count 60088320 block-size 512
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00082000,59555840 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1 r-dir 2 r-sec 0
PM_RSTS: 0x00001000
Trying partition: 0
lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1 r-dir 2 r-sec 0
Read config.txt bytes     1784 hnd 0x000003e6 sha256 6c0f63e7068a74e6
Read start4.elf bytes  2213024 hnd 0x000118be sha256 b42d4b60ca45b461
Read fixup4.dat bytes     5423 hnd 0x000003f9 sha256 df4a063e368cebad
0x00c03140 0x00000000 0x0000003f
MEM GPU: 76 ARM: 947 TOTAL: 1023
Starting start4.elf @ 0xfeb00200 partition 0
PCI reset
Stop display
Display stopped
```

Maybe someone can help me?

timg236, 2020-12-08:
It looks as though the bootloader was able to load start4.elf from the USB MSD. Please can you add uart_2ndstage=1 to config.txt to see if start.elf logging reveals anything.

trumpetsven, 2020-12-09:
Sorry, my fault. It is working exactly as I described. I forgot that a new Image does not do any serial output or has SSH enabled and I do not have a display... sorry! Thanks for your help!

closed 2020-12-09 by trumpetsven