raspberrypi/firmware #1511: No booting with kernel 5.4.83

mirak011, 2020-12-15:
After last update to kernel 5.4.83 Raspberry Pi is not booting, tested with Raspberry Pi B and Raspberry Pi A. (It is described as
kernel: Bump to 5.4.81, however it is Bump to 5.4.83)

HiassofT, 2020-12-15:
Kernel 5.4.83 looks fine here, but the (GPU) firmware bump seems to prevent booting on RPi0-3 (checked here on 3B+). Going back to Dec 7 firmware (start*.elf, fixup*.dat, bootcode.bin) restored proper boot
```
root@raspberrypi:~# uname -a
Linux raspberrypi 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux
root@raspberrypi:~# vcgencmd version
Dec  7 2020 18:55:36 
Copyright (c) 2012 Broadcom
version e8da211fb125d8be6100f678806deb2af47a8064 (clean) (release) (start)
```

timg236, 2020-12-15:
Thanks. We've reproduced the problem, will post an update once we know more.

popcornmix, 2020-12-15:
@mirak011 please test latest update.

mirak011, 2020-12-15:
> @mirak011 please test latest update.

Tested with Raspberry Pi 0W, 1A, 1B, 2B, 3B, 3B+ and it has been fixed. Thank you very much!

closed 2020-12-15 by popcornmix

bulletmark, 2020-12-16:
Can somebody suggest how to recover from this? I downloaded the new/fixed `raspberrypi-bootloader-20201215-2` package which had the following files:
```
-rw-r--r-- 1 mark mark   52456 Dec 16 08:49 bootcode.bin
-rw-r--r-- 1 mark mark    3172 Dec 16 08:49 fixup4cd.dat
-rw-r--r-- 1 mark mark    5433 Dec 16 08:49 fixup4.dat
-rw-r--r-- 1 mark mark    8438 Dec 16 08:49 fixup4db.dat
-rw-r--r-- 1 mark mark    3172 Dec 16 08:49 fixup_cd.dat
-rw-r--r-- 1 mark mark    7299 Dec 16 08:49 fixup.dat
-rw-r--r-- 1 mark mark   10288 Dec 16 08:49 fixup_db.dat
-rw-r--r-- 1 mark mark  790460 Dec 16 08:49 start4cd.elf
-rw-r--r-- 1 mark mark 3715592 Dec 16 08:49 start4db.elf
-rw-r--r-- 1 mark mark 2214880 Dec 16 08:49 start4.elf
-rw-r--r-- 1 mark mark  790460 Dec 16 08:49 start_cd.elf
-rw-r--r-- 1 mark mark 4787592 Dec 16 08:49 start_db.elf
-rw-r--r-- 1 mark mark 2939168 Dec 16 08:49 start.elf
```
I then copied these to the partition on my SD card where they were already existing but the Pi3 still failed to boot.

popcornmix, 2020-12-16:
Assuming you are on a Pi0/1/2/3 and you don't have custom config.txt settings (e.g. gpu_mem=16, start_x=1 or start_file=xxx) and are not using noobs then the two files needed are:
[start.elf](https://github.com/raspberrypi/firmware/raw/master/boot/start.elf) and [fixup.dat](https://github.com/raspberrypi/firmware/raw/master/boot/fixup.dat).
Replacing those two of the boot partition of sdcard should fix it.
If that doesn't work post the content of config.txt and cmdline.txt

bulletmark, 2020-12-16:
@popcornmix thanks but I discovered it was a faulty power supply. Timing seems odd but I guess the repeated power cycling flaked out the supply so it was only "partially" booting. Swapped the supply out and all good.

coastalplain, 2021-01-12:
Was 5.4.83 removed?  I see 5.4.81 and then 5.10.0 in the [commits](https://github.com/raspberrypi/firmware/commits), but not 5.4.83.  I'm sure I must have missed it somewhere.

popcornmix, 2021-01-13:
Looks like the commit message was wrong and that was actually 5.4.83
https://github.com/raspberrypi/firmware/blob/8cd76653b88939baf25c3f9d9ce90657bcc19b76/extra/uname_string7