raspberrypi/firmware #1512: Pi4+ FKMS does not appear to robustly support 4K at 60hz (4k60), only 30hz

alanbork, 2020-12-16:
summary: Tested Pi4 running FKMS on two different 4k displays, with 2 different HDMI cables.  both work at 30hz but neither work at 60hz. hdmi_enable_4kp60=1 no help. Using the HDMI port closest the USB-C power.

details:

I'm trying to get aPi4B (purchased right when they were launched) to work at 4k resolution and 60hz refresh. But the best I can consistently get is 4k30p. It seems like a lot of people have complained on the forums about not getting 4k60 to work, with no followup suggesting they ever found a solution, so I'm starting to wonder how many confirmed instances of 4k60 working there are.

I've tested two displays and neither work.

The PC monitor I tested is: Samsung U28H750

The TV I tested is: LG UK6300

both support 3840x2160x60hz using non raspberry pi hardware (desktop pc and apple tv). The pi knows about the resolutions I want to use, see photo of calls to tvservice; https://imgur.com/a/Uq2MYbT. I'm not using TVservice to set the mode, just to report all modes supported. For testing the actual desired mode I used the xrandr gui front-end on the pi desktop.

mode 95 works fine on both displays but if I try mode 97 on either I get "no signal" from both displays. One time out of many attempts the TV actually did work in mode 97, but only after plugging and unplugging the HDMI port a lot.

At first I tested with a HDMI cable that worked fine using an apple tv at 4k, but that required an adapter to plug into the pi4. 
I suspected the adapter could be the issue, so I also tried a HDMI cable from monoprice that advertised explicitly supporting the mode in question and could be plugged in without an adapter: https://www.monoprice.com/Product?p_id=7557

I'm using FKMS and the stock, latest version of the OS downloaded 12/11/2020:
https://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2020-12-04/2020-12-02-raspios-buster-armhf.zip

I also tried an older version of the OS downloaded in July, with no improvement. 

My config.txt:

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0

# uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
hdmi_drive=1

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack

dtoverlay=vc4-fkms-v3d
max_framebuffers=1
hdmi_group=1
hdmi_mode=4
hdmi_enable_4kp60=1

[all]

#1.2mbs, "high speed"
dtparam=i2c_arm=on
dtparam=i2c_arm_baudrate=1200000

nanoant, 2021-04-13:
Looks like a duplicate of https://github.com/raspberrypi/firmware/issues/1392

alanbork, 2021-04-13:
> Looks like a duplicate of #1392

idk - it worked on the older firmware. #1392 is a bug about EDID, which would have meant it never worked.

nanoant, 2021-04-13:
> possible. but since it did work once

Interesting, maybe it would work for me too if I could downgrade my brand new Pi 400 firmware 😄

Anyways there in https://github.com/raspberrypi/firmware/issues/1392 are 3 more reports that do not look like just an EDID issue. And I am trying to connect all the similar/same issues/reports including posts on the forum so we can gain some momentum with this issue.

alanbork, 2021-04-13:
if you look I posted anther report of problems with 4k60 that's definitely drivers not EDID because it worked (but with tearing) in January. So rolling back the drivers might help you.

nanoant, 2021-04-15:
> if you look I posted anther report of problems with 4k60 that's definitely drivers not EDID because it worked (but with tearing) in January. So rolling back the drivers might help you.

Oh okay. That's interesting piece of information. But you don't have an exact date or hash of the firmware that was working for you by chance?

alanbork, 2021-04-15:
I still don't have access to the hardware listed in the original report, but two other 4k tvs work fine at 3840x and neither like 4096x at 60hz. rpi engineer is currently looking into it. it's very possible that I am miss-remembering and never got 4096x to work, just 3840x.

nanoant, 2021-04-15:
> if you look I posted anther report of problems with 4k60 that's definitely drivers not EDID because it worked (but with tearing) in January. So rolling back the drivers might help you.
> I still don't have access to the hardware listed in the original report, but two other 4k tvs work fine at 3840x and neither like 4096x at 60hz. rpi engineer is currently looking into it. it's very possible that I am miss-remembering and never got 4096x to work, just 3840x.

I am re-reading what you have wrote, and I am getting confused. I cannot find that "another report" your are mentioning. Also I believe 4K UHD@60Hz never worked Samsung U28H750 and LG UK6300 UHD devices from the first post.

Do I understand you right that it works on two other 4k TVs which are completely different hardware than Samsung U28H750 and LG UK6300 UHD? If Yes, then ok, because Pi4 seems drive 4K@60Hz on some devices with FKMS, but fails with others, including mine. But it does not mean that downgrading will make my monitor work.

alanbork, 2021-04-15:
> find that "another report" your are mentioning. Also I believe 4K ***@***.***Hz
> Yes, then ok, because Pi4 seems drive ***@***.***Hz on some devices with FKMS,

sorry. to restate: 4096x2160 has never worked for me on any display. only
fkms even attempts to run it, but it's unknown if that code path actually
works (by me, anyway).
3840x2160-60 works on both a sony and TLC tv I've personally tested on.

alanbork, 2021-04-21:
it seems to work now both with fkms and kms. kms only works reliably with 

core_freq=600
core_freq_min=600

in config.txt

and fkms requires those too lines to support 4096x2160.

I'm calling this closed.  dependence of 4096x on overclocking should be documented somewhere.

closed 2021-04-21 by alanbork