raspberrypi/firmware #1513: Firmware/Kernel update is bricking multiple RPi Zero W

yik77, 2020-12-16:
**I believe this is right place (or one of) to report this issue. It is potentially very high impact issue.** 

Simply put, upgrading to latest stable and default version of kernel/firmware reproducibly causes frozen system (LED sometimes keeps blinking-7 blinks) and results in a bricked, unbootable device. [screenshot](https://i.postimg.cc/fyp6NsBQ/Screenshot-from-2020-12-15-21-26-50.png)

I understand your healthy skepticism with this statement, but I did verified everything I can, before voicing this strong statement. Let me elaborate:

I have tested it with MULTIPLE brand new SD cards, [Sandisk, bought in a brick and mortar store] with MULTIPLE power supply options [Canakit official power supply unit, a table lamp with USB, a power bank, and finally with laboratory electronic power supply unit. I have tested it with "old" and with another "new" pi zero w board, so with 2 different pi zero w boards, with 2 completely separate devices. Same faulty behavior.
I have been monitoring them by having my HDMI monitor connected in, and watching boot messages and system log. I have also replaced power cables few times. There are NO peripherals on either of my pi zero w boards. No keyboads, no USB, nothing attached. Nothing put power.

I have tested it with my "old" pi zero w, running very light load, mostly pihole, dnscrypt-proxy and local intranet clipboard. This device was rock stable and I have seen maybe 4-5 kernel level crashes in the last 2 years, so 5x in last 500+ days. In last few days, however, with new kernel released, during/after upgrading, it start failing on constant basis. Often during the upgrade itself or during the first consequent boot. Very bad situation.

I do have 2 more pictures of kernel panic, it may be related to wifi firmware/drivers/kernel/kernel modules. I believe this is rather important finding, which may possibly impact many users and hit raspberry ecosystem...

2 more screenshots are [crash during apt-get update](https://i.postimg.cc/fTdzB3Fz/503-E0-E6-E-CA57-4-AB6-BF88-116-E916-F71-F6.jpg) and [crash of freshly burned Pi OS lite image during first boot](https://i.postimg.cc/3wLzMvx0/D1-B00866-1-E4-C-4-BCB-B555-8-C8-A1-FC17926.jpg)

**To reproduce**
install latest package with kernel/firmware

**System**
plain vanilla Pi OS lite image. 

* PiZeroW (tested on 2 separate devices)
* Which kernel version? This is shown in the linked screenshot.

mloebl, 2020-12-17:
I think I may be seeing this too, not sure.  I foolishly was having some issues with wireless and not knowing the age of this Pi I just got, and, I tried running rpi-update.  Not something I do often at all, but was curious if there may have been an update/fix to come down.  Board will no longer boot, even with bt and wifi disabled.  I get kernel panic on BCM2835 on boot typically after it tries to do anything related to networking after wpa_supplicant.  Reflashed with a couple SD cards.

yik77, 2020-12-17:
I do see multiple random reboots in the middle of night...something wrong 

```
Dec 17 07:17:09 oldpizero kernel: mc: Linux media interface: v0.10
Dec 17 07:17:09 oldpizero kernel: vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
Dec 17 07:17:09 oldpizero kernel: bcm2835_vc_sm_cma_probe: Videocore shared memory driver
Dec 17 07:17:09 oldpizero kernel: [vc_sm_connected_init]: start
Dec 17 07:17:09 oldpizero kernel: [vc_sm_connected_init]: installed successfully
Dec 17 07:17:09 oldpizero kernel: videodev: Linux video capture interface: v2.00
Dec 17 07:17:09 oldpizero kernel: bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
Dec 17 07:17:09 oldpizero kernel: bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
Dec 17 07:17:09 oldpizero kernel: bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
Dec 17 07:17:09 oldpizero kernel: bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
Dec 17 07:17:09 oldpizero kernel: bcm2835-isp bcm2835-isp: Device node output[0] registered as /dev/video13
Dec 17 07:17:09 oldpizero kernel: bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
Dec 17 07:17:09 oldpizero kernel: bcm2835-isp bcm2835-isp: Device node capture[0] registered as /dev/video14
Dec 17 07:17:09 oldpizero kernel: bcm2835-isp bcm2835-isp: Device node capture[1] registered as /dev/video15
Dec 17 07:17:09 oldpizero kernel: bcm2835-isp bcm2835-isp: Device node stats[2] registered as /dev/video16
Dec 17 07:17:09 oldpizero kernel: bcm2835-isp bcm2835-isp: Register output node 0 with media controller
Dec 17 07:17:09 oldpizero kernel: bcm2835-isp bcm2835-isp: Register capture node 1 with media controller
Dec 17 07:17:09 oldpizero kernel: bcm2835-isp bcm2835-isp: Register capture node 2 with media controller
Dec 17 07:17:09 oldpizero kernel: bcm2835-isp bcm2835-isp: Register capture node 3 with media controller
Dec 17 07:17:09 oldpizero kernel: bcm2835-isp bcm2835-isp: Loaded V4L2 bcm2835-isp
Dec 17 07:17:09 oldpizero kernel: snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
Dec 17 07:17:09 oldpizero kernel: bcm2835-codec bcm2835-codec: Device registered as /dev/video10
Dec 17 07:17:09 oldpizero kernel: bcm2835-codec bcm2835-codec: Loaded V4L2 decode
Dec 17 07:17:09 oldpizero kernel: bcm2835-codec bcm2835-codec: Device registered as /dev/video11
Dec 17 07:17:09 oldpizero kernel: bcm2835-codec bcm2835-codec: Loaded V4L2 encode
Dec 17 07:17:09 oldpizero kernel: bcm2835_audio bcm2835_audio: card created with 8 channels
Dec 17 07:17:09 oldpizero kernel: bcm2835-codec bcm2835-codec: Device registered as /dev/video12
Dec 17 07:17:09 oldpizero kernel: bcm2835-codec bcm2835-codec: Loaded V4L2 isp
Dec 17 07:17:09 oldpizero systemd[1]: Found device /dev/serial1.
Dec 17 07:17:11 oldpizero systemd[1]: Found device /dev/disk/by-partuuid/4b988ef2-01.
Dec 17 07:17:12 oldpizero kernel: cfg80211: Loading compiled-in X.509 certificates for regulatory database
Dec 17 07:17:12 oldpizero kernel: cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
Dec 17 07:17:13 oldpizero kernel: brcmfmac: F1 signature read @0x18000000=0x1541a9a6
Dec 17 07:17:13 oldpizero kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
Dec 17 07:17:13 oldpizero kernel: usbcore: registered new interface driver brcmfmac
Dec 17 07:17:13 oldpizero kernel: brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt failed with error -2
Dec 17 07:17:13 oldpizero kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
Dec 17 07:17:13 oldpizero kernel: brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
Dec 17 07:17:13 oldpizero kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3
```

mloebl, 2020-12-17:
So for mine I was able to run, `sudo apt update; sudo apt install --reinstall raspberrypi-bootloader raspberrypi-kernel` however still dies on boot with the RaspOS Lite, **2020-12-02-raspios-buster-armhf-lite**. If I hold the shift key down during the boot sequence for the cpu governor during boot, that's the only way I can get it to boot without a stack track and a "kernel stack is corrupted".  I've used several SD cards at this point as well, so confident it's not that.  Using a Canakit 2.5a p/s, but tried a 2nd p/s as well, so I don't believe it's a power supply issue either (considering it's a Pi Zero w, not much power is required anyways so that's overkill.)   I am at least able to get it up and connect via wifi.

mloebl, 2020-12-17:
@yik77 I'm curious if this is related or the same issue... I reverted my `/etc/init.d/raspi-config` to `ondemand`, then in the `/boot/config.txt`  I set the following and now booting cleanly:
```
arm_freq_min=100
arm_freq_max=700
arm_freq=700
```
It's booting cleanly now and wireless working fine.  I stumbled on this bug and wondered if related; https://github.com/raspberrypi/firmware/issues/1431

yik77, 2020-12-17:
maybe there is something with powersave...
```
Dec 17 07:17:25 oldpizero raspi-config[292]: Checking if shift key is held down:Error opening '/dev/input/event*': No such file or directory
Dec 17 07:17:25 oldpizero systemd[1]: Started Avahi mDNS/DNS-SD Stack.
Dec 17 07:17:25 oldpizero raspi-config[292]:  No. Switching to ondemand scaling governor.
Dec 17 07:17:25 oldpizero systemd[1]: Started LSB: Switch to ondemand cpu governor (unless shift key is pressed).
Dec 17 07:17:26 oldpizero kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled 
```
last line is in red, and i see it often in those crashes...

mloebl, 2020-12-17:
@yik77 I spoke too soon, it's dying in an hour or two as well. Screen is all corrupted as well, so can't see any stack traces.

yik77, 2020-12-17:
some people suggest to use a dietpi (an alternative distribution) instead, claiming that their firmware/kernel is not broken...this is pretty weak solution...in my opinion...

MichaIng, 2020-12-18:
DietPi uses exactly the same kernel 😉. We apply `schedutil` CPU governor by default, but that does not seem to be the issue, if `powersave` leads to the crash as well..

@mloebl
Just in case you used it before, remove `arm_freq_min` from your `config.txt`, as it caused issues with prior kernel versions and I am not sure if everything about it has been solved: #1431
`arm_freq_max` is not valid setting, AFAIK, is it?

It's suspicious that the kernel error happens exactly when the CPU governor has been applied 🤔. Did anyone try to disable the service that sets it? `systemctl disable raspi-config`

mloebl, 2020-12-18:
@MichaIng Just removed `arm_freq_min`, thank you!  I upgraded to the last kernel/firmware from rpi-update and I think it's running the longest it has been... see if it makes it through the night.  I have another Pi Zero W coming on Sat if Amazon doesn't lose it, so can see if it's a fluke of maybe a faulty Pi.  I have not yet tried disabling `raspi-config` , but will try it on the 2nd one what I get if it has the same issues. It's funny as I've been using Pi's for years, but didn't realize until the last couple days that there was a raspi-config service until I was taking a look at the service script today.

yik77, 2020-12-18:
> .... I have another Pi Zero W coming on Sat if Amazon doesn't lose it, so can see if it's a fluke of maybe a faulty Pi. I have not yet tried disabling `raspi-config` , but will try it on the 2nd one what I get if it has the same issues. ...

That was the solution I tried, too, getting new rpi zero w board. No fix. no, it was not a HW issue, nor a sd card issue, nor power supply issue. But without attempting to do sudo apt get upgrade, it is running for 11 hours, so far the record...

pelwell, 2020-12-18:
You say it is a "plain vanilla Pi OS lite image." Does that mean you haven't changed the configuration in any way - `config.txt`, `cmdline.txt` etc.? Please post the contents of those files so we can verify.

yik77, 2020-12-18:
> You say it is a "plain vanilla Pi OS lite image." Does that mean you haven't changed the configuration in any way - `config.txt`, `cmdline.txt` etc.? Please post the contents of those files so we can verify.

Confirm. I did not modified any of those. Never touched any of those in my life. I am afraid to mess with kernel. All I did was touch /boot/ssh (to enable ssh) and nano /boot/wpa_supplicant.conf with my wifi name and password (to enable wifi). That is all. 

Do you still need those files? they are in images of Pi OS Lite (used to download and dd them, and also tried using rpi-imager package. (that verifies hash/checksum of the downloaded lite image and its checksum is different than image you get from `https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-12-04/2020-12-02-raspios-buster-armhf-lite.zip`)

mloebl, 2020-12-18:
@yik77  Last night I updated to the latest kernel/firmware thru rpi-update to `42c900ed58a853a09f9d7abb4785b351777edabf` (latest) to see if I could see a difference. Without video connected currently, and over 12 hours up, longest since I flashed `raspios_lite_armhf-2020-12-04/2020-12-02-raspios-buster-armhf-lite.zip`

yik77, 2020-12-18:
> @yik77 Last night I updated to the latest kernel/firmware thru rpi-update to `42c900ed58a853a09f9d7abb4785b351777edabf` (latest) to see if I could see a difference. Without video connected [...]

I have almost 24 hours uptime with ssh pi@oldpizero.local on `Linux oldpizero 5.4.79+ #1373 Mon Nov 23 13:18:15 GMT 2020 armv6l`  In your opinion, is hooking up to hdmi monitor making it more or less stable? By Murphy's law, it may be less likely to fail, when under closed supervision, with screen, shh with journalctl -f, hdmi monitor, all watching system log...

mloebl, 2020-12-18:
> @yik77 Last night I updated to the latest kernel/firmware thru rpi-update to 
> I have almost 24 hours uptime with ssh [pi@oldpizero.local](mailto:pi@oldpizero.local) on `Linux oldpizero 5.4.79+ #1373 Mon Nov 23 13:18:15 GMT 2020 armv6l` In your opinion, is hooking up to hdmi monitor making it more or less stable? By Murphy's law, it may be less likely to fail, when under closed supervision, with screen, shh with journalctl -f, hdmi monitor, all watching system log...

That's what I am wondering too. I had noticed that when I had the monitor hooked up, when it crashed in the scenario where it was running for a period of time (usually 1 -2 hours), the screen was usually corrupted/distorted/unreadable.  I booted the Pi Zero W this last time yesterday with no monitor and have been doing everything via SSH since it's been "stable".

pelwell, 2020-12-18:
If you've not edited the config files at all then I don't need to see them.

> By Murphy's law, it may be less likely to fail when under close supervision

Poor old Murphy is so misunderstood. The British "Sod's Law" is closer to what you mean.

Back in the 80s (I'm that old) I wrote a printer driver for a dot matrix printer to print screenshots from a C64. It would occasionally drop a single byte, causing an 8-high partial row of pixels to shift to the left by one dot, _except_ when I stood over it watching.