raspberrypi/firmware #1515: Difficulty booting from SSD drive using stable ieeprom-2020-12-11.bin

Britinparis18, 2020-12-19:
I have my Raspberry Pi 4 Model B Rev 1.1 booting from an SSD drive. 
I have my firmware set to the stable branch but after the recent update to pieeprom-2020-12-11.bin I have been experiencing startup problems.

Specifically, on booting the SSD flashes once, and its blue indicator led turns a faint red.
The drive is then not recognized until it is disconnected and reconnected.
If the raspberry boots from its SD card, the SSD drive is not recognized, until it is reconnected.
A Reinsert during the booting loop, causes the disk to be recognized and to successfully boot from it.

My SSD drive is reported as: ID 125f:a88a A-DATA Technology Co., Ltd. 
I have a changed /boot/cmdline.txt file adding: 
usb-storage.quirks=125f:a88a:u

Initially, I thought I had developed a USB 3 port problem, or insufficient power supply from my Raspberry power brick.
Booting the same code from a USB Stick works without issue. 

Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage4
Nov 30 2020 22:12:08 
Copyright (c) 2012 Broadcom
version ab1181cc0cb6df52bfae3b1d3fef0ce7c325166c (clean) (release) (start)
Linux raspberry 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux

vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41

Changing the Firmware to the 'critical' version, resolved my problem.  
sudo rpi-eeprom-update -d -f /lib/firmware/raspberrypi/bootloader/critical/pieeprom-2020-09-03.bin

Changing it back to pieeprom-2020-12-11.bin reintroduces the issue

achessoux, 2020-12-20:
Hello,

I observe exactly the same issue with my Pi 4 and my SSD NVMe connected through x872 board powered by internal 5V power supply (provided by x735 board): with the stable firmware from Dec 11, SSD is only recognized (during boot or even after when system is started) if I unplug it then plug. Whereas with the critical firmware from Sep 3, everything is fine.

Shall I provide some logs or other information? (Tell me which one)

Regards,
Aymeric

Roberto6969, 2020-12-20:
Same problem at me on two RPi 4, both with Kingspec SSD. One disk is 128 GB, the other is 256 GB.
After yesterday's update to the latest version both doesn't boot anymore. Only if I pull out SSD and insert it again in right moment, it boots. Before updates both booted without single problem. 

**$ ls /lib/firmware/raspberrypi/bootloader/stable/**
pieeprom-2020-04-16.bin  pieeprom-2020-07-16.bin  pieeprom-2020-09-03.bin  recovery.bin        vl805-000138a1.bin
pieeprom-2020-06-15.bin  pieeprom-2020-07-31.bin  pieeprom-2020-12-11.bin  vl805-000137ad.bin
**$ sudo rpi-eeprom-update**
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Fri 11 Dec 11:15:17 UTC 2020 (1607685317)
 LATEST: Fri 11 Dec 11:15:17 UTC 2020 (1607685317)
 FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1
**$ vcgencmd bootloader_version**
Dec 11 2020 11:15:17
version c3f26b6070054bca030366de2550d79ddae1207a (release)
timestamp 1607685317
update-time 1608459922
capabilities 0x0000001f

ncodee, 2020-12-24:
Having the same issue here on the RPi 4 v.1.4.

Roberto6969, 2020-12-27:
Hello, will someone check this issue? Issue is really annoying and makes Raspberry pi 4 conditionally usable.

Thank You!

JamesH65, 2020-12-27:
The Trading office is shut until 4th January.  Hopefully someone will have time to look then.

Roberto6969, 2021-01-11:
@JamesH65  I don't want to be annoying but this bug makes RPi4 quite unreliable. Will be any new release of firmware in a short? Thank You!

JamesH65, 2021-01-11:
Have you tried the very latest kernel and firmware? @pelwell Could any recent fixes be relevant?

pelwell, 2021-01-11:
Does this comment help? https://github.com/raspberrypi/rpi-eeprom/issues/266#issuecomment-756005465

Roberto6969, 2021-01-12:
@pelwell of course!
I can confirm the Solution:
"sudo -E rpi-eeprom-config --edit" and adding following line:
USB_MSD_PWR_OFF_TIME=0

Tested on two RPi4 with today's latest updates. 
Thank You!

achessoux, 2021-01-15:
Firmware update to 2021-01-11 stable version solves the issues (without changing the USB_MSD_PWR_OFF_TIME which is more covenient).
[2021-01-14 - Promote pieeprom-2021-01-11 to STABLE (LATEST)](https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/release-notes.md#2021-01-14---promote-pieeprom-2021-01-11-to-stable-latest)

To perform the update:

1. change the release stream to stable by changing the FIRMWARE_RELEASE_STATUS entry to "stable" in the following file:
                    `sudo nano /etc/default/rpi-eeprom-update`
2. Install the firmware update using:
                    `sudo rpi-eeprom-update -a`
3. Reboot:
                    `sudo reboot`
4. Verify:
                    `sudo rpi-eeprom-update`

timg236, 2021-01-16:
@achessoux Thanks for the test report

Britinparis18, 2021-01-16:
Thanks achessoux, the last update added a 'default' firmware update, but changing to your stable stream fixes the the problem.

closed 2021-01-16 by Britinparis18

ApricotFarm, 2021-01-26:
@achessoux -- Thank you!! I can confirm that your solution worked for me (updating firmware to 2021-01-14). I can now boot normally from my 128GB Kingspec SSD on my 4GB Pi 4.

ur7x, 2021-01-27:
I'm trying to boot from a Samsung 970 Evo Pro running in a Geekworm X872.  Latest version of both OS and eeprom bootloader

Device refuses to recognize the SSD unless I "unplug and plug" the USB "horse shoe" connector.
Unplug and Plug and all is good.

Is this and X872 issues? a Samsung 970 issue? or something else.

ApricotFarm, 2021-01-27:
@ur7x I had the exact same behavior with my 128GB Kingspec SSD (USB C to Raspi USB 3). Make sure you're using the Jan 14 bootloader or later; I found that the standard updating tools wouldn't bring my bootloader all the way up, so I had to carefully follow the method that @achessoux posted, a few posts earlier. It fixed the problem for me entirely. 

Hop this helps -- good luck!

ur7x, 2021-01-28:
When I run the SUDO rpi-eeprom update I get this response:

BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Fri Dec 11 11:15:17 UTC 2020 (1607685317)
 LATEST: Fri Dec 11 11:15:17 UTC 2020 (1607685317)
 FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1

That's right?  Right?

ApricotFarm, 2021-01-28:
Nope! :-) 
That's the same issue I was having the "standard" update methods weren't updating. You need to change your release channel so you get the newer release. Follow @achessoux's post above meticulously (the four-step process): https://github.com/raspberrypi/firmware/issues/1515#issuecomment-761245134  and when that's done you should have something more like this:
```
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Sat 16 Jan 2021 02:10:13 PM UTC (1610806213)
 LATEST: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/default
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1

```
I don't know why it reports "latest" as 3 Sep 2020, but the current version of 16 Jan is the one that did the trick for me.

ur7x, 2021-01-28:
GOT IT ! 
Thanks so much...  Command line updates were not working...  When into Raspi-config and did the bootloader update there and it took!  Now running "Sat 16 Jan 2021" and surprise surprise... it boots to the Samsung 970 EVO...  Plug and unplug resolved!

achessoux, 2021-01-28:
Nice to see your issue is solved. We should thank the developer of the firmware!

ur7x, 2021-01-29:
Agreed! with one small caveat.  Clearly the  "sudo rpi-eeprom-update -a" tool is lacking.  

I followed your instructions to the letter and was left with not only an old version of the boot loader, but the message I got from the systems is that my old boot-loader was the "latest" one.   Even  ApricotFarm  gets a message that his "Jan 16 2021" bootloader is somehow older than one from  September 03 2020. 

The only way I got the newest Boot loader was to use "sudo raspi-config" and then ask for it under advanced options.  Command line updates were unable to download and update the boot loader... Weird. 

With that, Thanks to the developers of this firmware patch! I love it!

IRTrail, 2023-11-27:
I have the same issue as ur7X here: https://github.com/raspberrypi/firmware/issues/1515#issuecomment-768591440

I have updated the bootloader as per achessoux and also via raspi-config.

I cannot get it to boot from the Geekworm X876/PNY 500G SSD unless I plug and unplug. If power is lost, it reverts to the sdcard.

FWIW this is a rpi for Klipper 3d printer firmware.

Any ideas?