raspberrypi/firmware #1516: Qt5 app with audio freezes Raspberry Pi 400

piec, 2020-12-21:
Hi,

I'm using a Raspberry Pi 400 with default Raspberry Pi OS provided in the pack in the MicroSD card. Upgraded today.
The system freezes after ~1s after starting `gcompris` an educative software for kids (qt), always reproducible.
To my opinion It seems very much related to a firmware getting stuck according to the kernel log.
Interestingly when running without audio everything works fine.

This bug was initially reported here https://bugs.kde.org/show_bug.cgi?id=429125

**To reproduce**
* Install Raspberry Pi OS on a rpi 400
* `sudo apt update && sudo apt upgrade && sudo apt install gcompris-qt`
* `gcompris -m`: without sound, works
* `gcompris -s`: freezes system

Could the audio make the firmware freeze and make further firmware calls fail?

Cheers,
Pierre

---

**Expected behaviour**
System does not freeze

**Actual behaviour**
System freezes

**System**
<details>
  <summary>raspinfo before crashing</summary>

```
pi@raspberrypi:~ $ raspinfo
System Information
------------------

Raspberry Pi 400 Rev 1.0
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage5

Linux raspberrypi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux
Revision	: c03130
Serial		: 10000000b4f1f308
Model		: Raspberry Pi 400 Rev 1.0
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Nov 30 2020 22:12:08
Copyright (c) 2012 Broadcom
version ab1181cc0cb6df52bfae3b1d3fef0ce7c325166c (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Sys. de fichiers blocs de 1K Utilisé Disponible Uti% Monté sur
/dev/root           14986672 9417256    4873540  66% /
devtmpfs             1827800       0    1827800   0% /dev
tmpfs                1959896       0    1959896   0% /dev/shm
[... 603 lines cut ...]
pi@raspberrypi:~ $ sudo rpi-eeprom-update
BCM2711 detected
VL805 firmware in bootloader EEPROM
BOOTLOADER: up-to-date
CURRENT: jeudi 3 septembre 2020, 12:11:43 (UTC+0000) (1599135103)
 LATEST: jeudi 3 septembre 2020, 12:11:43 (UTC+0000) (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1
```

```
pi@raspberrypi:~ $ sudo rpi-eeprom-config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41
```

popcornmix, 2020-12-23:
Do you have two hdmi displays connected?

piec, 2020-12-23:
No just a single 1080p hdmi samsung screen. I tried the 2 mini hdmi outputs, same effect

popcornmix, 2020-12-23:
Might be worth trying latest firmware/kernel from rpi-update.
This moves to 5.10 kernel. Backing up is recommended first.

Donno191, 2021-01-03:
Same Issue with multiple apps Gcompris, snap ect. Using Raspberry Pi 4B 8GB(Note Broser working fine and aplay in CLI)

cipy, 2021-01-04:
Hi there,

Unfortunately this issue is generic today for all RPi 3B+, 4B with the latest stock kernels from Raspbian, including the latest pushed (yesterday) with rpi-update @popcornmix; it doesn't matter if you set the sound beforehand to go to the AV jack or HDMI.

Here's a kernel dump https://mail.kde.org/pipermail/gcompris-devel/2021-January/005936.html

Thanks,
Ciprian

cipy, 2021-01-04:
looks like switching back to the EL33T kernel version (1337) solves this 💯 for now 

> sudo rpi-update e70a90efc09925adbc84e9dded5526ee155e00e0

also the build before this one seem to work, I'll let the fun of finding the breaking change to @popcornmix

cipy, 2021-01-04:
039088f7d929b0d1cc49e8d2012df0652f908518 is the last build when GCompris +audio works on both 3B+ & 4B
ca14404412d6c2a9966d7766bbad6cc24989a33f is the very first iteration when things go kernel blurps

```
firmware: hdmi/edid: Add option to ignore any odd horizontal timings on Pi4
firmware: sdcard: Hybrid MBR - only select GPT if it is the first primary partition See: raspberrypi/firmware#1465
firmware: audioplus: Avoid broken audio when requesting hdmi audio device when using composite display See: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=283639
```

cipy, 2021-01-04:
looks like installing the gstreamer1.0-pulseaudio package solves this problem 🥇

popcornmix, 2021-01-04:
@piec do either of these fix the issue for you?
- installing the gstreamer1.0-pulseaudio package
- switching to firmware/kernel 039088f7d929b0d1cc49e8d2012df0652f908518 with rpi-update?

petitlapin, 2021-01-04:
> @piec do either of these fix the issue for you?
> 
>     * installing the gstreamer1.0-pulseaudio package
> 
>     * switching to firmware/kernel 039088f7d929b0d1cc49e8d2012df0652f908518 with rpi-update?

Both work: (from the GCompris mailing-list https://mail.kde.org/pipermail/gcompris-devel/2021-January/005942.html):
> Yes, you may:
> a) have an up to date Raspbian OS and just install the gstreamer1.0-pulseaudio on top of it,
> b) downgrade your kernel (no need for gstreamer1.0-pulseaudio) to the EL33T version

piec, 2021-01-05:
Thanks for the answers

@popcornmix 
* with `gstreamer1.0-pulseaudio`: works fine (audio as well)
* `039088f7d929b0d1cc49e8d2012df0652f908518` without `gstreamer1.0-pulseaudio`: gcompris works but no audio, no audio at all in the system.
* `039088f7d929b0d1cc49e8d2012df0652f908518` and `gstreamer1.0-pulseaudio`: works but no audio (same as previous)

I'm using the speakers of my hdmi screen

cipy, 2021-01-05:
Hi there,

I had to make it work/test with pulseaudio as this one is the recommended way forward in most recent RPi kernels

Therefore my setup had also:

> apt-get purge -y bluealsa
> apt-get install -y pulseaudio-module-bluetooth

and all the other packages up to date. If you still have *alsa it might explain the difference, or maybe I was off by one 👎 

My backup plan was to find the very first/last kernel at which the GCompris won't freeze, which I believe it is 0390

piec, 2021-01-05:
More info, discussed on the [kde thread](https://bugs.kde.org/show_bug.cgi?id=429125) 
Qt uses gstreamer as audio backend, itself can use different audio sinks. When gstreamer1.0-pulseaudio is installed gstreamer will use that which works fine, if not present it will use omxanalogaudiosink which generates the bug described in the issue (firmware freeze).
It can be reproduced using: `gst-launch-1.0 audiotestsrc ! omxhdmiaudiosink`

piec, 2021-01-05:
@cipy as far as I know I need alsa, it's the layer below pulseaudio
(pulseaudio depends on libasound2)

popcornmix, 2021-01-07:
I've reproduced a hang with `gst-launch-1.0 audiotestsrc ! omxhdmiaudiosink`. Investigating.

popcornmix, 2021-01-07:
Potential fix has been pushed to rpi-update firmware. Can you test?

piec, 2021-01-07:
Thanks I should be able to test tomorrow

piec, 2021-01-08:
@popcornmix :+1: after `rpi-update` `gst-launch-1.0 audiotestsrc ! omxhdmiaudiosink` doesn't crash anymore. It doesn't output sound though.

popcornmix, 2021-01-08:
I get audio here. A fixed tone comes out of hdmi display.
```
pi@pi4:~ $ vcgencmd version
Jan  7 2021 18:27:29 
Copyright (c) 2012 Broadcom
version fb345a0c2d5544957f4ba1a2b9e968970e3312c4 (clean) (release) (start)
pi@pi4:~ $ gst-launch-1.0 audiotestsrc ! omxhdmiaudiosink
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstAudioSinkClock
```
Does the simpler:
```
/opt/vc/src/hello_pi/hello_audio/hello_audio.bin 1
```
produce audio? Has it with any version of firmware?

piec, 2021-01-08:
No luck with these:
```
pi@raspberrypi:.../hello_pi/hello_audio $ vcgencmd version
Jan  7 2021 18:27:29
Copyright (c) 2012 Broadcom
version fb345a0c2d5544957f4ba1a2b9e968970e3312c4 (clean) (release) (start)
pi@raspberrypi:.../hello_pi/hello_audio $ gst-launch-1.0 audiotestsrc ! omxhdmiaudiosink
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstAudioSinkClock
```
--> no sound on my hdmi monitor

```
pi@raspberrypi:.../hello_pi/hello_audio $ ./hello_audio.bin 1
Outputting audio to hdmi
```
--> no sound on my hdmi monitor

```
pi@raspberrypi:.../hello_pi/hello_audio $ gst-launch-1.0 audiotestsrc ! pulsesink
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstPulseSinkClock
```
--> works, I hear sound on my hdmi monitor

---
I don't know if hello_audio is influenced by volumes set in alsamixer, but I checked them anyway

popcornmix, 2021-01-08:
What does "tvservice -a" report?

piec, 2021-01-08:
```
$ tvservice -a
     PCM supported: Max channels: 2, Max samplerate:  48kHz, Max samplesize 24 bits
```
I'm going to test hello_audio.bin after `rpi-update 039088f7d929b0d1cc49e8d2012df0652f908518` to see if it worked

piec, 2021-01-08:
I have no sound on hdmi on my rpi 400 with revision 039088f7d929b0d1cc49e8d2012df0652f908518
Tested with:
* hello_audio.bin 1
* gst-launch-1.0 audiotestsrc ! pulsesink
* gst-launch-1.0 audiotestsrc ! omxhdmiaudiosink

popcornmix, 2021-01-08:
What does "tvservice -s" report?
Not sure what to say if audio wasn't working that long ago (it's obviously not a problem most users have).
Can you try with a different display?
Does hdmi_drive=1 in config.txt make a difference?

piec, 2021-01-11:
I'm checking trying this as soon as possible

piec, 2021-01-15:
```
$ tvservice -s
state 0xa [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive
```
I'll try another display too

do-dat-thang, 2021-03-13:
Has anyone worked to root cause on this? I am hitting an almost identical problem. The end result is the same as originally reported here but the trace-back and apps being run are a bit different
[raspinfo.txt](https://github.com/raspberrypi/firmware/files/6134085/raspinfo.txt). 
[raspinfo.txt](https://github.com/raspberrypi/firmware/files/6134086/raspinfo.txt)

I have Rpi4 with dual HDMI and camera connected.  I have an app that creates and destroys windows and takes pictures of the display to validate. After a bit, the mmal timeout occurs hanging raspistill which is used to take the pictures. Then Xorg hangs up and croaks about every 2 minutes or so.

I didn't think this was related to the audio, but I went ahead and "apt install  gstreamer1.0-pulseaudio".  This had no effect to help the problem.

Attaching raspinfo output.
[raspinfo.txt](https://github.com/raspberrypi/firmware/files/6134087/raspinfo.txt)

Please let me know if this is the right place to ask, or where I should otherwise.

do-dat-thang, 2021-03-13:
Hmmm - looking at raspberrypi.c's rpi_firmware_transaction() source code, the error is very generic.  So, although my problem ends up with the same issue, it is probably worlds apart.

But... still I'm kind of thinking this may be the right place to start(?)  Is it best to put a new issue on the pile? I'm (if you haven't noticed) new to the work flow here. Thought I'd start by trying to find out if my problem has already been resolved before I dig too deep through the code.

popcornmix, 2021-03-14:
If your issue doesn't involve Qt5 and it's only connection with this issue is rpi_firmware_transaction (which basically means firmware has stopped responding which can have many causes) then create a new issue.