raspberrypi/firmware #1517: ioctl(fb, FBIO_WAITFORVSYNC, &dummy) not working at high refresh rates

bill-connelly, 2020-12-23:
**Describe the bug**

We are using ``ioctl(fb, FBIO_WAITFORVSYNC, &dummy)`` to capture monitor timings for a program that directly writes to framebuffer (we have commented out dtoverlay=vc-fkms-v3d in /boot/config.txt). It works perfectly at various 50Hz and 60 Hz refresh rates. However, when set the refresh rates to 100 or 120 Hz with tvservice, the function hangs approximately 1 second. Instead of returning in 1/100 or 1/120th of a second as would be predicted. 

**To reproduce**
A minimal C function to show behaviour is

` #include <stdio.h>
 #include <stdlib.h>
 #include <time.h>
 #include <linux/fb.h>
 #include <string.h>
 #include <sys/ioctl.h>
 #include <fcntl.h>

 int main(void) {
   int fb = open("/dev/fb0", O_RDWR);
   __u32 dummy = 0;

   int i;
   struct timeval tv;
   unsigned long time_us;
   int error_code;
   for (i = 0; i < 10; i++) {
     error_code = ioctl(fb, FBIO_WAITFORVSYNC, &dummy);
     gettimeofday(&tv, NULL);
     time_us = 1000000 * tv.tv_sec + tv.tv_usec;
     printf("Time is %lu, error code is %i\n", time_us, error_code);
   }
 }
`

**Expected behaviour**
Setting the monitor to 50 or 60 hz with a command like ``tvservice -e "cea 31"`` or ``tvservice -e "cea 16"`` and calling the above C function with ``./refreshrate``  should produce an output like a

`Time is 1007133697, error code is 0
Time is 1007150314, error code is 0
Time is 1007166978, error code is 0
Time is 1007183643, error code is 0
Time is 1007200319, error code is 0
Time is 1007216987, error code is 0
Time is 1007233649, error code is 0
Time is 1007250319, error code is 0
Time is 1007266986, error code is 0
Time is 1007283648, error code is 0
`

Note, there is approximately 17 ms between times, as expected for a 60 Hz update. However, setting the monitor to 100 or 120 Hz with ``tvservice -e "cea 63"`` produces the following output

` Time is 1106366765, error code is 0
Time is 1107575047, error code is 0
Time is 1108783371, error code is 0
Time is 1109991698, error code is 0
Time is 1111200022, error code is 0
Time is 1112408352, error code is 0
Time is 1113616677, error code is 0
Time is 1114824996, error code is 0
Time is 1116033327, error code is 0
Time is 1117241653, error code is 0
`

Note, times are separated by 1.1 seconds. Not 1/120 seconds as would be expected.

**System**

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
[... 566 lines cut ...]
[   10.644269] fuse init (API version 7.27)
[   12.808262] bcmgenet fd580000.genet eth0: Link is Up - 1Gbps/Full - flow control off
[   12.808285] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   13.168636] Bluetooth: Core ver 2.22
[   13.168679] NET: Registered protocol family 31
[   13.168683] Bluetooth: HCI device and connection manager initialized
[   13.168695] Bluetooth: HCI socket layer initialized
[   13.168701] Bluetooth: L2CAP socket layer initialized
[   13.168721] Bluetooth: SCO socket layer initialized
[   13.178231] Bluetooth: HCI UART driver ver 2.3
[   13.178238] Bluetooth: HCI UART protocol H4 registered
[   13.178275] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   13.178380] Bluetooth: HCI UART protocol Broadcom registered
[   13.419762] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   13.419769] Bluetooth: BNEP filters: protocol multicast
[   13.419784] Bluetooth: BNEP socket layer initialized
[   13.458557] Bluetooth: RFCOMM TTY layer initialized
[   13.458570] Bluetooth: RFCOMM socket layer initialized
[   13.458583] Bluetooth: RFCOMM ver 1.11

* Which model of Raspberry Pi? Pi4b
* Which OS and version? Buster
* Which firmware version (`vcgencmd version`)? version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start)

* Which kernel version (`uname -a`)? Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

alanbork, 2021-07-07:
why not run with fkms (or kms)?

bill-connelly, 2021-07-07:
@alanbork Because I want to write directly to the frame buffer.

alanbork, 2021-07-07:
out of curiosity only, why? do you get better performance doing that? I've been able to do 120hz  @1080p under kms. it's possible it's dropping frames once in a while though.

bill-connelly, 2021-07-07:
@alanbork If you can point me in the direction of documentation that shows me how to display a bitmap/pixelmap and (and this is absolutely critical) get the timings of vsync, under KMS, I would be very grateful. The whole point of the software I have written is to have the user know when frames are being displayed.

alanbork, 2021-07-07:
are you trying to build a lag tester, like

 https://alantechreview.blogspot.com/2020/08/pilagtester-pro-order-page.html

?
kms has a vsync callback, look up the source to modetest. I haven't tried
to use it, I just infer timings from the return from swapbuffers() or
equivalent.  Note that the timings you get from this are pretty tight, see

https://github.com/alanbork/kmsmodeline

for an example. The average is exactly the refresh rate, even if there's a
tiny bit of nose per frame.

bill-connelly, 2021-07-08:
It's for this:

https://github.com/bill-connelly/rpg
https://www.biorxiv.org/content/10.1101/2020.03.05.979724v1

Scientists want to display stimuli to subjects, but not knowing when they actually turn up on the screen is a pain. 16 milliseconds is a big deal. 

Anyway, there seems to be lots of stuff on your github, and on 
https://github.com/dvdhrm/docs/blob/master/drm-howto/modeset-vsync.c
That I'll try to get into some time.

Still, the bug report is real.

alanbork, 2021-07-08:
have you considered using/raiding the C code base of the psychtoolbox  like
psychopy does? Mario Kleiner has tested on the pi and is pretty meticulous
about timing.  At the very least his code should be useful for
determining how to accurately get timestamps, including workarounds for
buggy drivers, though I don't think any of it is pi specific like your
project. Your comparison to the psychtoolbox is probably contaminated by
running it on windows, which mario likes to stress is not good for real
time graphics. A fairer comparison would be running PTB on a pi.

My own pi work suggests that sometimes the pi goes off to lunch and has
miserable latency, but usually it's quite good, like your data.

But if you really care down to 16ms, you'll need to also characterize your
monitor's lag and response time, plus there's the ulgy fact that previous
frames change how fast the pixels change intensity.

bill-connelly, 2021-07-08:
@alanbork I thought Psycopy was pure python? It rests on pygame doesn't it. And I hadn't thought of looking at what psychtoolbox does because it doesn't/didn't work on Pi. I started working on this project when the Raspberry Pi 3 was new. I just wanted to write to the frame buffer. KISS. Obviously, what has happened with the RPI4 has changed things, so I'm open to going in other directions.

alanbork, 2021-07-08:
Psycopy is't pure python but you might be right that the visual routines
are; I know mario was collaborating with them at some point but don't
recall on what.

Psychtoolbox runs even on the pi3, and mario has tested both pi3 and pi4
performance with good results tested against a photodiode (again, that's
from memory, which might be wrong). Mario was even involved in the coding
of the pi3 gpu driver (according to him). He posted an update on the
toolbox forum in the last year or so if you want to look it up.

from my personal experience the pi3 might be a better choice still as pi4
drivers are still buggy with respect to timing.

PeterPablo, 2021-07-08:
Once I did timing-sensitive research that was sensitive in the sub-us range. May I suggest that while a standalone SW solution is not available to you, that you consider introducing a trigger mechanism that is able to generate more accurate timing information? While I did not look at that publication,  "turn up on the screen" suggests that you might have a useable brightness trigger from the display that can be rather easily captured by an external photodiode and timestamped from a GPIO.

alanbork, 2021-07-08:
yes, a photodiode in the corner detecting a pixel that swaps from black to white on each successive frame. Never a bad idea.  Not really solving his problem though since he's looking for a reliable measure of when the pi sent the frame so he can start uploading the next one to the framebuffer.

bill-connelly, 2021-07-08:
@PeterPablo The whole point of the solution was to avoid those approaches. I just want to supply scientists with a robust turn key solution to displaying visual stimuli. Which I think I have, but only up to 60 Hz.